betaflight/betaflight #6893: Runaway Takeoff Prevention

Strangedave, 2018-10-06:
While running a log for Plazmatree I caused the runaway takeoff prevention feature to trigger. Resulting in a pretty harry crash. I have dvr video and the log and diff. Log and Diff attached. the video was too big so I uploaded it to YouTube - https://youtu.be/DJMH5McUjQY

Is it possible to turn the Runaway Takeoff down a little, or is this just a "don't do that any more" kind of thing? Just wanted to make sure someone knew about this in case it's a firmware issue.

I did preform the exact same moves on a matching quad yesterday, no problem on that one.

[BTFL_BLACKBOX_LOG_WIN_5R_20181006_115251.zip](https://github.com/betaflight/betaflight/files/2453225/BTFL_BLACKBOX_LOG_WIN_5R_20181006_115251.zip)
[Diff 3-5-1 v1.txt](https://github.com/betaflight/betaflight/files/2453227/Diff.3-5-1.v1.txt)

etracer65, 2018-10-06:
Doesn't look like you got throttle high enough with flight activity to disable the protection. You seem to be hovering < 20% for most of the flight and the default cutoff is 25%.

So you should lower `runaway_takeoff_deactivate_throttle_percent` to be close to your hover throttle.  Also you probably want to fly around for a few seconds before doing extreme stick moves like that.  Those big moves will cause pidsum spikes that look like things are going out of control.  Yaw is the most likely to trigger because it has the least authority and can't react to the inputs as quickly (causing error and the pidsum to grow rapidly).

etracer65, 2018-10-06:
Unrelated, but I notice that your rc smoothing is not working well because you're running your Taranis in 16 channel mode.  This causes the changes to be updated every 18ms instead of 9ms.

If you run in 8 channel mode then you'll have better results and all the channels will update every 9ms.

[image]

mikeller, 2018-10-07:
@Strangedave:

> If you run in 8 channel mode then you'll have better results and all the channels will update every 9ms.

Or use a non-stupid version of OpenTx that does not hobble 16 channel mode by sending updates only every 18ms: https://github.com/mikeller/opentx/releases

Strangedave, 2018-10-07:
No one has ever even mentioned such a thing to me. I'll definitely get that fixed tonight. There really isn't any need for 16 channels at this point anyways.

0crap, 2018-10-07:
> 
> 
> @Strangedave:
> 
> > If you run in 8 channel mode then you'll have better results and all the channels will update every 9ms.
> 
> Or use a non-stupid version of OpenTx that does not hobble 16 channel mode by sending updates only every 18ms: https://github.com/mikeller/opentx/releases

Interesting!
Is that the only difference compared to the "normal" OpenTX version?
(Set D16 Channel Range to CH1-8 by default.)

mikeller, 2018-10-07:
@0crap: Out of the two channel groups 1 - 8 and 9 - 16, it sends only the channels that have changed in every 9ms slot. Assuming that the channels 9 - 12 are switches that only change very infrequently, channels 1 - 4 (the sticks) are sent almost all of the time.

Strangedave, 2018-10-15:
Ok, so I have been on this issue since my last post. I did update the firmware on my radio as suggested. It dose seem to have helped my other rigs. Updated the firmware on RX and replaced VTX was damaged in the crash. When I got the Kwad running again I took it for a test spin and had twitching motors. Basically could not get it to fly smooth. It would fly, just erratic twitching. I assumed it was something that I had done, so I went through everything. Checked all wiring, all settings, and then reflashed betaflight 3.5.1. Nothing was fixing the issue. So I changed over to the MPU6000 gyro and everything is good to go. My thought is that the ICM20602 GYRO was bad all along. I ordered a new DALRC F722 and will report back once I have that running, but its going to be many days. Sorry for the complaint, and thanks for the advice. I appreciate everything that you all do for the hobby. Just wondering is there any way to conclusively confirm my diagnosis from the original log? I have video of the subsequent flights, and black box logs if anyone has interest in them.

Strangedave, 2018-10-15:
I'm attaching a short test flight log here, and a link to the video from that flight. I think they confirm problem solved.
[BTFL_BLACKBOX_LOG_WIN_5R_20181015_182108.zip](https://github.com/betaflight/betaflight/files/2480891/BTFL_BLACKBOX_LOG_WIN_5R_20181015_182108.zip)

Video is here;
https://youtu.be/d2uB7PA15Ms

closed 2018-10-15 by Strangedave

Strangedave, 2018-10-17:
I was wrong, new fc installed today problem is still there....

etracer65, 2018-10-17:
Gyro twitches (particularly on the yaw axis) are often indicative of electrical noise. The ICM gyros tend to be more affected.  Adding low ESR capacitors may help.

Strangedave, 2018-10-18:
I have one on there. I’ll switch it out and see if that helps. 

Dave Goddard
Garry Real Estate
Office (630)310-8315
Cell (630)992-3283
Email dave@dave-goddard.com
Web www.4ure.com

Strangedave, 2018-10-18:
Old cap replaced with new cap. The problem is still there. only thing left to replace is the motors and DALRC rocket esc. This video shows the issue see 1:15 as im going around the tree. You can hear the issue occur throughout the video. https://youtu.be/8WWUZgiSccc . The crazy thing is I built 2 of the exact same thing. one runs, the other not so much. Checked the motors on betaflight for vibration, they are roughly half of the vibration of the working rig. I'm going to post this video to the betaflight facebook page https://youtu.be/ZcrI0yCEw2M I'm out of ideas. Thanks for your help so far.

Strangedave, 2018-10-19:
Giving up, Flashing 3.5.2 to default and I'll tune the two rigs independently. I'm pretty sure I can make it run ok. if not; its gonna be an MPU6000 rig and I know that works ok.

Strangedave, 2018-10-30:
Now it has a new set of motors. Still twitching. https://youtu.be/rCFJWtGcPTQ
and slomo https://youtu.be/94eQXlxDq8I.

etracer65, 2018-10-30:
This is really not the place to continue this discussion as it's closed and your issue isn't related to Runaway Takeoff Prevention at all. If you think there's a specific problem with your flight controller target it's probably best to open a new issue or continue discussion in and existing one if exists. That being said Betaflight issues are for discussing firmware issues and not specific hardware problems users may be having. If you feel it's a problem specific to a target then it's appropriate to discuss in its own issue.

Strangedave, 2018-11-26:
This problem was caused by a defective ICM20602 gyro. The problem was solved by replacing the Flight Controller.