betaflight/betaflight #6894: Crossfire telemetry in 50Hz mode

slackista, 2018-10-06:
Telemetry refresh rate is very low. and not consistent in 50Hz mode. Sometimes is 1/5 Hz and sometimes > 1/10Hz causing 'Sensor lost' in Taranis. All other sensors belongs to strict crossfire radio, update very fast.

In 150Hz mode BF telemetry update is relatively slow, around 1Hz.  

Steps to reproduce the behaviour:
Force Crossfire receiver to operate in 50Hz mode and wait couple of seconds.
Results: 'Sensor lost' warning in Taranis.

BF config:
```
# version
# Betaflight / MATEKF405 (MKF4) 4.0.0 Oct  6 2018 / 22:17:26 (a5a0f9aa7) MSP API: 1.41

board_name MATEKF405
manufacturer_id 

# name

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_RX 2 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1400 2100 0 0
aux 1 1 3 1700 2100 0 0
aux 2 2 3 1325 1600 0 0
aux 3 13 4 1750 2100 0 0
aux 4 26 0 1775 2100 0 0
aux 5 28 2 1600 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000

# master
set gyro_lowpass_hz = 120
set acc_calibration = -59,-30,-80
set mag_hardware = NONE
set min_check = 1100
set max_check = 1990
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set align_board_yaw = 270
set vbat_max_cell_voltage = 42
set vbat_min_cell_voltage = 34
set vbat_scale = 111
set serial_update_rate_hz = 300
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 44
set osd_cap_alarm = 1400
set osd_alt_alarm = 200
set osd_vbat_pos = 2081
set osd_rssi_pos = 2094
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2497
set osd_mah_drawn_pos = 2113
set osd_altitude_pos = 2518
set osd_disarmed_pos = 2347
set osd_stat_max_spd = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 1
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5805
set vcd_video_system = PAL

# profile
profile 0

set feedforward_transition = 20
set iterm_relax_cutoff = 12

# rateprofile
rateprofile 0

set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 105
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 15
set tpa_breakpoint = 1050
```

My setup:
Taranis x9D plus SE, OpenTX v2.2.2
TBS Crossfire Micro TX v2.41
TBS Crossfire Nano RX v2.41
Matek F405 CTR, Betaflight v3.5.1

Problem still persists when:
Downgrade BF to 3.3
Downgrade Crossfire fv to v2.40 and v2.24

etracer65, 2018-10-07:
Sorry, on mobile and accidentally closed.

I believe this has already been reported to TBS.  Don’t think it’s a Betaflight issue as we have no visibility into whether Crossfire is in 50hz mode.

lostcontrol, 2018-10-15:
I initially opened a ticket against OpenTX to be able to set a different timeout per sensor. I then reported the issue to TBS. The ticket got closed, reopened, closed, etc, etc.

Last comment I got from them was:

> because some of the sensors are low in priority to keep the sensors that matter functioning longer. if your LQ is fine, and your SNR is fine then you shouldnt have to worry,

I'm now somewhat used to those occasional "sensor lost" alarms...

https://github.com/opentx/opentx/issues/6132
http://team-blacksheep.freshdesk.com/support/tickets/741290

slackista, 2018-10-15:
Sensors refresh rate over 1 second is even to high for proper use of alarms. If my battery voltage drops and I will have alarm after 15 seconds it is totally useless.

I also opened ticket in tbs support desk. For two weeks, no answer, except questions like someone new in that particular issue.
https://team-blacksheep.freshdesk.com/support/tickets/806370
To be honest, after spending hundreds of $ for crossfire system and unable to use BF telemetry I am very disappointed.

lostcontrol, 2018-10-16:
@slackista same here. First answer was quick, just a bunch of random questions not really related to the topic. Then just the usual "in 50Hz mode, telemetry bandwidth is lower, some sensors have low prio, etc, etc" answer over and over. I provided a video showing the issue and also some telemetry logs where you see that sometimes those sensors have a very low refresh rate.

I have several friends that fly XF on OpenTX with KISS FC and they never complained about this issue. That's why it might be still related to Betaflight somehow.

I don't know but is Betaflight sending more data over telemetry than KISS for example?

mikeller, 2018-10-16:
@lostcontrol: From your description it looks like the problem that causes this is that the bandwidth that is available when the link is running in 50 Hz simply is not enough to transfer data for all of the sensors that Betaflight supports within the timeframe that OpenTx allows before it starts generating 'sensor lost' alerts. The only way that Betaflight could potentially 'fix' this would be to stop sending some of the telemetry values, in order to enable faster updates on the remaining ones. This is obviously not a good solution, and it would be preferable for the user if they still got data for all of the sensors, and OpenTx addressed this by allowing higher timeouts before it is generating alerts.

lostcontrol, 2018-10-16:
@mikeller well, as pointed out by an OpenTX dev and as @slackista said before, having a sensor value refreshed less than once every 10 seconds is more or less of no use, especially for battery voltage.

Looking at the code, the telemetry data are sent at 10Hz. There are 4 "groups" (attitude, battery, flight mode and GPS) so that is ~2.5Hz for a specific sensor I guess. Would having a different schedule like battery data sent out twice as much as the rest for example change anything? Do you know how Crossfire/CRSF handles low telemetry bandwidth situations? Like a FIFO and it drops anything newer?

If I find time, I'll try to only keep the battery data in the telemetry and give it a try just to see if this would improve things.

Still, I also wonder if that super low telemetry bandwidth in 50Hz mode is normal. I get those at spots where my FrSky link would not even trigger a "RSSI low" warning.

mikeller, 2018-10-17:
@lostcontrol: Good point, there is a big discrepancy between 0.4 s delay between frames for one sensor type and 10 s packet timeout. I do not know anything about the internals of the crossfire hardware, so I can only make somewhat educated guesses about the reason for this discrepancy:

1. potentially the buffer for the telemetry back channel on the RX is implemented in a bad way, and this causes packets in the buffer to be lost or even corrupted when there is back pressure from a slow link;

2. or it could be that the data for one telemetry sensor value has to be split over multiple frames, and the error correction on the TX is not able to cope with frame losses, and has to discard entire sensor values when individual frames are lost. This would mean that as soon as the frame loss rate is non-negligible the probability for complete telemetry frames being received will drop to almost 0. (FrSky X has this problem to some extent.)

Good idea to test with only sending one sensor value (battery or otherwise does not matter for the test).

We could also test what happens if we up the telemetry frame rate to 20 or 40 Hz. If the problem is 2., and the buffer on the RX is sound, this might improve things by making sure the buffer is always full and no frames are sent back to the TX without any telemetry data.

Another thing that might (or might not) be useful is logging on the TX. I think OpenTx can log all telemetry frames, and this would shed some light on what is actually received.