betaflight/betaflight #6896: Add WHITE when changing LED colors via aux channel

FieserKiller, 2018-10-07:
**Is your feature request related to a problem? Please describe.**
I need the following LED colors on a knob for FAI race in shenzhen using the betaflight led control feature:
Red Blue Yellow Green White Purple 
I followed the wiki and everything works straightforward except white. 
Betaflight rotates nicely the Hue in HSV and I get all colors except white, because for white it needs to set Saturation to 255 which it does not. 

**Describe the solution you'd like**
I'd love to be able to set up betaflight in a way that it rotates the colors through colors I define via the color command. The current hue rotation looks cool at all but is of no real use for racer. The event managers define a bunch of colors to use and thats exactly the colors I'd like to be able to set up and rotate through without betaflight interpolating hue between the main colors.

**Describe alternatives you've considered**
A quick and dirty solution I'll use if this feature request is rejected or not finished in time is to modify code so that LEDLOW mode does not set the LED strip to HSV(BLACK) but to HSV(WHITE) and use this mode when white is needed.
This hack could be generalized by making the LEDLOW color configurable via mode_color command.

jflyper, 2018-10-07:
@FieserKiller As I saw your FB comment this evening, I came to remember that I've removed the color selection feature from CMS, leaving it for future work to put it back in.

I have an experimental code that reinstate the feature.

- There will be a "RACE COLOR" menu entry under the "LED STRIP" feature menu.
- When _ALL_ of the LEDs with function being set to "Color" have the same color set, you can select the color with the "RACE COLOR" menu entry.
- If not all colors are the same, then the "RACE COLOR" menu entry is "DISABLED".

What do you think? I can provide a test build. Which FC/BF version are you using?

jflyper, 2018-10-07:
[betaflight_3.5.1_KAKUTEF4V2-racecolor-00.hex.zip](https://github.com/betaflight/betaflight/files/2453857/betaflight_3.5.1_KAKUTEF4V2-racecolor-00.hex.zip)

FieserKiller, 2018-10-07:
I've to admit I don't understand your instruction.
I flashed the image and restored my back up settings.
Selected all leds, selected function color and chose white as default color. 
But here I'm stuck what to do next?

[image]

jflyper, 2018-10-07:
Open CMS (the OSD menu) and navigate to FEATURES->LED STRIP.

If you have these in the same color (I presume GREEN), then you will see "RACE COLOR       GREEN" menu entry. Use left or right to change the color.

mikeller, 2018-10-07:
@FieserKiller: I don't know what exactly your problem is, by reading what you want to do and reading how colours can be changed through an AUX channel (https://github.com/betaflight/betaflight/wiki/LED-Strip-functionality#how-to-set-up-throttle-functionality-with-leds) this is already possible:
- set the LEDs to 'Color', choose what end colour you want to use (don't use 0 or 15);
- set the colour one below the chosen colour to the colour farthest away from the chosen colour;
- set the colour one above the chosen colour to white;
- set the 'throttle' modifier to the AUX channel you want to use;

As a result, the first half of the chosen AUX channel will cycle through all of the colours, the second half will fade to white.

Sample config with colour 14 chosen as pre-set colour:

```
# led
led 0 3,0::CT:14
led 1 4,0::CT:14
led 2 5,0::CT:14
led 3 6,0::CT:14
led 4 7,0::CT:14
led 5 8,0::CT:14
led 6 9,0::CT:14
led 7 10,0::CT:14
led 8 11,0::CT:14
led 9 12,0::CT:14

# color
color 13 0,0,255
color 14 359,0,255
color 15 0,255,255

# mode_color
mode_color 7 0 6
```

jflyper, 2018-10-07:
@mikeller Fantastico!!

FieserKiller, 2018-10-07:
@jflyper ach I didn't think of using the osd :D But it works beautifully! Not on the knob but I get all colors i need and they are changeable fast enough for sure!

FieserKiller, 2018-10-07:
@mikeller's solution works as advertised as well :)
only thing is no there is even more colors more tightly mixed into half of the channel but thats pure nitpicking, mein point is: it works! 

But I love and will go with @jflypers solution, because here i have clearly named distinct colors to chose from. 
One remark: I guess you use the color names from enum colorId_e (line 99 ledstrip.c)? Then I suggest to use the HSV definitions from const hsvColor_t below because this way name and color will be in sync as well. Currently the user defined colors are used where name and color fit in default configuration but don't  anymore when the user changed colors.
How are the chances to get this race color feature into a release? For now I can flash your custom uild to my quads but betaflight will progress for sure :)