betaflight/betaflight #6897: 3.5.1 BlackBox log numbering scheme

0crap, 2018-10-07:
I have configured my TX so that if I flick a switch it will activate BlackBox logging.
(Using Modes -> BLACKBOX with Aux4 in the Betaflight Configurator.)

What I expect is that when I activate logging it will make a file LOG00001.BFL and the OSD Post Flight Statistics shows log number 1 is created. Then 2 , 3 and so on.

However this is not always the case.
When the logging switch is disabled and I arm->disarm->arm->disarm then enable the logging switch, arm->disarm, a log file named LOG00003.BFL is created.

In other words, with the logging switch disabled, it does seem to keep counting log numbers under the hood. 

Target: MATEKF405 (MKF4)

etracer65, 2018-10-07:
Every arm creates a blackbox log header even if no other information is logged. This is normal and by design and has been this way forever.

0crap, 2018-10-07:
OK, Thx for the comment.

closed 2018-10-07 by 0crap