betaflight/betaflight #6902: Continue Lipo Feature for Current Draw

tomstom, 2018-10-08:
If I compare BF with Kiss, one big feature that I'm missing is to continue a Lipo after a reboot of the flight controller. For example you crashed and have to swap a broken propeller with a new one. So I do this only when the copter is off... But after this, it would be nice If the OSD starts at the last value for current draw of the last Lipo...

**Describe the solution you'd like**
In Kiss you get a question for some seconds after booting, if you want to contiune the last lipo. You have to confirm this with a stick input, for example yaw left.

timman2er, 2018-10-08:
I don't like the stick input method. We could detect if it's the same battery based on it's voltage, and continue current draw if voltage is within a small range of previous known voltage. 

This'd be activated in CLI with: `lipo_current__draw_continue` .

ledvinap, 2018-10-08:
@timman2er : Voltage method won't work well - there is noticeable difference under load and even without load it takes some time until it reaches stable value.

mikeller, 2018-10-08:
@ledvinap: Agreed re 'auto - detect' not being an option - particularly older batteries tend to 'refresh' quite a bit within seconds to minutes after the load is taken off, resulting in a significant voltage increase.

Do you think we'd need wear levelling on the flash used to store the last battery usage, considering this will have to be written once a second or so to yield reasonable accuracy. Or is there a way around having to persist the latest battery usage level into flash that I am overlooking?

closed 2018-11-14 by stale[bot]