betaflight/betaflight #6908: FC freeze during fly

roki97, 2018-10-08:
Hi to all.

I have this configuration:
Omnibus F4 SD with OSD and Vtx Integrated
35A ESC with Dshot1200
Sbus Receiver

This is what happens:
I fly for 20 seconds, sometimes 1 minute, or 40 sec and when I flip on roll axis it do like a desync and osd freeze, no more timer, no lipo voltage, it is completely freeze, but video still works.
I tried to fix this problem by changing motors and esc (from dshot1200 to dshot300), then I changed Fw, from 3.5 to 3.3, I have different quads but no one give me a problem like this. (maybe too low stick min? Nope I checked it, it go from 1002 to 1998)
One time I inserted sd on the slot of fc and I recorded blackbox. When I flip and the problem happens the blackbox stop recording without any info (no disarmed, no strange data).

Now my quad is open and waiting to be fixed but in the next days I can build it and record a DVR to show the problems.

I didn't found any problem like these so I open this post.

Sorry for my bad english.

Hope you can help me, thank you.

Romeo

My config:
ESC Quadrant 35A dshot1200
Amax Inno 2550kV 2306 Motors
XM+ Receiver
F4 OSD and 800mW Vtx by Rchalarge

jflyper, 2018-10-08:
@roki97 
- You are not clear about the version you have trouble with. Does it work on any of the versions? Or, it doesn't work at all?
- You need to attach CLI `diff` from non-working version.
- How are you powering your flight controller? 800mW VTX onboard freaks me out.

roki97, 2018-10-08:
@jflyper 
Thak you for your interesting.
I tried different release but none work, all the release (stable) give me same problem.
I use only 25mW VTX power, not 800mW, the temperature on fc is around 50-55 degree, and I power it from ESC's BEC, 5V. 

[diff.docx](https://github.com/betaflight/betaflight/files/2457641/diff.docx)
[diff.txt](https://github.com/betaflight/betaflight/files/2457642/diff.txt)

etracer65, 2018-10-08:
The only time I've seen something like this was when I had a damaged ESC that had lost some capacitors (prop strike). Every time I would do a roll the FC would crash/freeze.  Turned out it was because of the massive voltage spike coming from the ESC under active braking (when the roll was stopping).  Replacing the ESC solved the problem.

So while your problem may or may not a be a damaged ESC, the cause is probably the same - massive voltage spike affecting the electronics.  Assuming nothing is damaged you can try adding capacitors and that may help.

etracer65, 2018-10-09:
Closing this since it's not a firmware issue.  Since the problem just recently started and you still have the problem with all versions of the firmware it's clearly a hardware problem. In addition to the suggestions above, try more appropriate areas for support like the Betaflight slack, the RC Groups discussion thread, or the Betaflight official Facebook group.

https://slack.betaflight.com/
https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread
https://www.facebook.com/groups/betaflightgroup/

closed 2018-10-09 by etracer65