betaflight/betaflight #6914: F7: Possible racing in CDC_Send_DATA

jflyper, 2018-10-09:
There seems to be some problem with `CDC_Send_DATA` in `vcp_hal/usbd_cdc_interface.c`.

`CDC_Send_DATA` blocks indefinitely at some point when it is repeatedly called back to back.
Possible racing condition on ring buffer management against asynchronous transmitter (ring consumer).

- To reproduce, build a F7 firmware with `APP_TX_DATA_SIZE` in `vcp_hal/usbd_cdc_interface.c` being set to some small value, say `512` then issue CLI `set` command.

- It does not develop with debug build.

- Here's a where it is executing when `set` is wedged.
```
Reading symbols from obj/main/betaflight_MINIMALF7X2.elf...(no debugging symbols found)...done.
(gdb) y
Undefined command: "y".  Try "help".
(gdb) load
Loading section .isr_vector, size 0x1e0 lma 0x8000000
Loading section .pg_registry, size 0x2f8 lma 0x80001e0
Loading section .pg_resetdata, size 0xa0 lma 0x80004d8
Loading section .data, size 0x4d0 lma 0x8000578
Loading section .fastram_data, size 0x4 lma 0x8000a48
Loading section .text, size 0x196a8 lma 0x8008000
Loading section .tcm_code, size 0x1e70 lma 0x80216a8
Start address 0x801c144, load size 114532
Transfer rate: 18 KB/sec, 6737 bytes/write.
(gdb) cont
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x080103c0 in delay.constprop ()
(gdb) where
#0  0x080103c0 in delay.constprop ()
#1  0x080106b8 in CDC_Send_DATA ()
#2  0x0801075e in usbVcpWriteBuf ()
#3  0x0800a17e in bufWriterFlush ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

ledvinap, 2018-10-10:
@jflyper : ad debug build - gcc sometimes gets too clever and breaks register access (2x halfword instead of 1x word when it can infer even address). Enabling different optimization levels on per-function basis did help to pinpoint it last time ...

ledvinap, 2018-10-10:
UserTxBufPtrIn / UserTxBufPtrOut muse be declared `volatile` (or some memory barrier used for access)

ledvinap, 2018-10-10:
https://github.com/betaflight/betaflight/blob/2f1f07d186cd4023a4b7486e6fcf9f6237e5b847/src/main/vcp_hal/usbd_cdc_interface.c#L389 - first byte is written even when buffer is full (effectively discarding all data)

ledvinap, 2018-10-10:
https://github.com/betaflight/betaflight/blob/2f1f07d186cd4023a4b7486e6fcf9f6237e5b847/src/main/vcp_hal/usbd_cdc_interface.c#L383
This test makes little sense - buffer space should be checked, not USB request pending ...

ledvinap, 2018-10-10:
rxAvailable and rxBuffPtr needs `volatile` too ...

MJ666, 2019-01-26:
For me the problem is also present in debug builds. I can see if i switch to CLI mode i get some garbage at the beginning. Setting the variables to volatile and move order of the buffer full test will fix the garbage problem but not the SPI flash CRC error issue from #7448.

jflyper, 2019-02-08:
I did some reading.

- Low level driver is called from interval timer using TIM7.
- At least the transmitter side is using DMA using USB core embedded DMA controller, and `TxState` seems to indicate if the DMA is in progress or not. Therefore, waiting for `TxState` to become zero

tavdog, 2022-05-26:
was this fixed ?

haslinghuis, 2022-05-26:
> It does not develop with debug build.

Looks like a compiler optimization issue.

haslinghuis, 2022-05-26:
Anyway configurator 10.8 does filter out most of the garbage now. Closing now.

closed 2022-05-26 by haslinghuis