betaflight/betaflight #6915: betaflight LED not blinking on LVC but turns red only

simtcrom, 2018-10-11:
**Describe the bug**
As per betaflight github [documentation](https://github.com/martinbudden/betaflight/blob/master/docs/LedStrip.md) (Modes->Warning) 

"Low Battery	-- flash red and off -- battery monitoring must be enabled. May trigger temporarily under high-throttle due to voltage drop"

So isn't it supposed to blink when reaches configured LVC?

**To Reproduce**
Set one of the LED to Warning, When reaches configured LVC, LED not blinking but turns red only. Buzzer is not wired.

**Expected behavior**
Low Battery	-- flash red and off.

**Flight controller configuration**

> 
$M>ni
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / OMNIBUSF4FW (OBFW) 3.5.1 Sep  8 2018 / 05:39:57 (d9fb5ca13) MSP API: 1.40

board_name OMNIBUSF4FW
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature LED_STRIP

# beeper

# beacon

# map

# serial
serial 5 1 115200 57600 0 115200

# led
led 0 8,15::CW:2
led 1 7,15::L:0

# color

# mode_color

# aux
aux 0 0 2 1675 2100 0 0
aux 1 1 3 900 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW270
set gyro_to_use = SECOND
set align_acc = CW270
set acc_calibration = 62,91,-164
set align_mag = CW270
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT300

# profile
profile 0

# rateprofile
rateprofile 0
> 

**Setup / Versions**
 - Airbot Omnibus F4 Nano V6
 - HGLRC Super Mini 1.38g WS2812.

mikeller, 2018-10-11:
@simtcrom:

```
led 0 8,15::CW:2
```

This LED has warnings enabled (`W`), but the colour is set to red already, so you won't be able to see when it is blinking red.

```
led 1 7,15::L:0
```

This LED is set to show the battery **level** (`L`), but no warnings.

(Also, note that your source for the documentation is outdated, the current documentation is at
https://github.com/betaflight/betaflight/blob/master/docs/LedStrip.md)

mikeller, 2018-10-11:
I am not suggesting anything, just pointing out that a red blinking warning overlaid on a constant red LED will not be visible. ;-)

But yes, having the warning and the battery level on the same LED is one way to make the blinking visible.

simtcrom, 2018-10-13:
> I am not suggesting anything, just pointing out that a red blinking warning overlaid on a constant red LED will not be visible. ;-)

I set color and warning on a LED and it blinks. It blinks when battery is low till then no light,

mikeller, 2018-10-14:
@simtcrom: Yes, good point, they cannot, I overlooked that. So you'll have to use either the battery indicator, or the battery warning, on any given LED.

closed 2018-11-20 by stale[bot]