betaflight/betaflight #6922: Getting error 12 Calibration not complete

mchaos, 2018-10-13:
I keep getting error 12. I let it sit and I still get error 12. If I try unplugging the batt, and plut it back in, eventually it will arm.

I have Ublock GPS installed. Does it consider waiting for a fix part of calibration?

I have a servo attached to a single axis gimbal and I have a switch set up for auto stabilize which keeps the cam level, when I am having the error 12, it is really sluggish to catch up. When there is no error 12, it moves quick. 

I have tried disabling GPS and it still does it.

any idea's?

etracer65, 2018-10-13:
Unless you provide the information requested in the issue reporting instructions, nobody can help you. In particular, what Betaflight version? What flight controller? A `diff` of your settings.

mchaos, 2018-10-13:
Betaflight / FPVM_BETAFLIGHTF7 (FBF7) 3.5.1 Sep 8 2018 / 05:18:18 (d9fb5ca
<https://github.com/betaflight/betaflight/commit/d9fb5ca131af9f03c69cd69a6e5db05931d38827>)
MSP API: 1.40

board_name FPVM_BETAFLIGHTF7
manufacturer_id
name

name EPIC
resources

resource MOTOR 1 E09
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 B01
resource SERVO 1 A03
mixerservo

servo 0 1050 1830 1500 100 4
servo mixfeature

feature SERVO_TILT
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature AIRMODE
beeper

beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -DISARM_REPEAT
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
beacon

beacon RX_LOST
beacon RX_SET
map

map TAER1234
serial

serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
ledcolormode_coloraux

aux 0 0 1 1700 2100 0 0
aux 1 2 2 1300 1700 0 0
aux 2 8 4 1300 1700 0 0
aux 3 13 4 1700 2100 0 0
aux 4 26 6 1700 2100 0 0
aux 5 35 2 1700 2100 0 0
aux 6 37 3 1300 1700 0 0
adjrangerxrangevtxrxfailmaster

set gyro_use_32khz = ON
set gyro_to_use = SECOND
set acc_calibration = 234,-62,-45
set mag_hardware = NONE
set baro_hardware = NONE
set rc_interp_ch = RPT
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set align_board_yaw = -90
set bat_capacity = 1300
set beeper_dshot_beacon_tone = 5
set small_angle = 35
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 35
set gps_rescue_min_sats = 0
set pid_process_denom = 2
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1300
set osd_vbat_pos = 2081
set osd_rssi_pos = 2107
set osd_tim_2_pos = 2369
set osd_flymode_pos = 2392
set osd_ah_pos = 2126
set osd_gps_speed_pos = 2090
set osd_gps_lon_pos = 2432
set osd_gps_lat_pos = 2449
set osd_home_dir_pos = 2318
set osd_home_dist_pos = 2349
set osd_compass_bar_pos = 2410
set osd_altitude_pos = 2097
set osd_battery_usage_pos = 2057
set cpu_overclock = 240MHZ
set vcd_video_system = NTSC
profile

profile 0

set dterm_notch_cutoff = 0
set anti_gravity_threshold = 350
set f_yaw = 0
rateprofile

rateprofile 0

set roll_rc_rate = 180
set pitch_rc_rate = 180
set yaw_rc_rate = 180
set roll_expo = 32
set pitch_expo = 32
set yaw_expo = 32

etracer65, 2018-10-13:
You're apparently trying to use the noisier ICM gyro in 32KHz mode. Try increasing `gyro_calib_noise_limit` until the gyro calibrates reliably.  If it only does this while powered via the battery (using the on board regulators) and has less problem when powered by USB then the on board regulators might be providing noisy power and extra capacitors might be needed.

mchaos, 2018-10-13:
I sincerely appreciate your help. I have a feeling that will do it.

mchaos, 2018-10-13:
Something else I noticed... When I plug into my PC with BFConif, there is a lag for the on screen model depicting the angle for a moment... I haven't seen that before.

mchaos, 2018-10-13:
FYI... Raising the limit fixed the issue. Thanks..

closed 2018-10-13 by etracer65