betaflight/betaflight #6926: BF 3.5.1 - BF Lua Script VTX page not working and OSD setup screen impossible to access

thomasekdahlN, 2018-10-13:
Hobbywing F4 FC and 4 in 1 40A stack
Matex HV VTX - IRC Tramp on UART6
Fport
X4RSB receiver

After upgrading to BF 3.5.1 (From 3.5.0) the following stopped working

- It is not possible to change any of the settings in the BF Lua VTX Page (Pressing enter does not make the information editable). The same LUA script on BF 3.5.0 works fine, and it worked on this model prior to upgrading. It is not possible to configure the VTX from the buttons. It is possible to set the VTX values from the CLI (so then it seems that the wiring is OK)

- Its not possible to access the OSD configuration screen at all. It only flashes and no menues are shown. Again, this worked perfectly before upgrading from 3.5.0 to 3.5.1

I'm open for that this can be a configuration fault, but I can't see how to fix it?

JVFPV22, 2018-10-13:
Im having the same issue. 
TBS Unify Pro v2 5v smart audio on 3.5.0 was working fine. Updated to 3.5.1 and when I "Mid Throttle-yaw left-pitch up", my fc beeps twice and no menus pop up on the osd as if the stick commands are being denied.

mikeller, 2018-10-14:
@thomasekdahlN: This is an issue with the Betaflight TX lua scripts. It should really go into https://github.com/betaflight/betaflight-tx-lua-scripts/issues.

closed 2018-10-14 by mikeller