betaflight/betaflight #6929: Increase "Warning Cell Voltage" field to 2 decimals.

simtcrom, 2018-10-14:
**Is your feature request related to a problem? Please describe.**
I am running BF 3.5.1 on a micro quad with a 800mah 2s lipo.
Currently "Warning Cell Voltage" is set to 3.5. When balance charge around 500mah is being loaded.
I guess ideal is 80% of total capacity, which mean I could use further more juice from the lipo.
But If I set "Warning Cell Voltage" to 3.4, then its too low.

**Describe the solution you'd like**
Increase "Warning Cell Voltage" field to 2 decimals. Then I can set 3.45v.

**Additional context**
Currently If I set to 3.45v, it becomes 3.5v.

mikeller, 2018-10-14:
@simtcrom: Changing the scale of `vbat_warning_cell_voltage` would be quite involved, since this is currently an 8 bit value, and as such would only go up to 2.55V if the scale was 0.01V. Instead, try setting `vbat_hysteresis = 1` in CLI and see if this makes it work for you.

simtcrom, 2018-10-14:
> @simtcrom: Changing the scale of `vbat_warning_cell_voltage` would be quite involved, since this is currently an 8 bit value, and as such would only go up to 2.55V if the scale was 0.01V. Instead, try setting `vbat_hysteresis = 1` in CLI and see if this makes it work for you.

its already set to 1 (default)
# get vbat_hysteresis
vbat_hysteresis = 1

I actually did not understand how vbat_hysteresis works. How can I use that to get warning voltage value increased by 0.05?

mikeller, 2018-10-14:
Hysteresis is the amount of sag below the warning voltage that is allowed before the warning kicks in. It is also in 0.1 V steps, but increasing it will result in the warning kicking in later. Try 2 then.

fernandodlc, 2018-11-19:
With higher voltage setups this is becoming more of a requirement. I appreciate the difficulty with changing the actual value but perhaps there would be a way to add an option tie it to an absolute voltage rather than a cell voltage.

fgiudice98, 2018-12-03:
If instead we use a voltage resolution of 0.02V the 8bit configuration value can still be used. If the value is multiplied by 2 we have 0V to 5.10V with 0.02V steps with no much effort. With 5.10V we have room even for HV Lipos max voltage. The only problem is with the firmware update.

mikeller, 2018-12-03:
@fgiudice98: Not sure if the overall effort / disruption to change to a 0.02V resolution will be better or worse than changing to 0.01V and 16 bit:
- users will still have to change their settings;
- the setting displayed in CLI will become totally unintuitive, which will probably lead to a lot of misconfigurations.

fgiudice98, 2018-12-03:
The settings can be changed automatically: if I have 35 as warning value with the new system it will be a warning value of 0.7V, so it can be changed to 175 automatically, since I don't think that values below 100 will be used.

mikeller, 2018-12-03:
@fgiudice98: How would these settings that are stored in the user's `diff all` output file be 'updated automatically' when the user updates the firmware?

fernandodlc, 2018-12-03:
It may be too late for a change like that now. However given the amount of configs that can break already when upgrading a major version, resetting this to a default value when moving from 3.6 ->4 would not be unreasonable but it might be a step too far for a minor one.

fgiudice98, 2018-12-03:
When the values are read from EEPROM they can be checked and changed automatically. If a user want to use the old values and the old way to set (*0.1V) there is no problem because they will be scaled automatically, if another want to use the new he can be set in the cli with the new way.

mikeller, 2018-12-03:
@fgiudice98: How would the firmware, when reading the values from EEPROM, know if the values have already been changed or not, and know what way the user wanted their value stored in CLI?
If you think that this is workable then feel free to open a pull request and we can discuss there.

fgiudice98, 2018-12-03:
If EEPROMvalue<100 -> old value type -> read as value*5 (35*0.1=35*5*0.02=175*0.02=3.5V)
If EEPROMvalue>100 -> new value type -> read as it is
This will be in /src/main/sensors/battery.c
I can even change in the cli using the old type, I will never see the difference.
In the next days I will work on a pull and test it

mikeller, 2018-12-03:
So there will be two possible ways to encode every voltage? This will be super confusing for users, so I don't think this should be added. If we really want to change this, we should rather go for a clean solution, even if this means **a lot** more work.

fgiudice98, 2018-12-03:
I don't thik it will be confusing. If I use the configurator in a normal way, I can't see the difference; if I use the cli and set the value as it is now, I can't see the difference; if I want more control over voltages I can set the new value in cli or use the configurator normally. This will ensure backward compatibility, so no problem in the dump/diff or problems with users not understanding the new type of value.
I think this solution is not a lot of work and will ensure backward compatibility, but if it's more reasonable a 16bit value ok.

mikeller, 2018-12-03:
And if you first use the CLI to set the value to 35, and then you use the configurator to set it to 3.4V, and suddenly you see 170 in CLI, how do you think that this is **not** confusing.

Feel free to open a pull request for configurator and firmware to change to a 16 bit value.

fernandodlc, 2018-12-03:
I'm not sure what the code complexity implications are but I can't find any cells that would use a greater than 100 value. From a user perspective either redefaulting voltage during a version change or having a "hidden" setting (for there are many already) in the cli to allow greater granularity would both be acceptable imo.

Are there any wider plans to move to 16bit values? If there is planning to make this happen more generally it might make this fix more sensible.

fgiudice98, 2018-12-03:
@fernandodlc The unused values of the old type will become the range of the new value type and vice versa.
As @mikeller is saying, the problem is for users seeing a "strange" value in the cli. Personally if I see a strange value I first read the doc/wiki. I see more peoples getting confused by drop of features in firmware update than adding a function in cli. For example, I use the "AND logic" in the aux settings in cli, this is not present in the wiki but present in the diff and implemented (it works really well for me). I didn't see nobody confused by this.

closed 2019-01-09 by stale[bot]