betaflight/betaflight #6932: 3.5.1 RSSI value issue with spektrum SPM4648

alexglvr, 2018-10-14:
**Describe the bug**
RSSI value is always 40 with spektrum SPM4648 receiver. It was working under betaflight 3.3.0.
FC = CLRACINGF4 (CLR4)

**To Reproduce**
Upgrade FC to 3.5.1

**Expected behavior**
Having the correct RSSI value

**Flight controller configuration**
```
``` # diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.5.1 Sep  8 2018 / 05:03:13 (d9fb5ca13) MSP API: 1.40

board_name CLRACINGF4
manufacturer_id 

# name
name Rooster A

# resources

# mixer

# servo

# servo mix

# feature
feature -ANTI_GRAVITY
feature MOTOR_STOP

# beeper

# beacon

# map
map TAER1234

# serial
serial 3 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 4 900 1500 0 0
aux 1 1 2 1750 2100 0 0
aux 2 2 2 1300 1700 0 0
aux 3 4 3 900 1350 0 0
aux 4 28 0 900 1500 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# rxrange
rxrange 0 1155 1845
rxrange 1 1155 1845
rxrange 2 1155 1845
rxrange 3 1155 1845

# vtx

# rxfail

# master
set gyro_lowpass_hz = 90
set min_check = 1100
set rssi_channel = 12
set rssi_scale = 41
set rc_interp_ch = RP
set fpv_mix_degrees = 30
set serialrx_provider = SPEK2048
set spektrum_sat_bind_autoreset = OFF
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1500
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 1
set tlm_halfduplex = OFF
set ibus_sensor = 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
set osd_vbat_pos = 2082
set osd_rssi_pos = 2457
set osd_tim_1_pos = 54
set osd_tim_2_pos = 470
set osd_flymode_pos = 2381
set osd_anti_gravity_pos = 2106
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 2253
set osd_ah_pos = 2126
set osd_current_pos = 385
set osd_mah_drawn_pos = 439
set osd_craft_name_pos = 2433
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2088
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5732
set camera_control_key_delay = 150
set camera_control_internal_resistance = 99

# profile
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_threshold = 350
set anti_gravity_gain = 1000
set iterm_windup = 50
set throttle_boost = 0
set p_pitch = 47
set i_pitch = 35
set d_pitch = 18
set p_roll = 40
set i_roll = 35
set d_roll = 18
set p_yaw = 63
set i_yaw = 39
set d_yaw = 20
set f_yaw = 0

# rateprofile
rateprofile 2

set roll_srate = 92
set pitch_srate = 92
set yaw_srate = 92```
```

**Setup / Versions**
 - Flight controller : CLRACINGF4 (Armattan)
 - Other components : SPM4648, TRAMP HV v2

**Additional context**

Thank you for your help

AndersHoglund, 2018-10-14:
You have this:
`set rssi_scale = 41`
Change to:
`set rssi_scale = 100`

jflyper, 2018-10-15:
lol

alexglvr, 2018-10-15:
Thank you for your answer. Effectively, it was simple...

In fact I did not see it because I am having a bigger issue around FailSafe.

I have 2 quads with the exact same config, especially SPM4648 receiver. I use a DX9 emitter from spektrum.

On BOTH quads, when I reach a distance around 70m, they go into failsafe. I then thought it was due to the emitter. Therefore I tried with another emitter from a friend (dx18), and the same results: Failsafe !
Is there a possibility to use the blackbox to find what is going on exactly?
Can I have something wrong in my betaflight configuration that can explain that?
Thank you again for your help,

Alex

jflyper, 2018-10-15:
@alexglvr Are you a stick armer?