betaflight/betaflight #6954: "No pid at min throttle" on a switch

HrrrPippi, 2018-10-18:
Hi, this is my first time, taking part in any of the development of this great firmware. 

Recently Mr Steel, and later wild willy posted videos showing how to have the quad a bit "safer" when first armed. 

They do not have the pid controller active, when first armed, and that way the motors and props have far less power active.

I wanted to try this, and used the "pid_at_min_throttle = off", but it seems "throttle boost" interferes, even though I used idle up, so when I quickly cut throttle, I would loose control for a second.

For now, I disabled smoothing of throttle and the throttle boost, and the problem have been solved. But i would really like to have the "pid_at_min_throttle" command on a switch, as a flight mode or perhaps turn off, once the throttle have been raised for 2 seconds.

Sorry, I don't have the knowledge to program this my self.

wind0r, 2018-10-18:
Airmode on a switch does exactly what you want.

RipperDrone, 2018-10-18:
Hm, can there potentially be a link btw this issue and https://github.com/betaflight/betaflight/issues/6018? I have pid_at_min_throttle = off on all my builds, and having those arm drops (mostly towards front right in my case, and this holds for minimum 5 different quads!), and 'losing control for a second' might be a similar event like seeing an arm drop on hard throttle cuts...?

RipperDrone, 2018-10-18:
maybe I will try pid_at_min_throttle = ON next time... :-)

DaveFPV, 2018-10-19:
For `pid_at_min_throttle`, this feature deactivates itself when you are above `airmode_start_throttle_percent `(1320µs by default) and never comes back. So, basically, it's only active before the first take-off.

wind0r, 2018-10-19:
Just to understand it. You throttle up switch is working totaly fine when you do
not have throttle boost enable but when you enable throttle boost it doesnt work fine?

So why do you ask for another new feature/mode instead of just fixing the issue?

pid_at_min_throttle = off exists since multply years it does exactly the same as with kiss what mr steele explained over a year ago and also it does the same as in this wild willy video. There is no need to create somthing what does exactly the same again. If there is the issue that should be fixed and not a workaround should get added

RipperDrone, 2018-10-19:
@wind0r - @DaveFPV gave the conformation I was looking for. The quad should not be able to fall back into the PIDs-off status after I have launched. Nevertheless, I experienced it many times on lightweight builds, where I probably did not exceed the 1320us throttle value before landing again and finding the quad go back to PIDs-off idling. This is why I was suspecting PIDs-off status to be responsible for losing control for a fraction of a second during flight when doing an abrupt throttle chop out of full speed. 

According to @DaveFPV's explanation, we have to rule out this rationale... :-/

(maybe I'll give it a try still, just to be sure... :-))

DaveFPV, 2018-10-19:
Yes, on some light and/or powerful quads, and you flight style, you have to check and adapt the airmode threshold if necessary because you can take off at 1100 for example, make some gentle moves below 1320 and have the feeling that the quad is "drunk" if you hit 0 throttle in this period of time because PIDs don't act normally. Just take off your props and make the test on the bench by raising/lowering the airmode value, you'll see that gyros don't respond at 0 throttle until this threshold will be exceeded.

HrrrPippi, 2018-10-19:
Wow thanks. Didn't know it would disable it self. Then I shut my mouth and go fly. Everything is super. Thanks again.

etracer65, 2018-10-19:
`pid_at_min_throttle = OFF` does absolutely nothing if you have airmode enabled and have reached the `airmode_start_throttle_percent` at least once.  Otherwise having `pid_at_min_throttle = OFF` and either having airmode disabled (no feature or mode switch active) or not yet activating airmode with the required throttle percent will mean no control if throttle drops below `min_check`.  This is how it's all supposed to work.

Throttle boost has nothing to do with any of this as the above all refers to throttle **inputs**.  Throttle boost is an **output** applied in the mixer before setting the motors.

closed 2018-10-19 by wind0r

RipperDrone, 2018-10-19:
@DaveFPV so all it's about is the throttle threshold of 1320us which you have to exceed. There is no watchdog timer looking at some minimum time above 1320us? I still get the feeling that these many, many reports of 'feeling like drunk' quads after hard throttle chops might have something to do with PID loop being interrupted for a fraction of a second. However, in our related issue thread it sounded convincing that all flight logs of these events showed normal PID behavior and it must have to do with some strange external forces on throttle chops which nobody can identify until now...

DaveFPV, 2018-10-19:
No timer, just the threshold. The PIDs are never interrupted.

DaveFPV, 2018-10-19:
Problems like you are showing are mostly related to tuning and the fact that chopping hard the throttle can be very challenging to compensate very quickly and perfectly, depending of the movements, the quad, the inertia, if the motor have a too low idle RPM, etc. Antigravity is an example of trick to deal with such behaviour with large throttle changes.

RipperDrone, 2018-10-19:
So what else could we try/tune then?

- we have tried AG off and very high values
- low idle RPM up to 8%

until now it doesn't cure the problem... but maybe we should continue this discussion in #6018 since this one is closed now ;-)

Pls take a reading there, I have summarized what has been tried out by many of us affected users...