betaflight/betaflight #6960: GPS Rescue in 3D Mode

3DRacer, 2018-10-20:
When 3D is activated then gps rescue is not available. Why?

mikeller, 2018-10-20:
Because there are some areas where 3D and GPS rescue don't work together (yet):
- GPS rescue does not support the bidirectional mixer that is used for 3D;
- there is no defined behaviour for when GPS rescue is activated while flying inverted.

Because both of these lead to undefined behaviour when GPS rescue is activated, and because 'long range 3D flight' was seen as a rather fringe case, the decision was made to simply prevent the non-working combination of the two from being configured.

If you want to change this, feel free to submit a pull request to fix the two issues above.

etracer65, 2018-10-21:
Also while inverted in 3D mode the GPS antenna will be posting towards the ground and have at least 2 layers of carbon blocking the signals from the satellites so you're unlikely to have a very good (if any) GPS fix.  GPS Rescue expects a continuous GPS fix to operate and has no ability to "loiter" while waiting for the lock to return.

HazMatFPV, 2018-10-21:
When using 3D mode on a switch will GPS Rescue work if 3d mode is disable?  I don't see the ability to set a switch to enable GPS Rescue if the quad is setup for 3D mode on a switch.  Would like to have GPS Rescue working with 3D Mode disabled.  Will GPS Rescue work if it's setup as the default failsafe?

watageek, 2018-10-28:
@SchonRichting
> 
> Is it possible to set angle mode in stage1 for about 0.8 seconds and then in stage 2 GPS-Rescue is used instead disarming the quad?

@etracer65 is implicitly say the GPS signal will be lost at some point and be unreliable as a safety feature:
> Also while inverted in 3D mode the GPS antenna will be posting towards the ground and have at least 2 layers of carbon blocking the signals from the satellites so you're unlikely to have a very good [...]

And because I was curious, it took 10-15s at the best attempt to have a lock signal when I flipped the quad on the bench.

SchonRichtig, 2018-10-29:
I don’t fly 3D, for me it ist the short time you fly inverted with your normal acro quad.
When I do long powerloops, or flip the quad being a litte bit higher I don’t 
loose much GPS sats in this seconds.
On the ground it is another situation, pointing the GPS antenna to the ground in 1-2 meter height. 
When I loose the RC Link in this situation the quad will drop to the ground, still having 10 or more sats in osd. 
So I will set failsave stage 1 to angle mode and throttle to hold or 1500 for this second.

closed 2018-12-05 by stale[bot]