betaflight/betaflight #6966: LED LOST WITH BTFL OMNIBUS 3.5.2

julc69100, 2018-10-21:
**Describe the bug**
After to install the new firmware. i don,t have the led strip
**To Reproduce**
Yesterday, before to install the new firmware ,i saved my configuration with "dump" 
[BTFL_cli_LD95_DUMP_20181020.txt](https://github.com/betaflight/betaflight/files/2499481/BTFL_cli_LD95_DUMP_20181020.txt)
then i installed the new version BTFL 3.5.2 Omnibus on my new drone Eachine Lizard 95 , and i saw the led strip was lost 

**Expected behavior**
normally , i must had a larson scanner when i connect the battery

In BETAFLIGHT / configuration the led strip stay inactive and in the led strip menu the value is 0 

[image]

normally i must have 

[image]

**Flight controller configuration**

[BTFL-DIFFALL-LD95-20181020.txt](https://github.com/betaflight/betaflight/files/2499476/BTFL-DIFFALL-LD95-20181020.txt)
[BTFL-DIFF-LD95-21-10-2018.txt](https://github.com/betaflight/betaflight/files/2499477/BTFL-DIFF-LD95-21-10-2018.txt)

**Setup / Versions**

Specification:
Brand name: Eachine
Item Name: Lizard95 95mm Micro FPV RACING DRONES BNF
Size:115mm*115mm*60mm
Weight: 66( battery not include)
Flight controller: Betaflight F3 Flight controller built-in OSD
Motor: Eachine 1104 KV6000 brushless motor
Propeller: 50mm 5-blades propeller
Camera: 600TVL HD CMOS 1/4inch
VTX: 5.8g 25MW/100MW Switchable 48CH NTSC/PAL Video transmitter
Battery: 11.1V 550mah 40C lipo battery
OSD: Betaflight OSD
Firmware of Flight controller :Betaflight 3.1
Flight time : 4minutes
Rear LED Ready( LED_Strip function)
Buzzer Ready

Flight controller Specification:
-Size:27mm*27mm*5mm
-Mount hole: 20mm*20mm
-Processor: STM32 F303 MCU
-Sensor: SPI Sensor MPU6000
-Built-in Betaflight OSD ( Batt voltage, RSSI , Artificial Horizon , Fly mode , -Flytime, Craft name etz.)
-Firmware version: Betaflight 3.1 Dshot600 ready
-Target: OmnibusF3
-Work voltage: 2S-3S lipo battery

ESC specification:
-Size:27mm*27mm*5mm
-Mount hole: 20mm*20mm
-Firmware version: BLHELI_S 16.5
- BB2, 48MHZ Chip
-Lipo Battery: 2S-3S
-Con. Current: 10A
-Peak Current: 15A (10S)
-BEC: 5V @0.1A
-Programming: YES
-Betaflight pass-trough ready
-Dshot600 ready(Default)
-Support oneshot42 oneshot125 Multishot Dshot
-Damped light on
 
Features:
-Extremely light and more powerful 3S power makes the Whisper 95 like a beast
-STM32 F303MCU + BLHELI_S ESC make your Operating feel more stability and silky
- Adjustable Angle of Camera
- Adjustable VTX Power 25mw/100mw
- Jaw-dropping flight performance
- Betaflight support
-Acro mode / Air mode /Angle mode support
-Built-in OSD(Using F3 MCU controls OSD over SPI bus in DMA mode)
-OSD Configuration has been include in BETAFLIGHT GUI
-Radio Stick to control PID Tunes
-BLEHELI_S pass-through ready
- Telemetry RSSI Output ready(Frsky Version)

**Additional context**
Add any other context about the problem here.

[ORIGINAL.lizard 95 frsky 3.2.0 dump.txt](https://github.com/betaflight/betaflight/files/2499502/ORIGINAL.lizard.95.frsky.3.2.0.dump.txt)

jflyper, 2018-10-21:
Please search issues before file a new issue.

#6802 
#6773 
#6642

closed 2018-10-21 by jflyper

jflyper, 2018-10-21:
@julc69100 You have to do a custom build; drop unused receiver protocols to make some room and enable `USE_LED_STRIP`. For details, please see other media. Thank you in advance.

julc69100, 2018-10-21:
merci

cvetaevvitaliy, 2019-01-04:
Hi!
I compiled the latest firmware from the source with LED_STRIP, but without the SD Card, since there is no free space in FLASH [(STM32F3)
[https://github.com/cvetaevvitaliy/BetaFlight-OMNIBUS-F3---LED_STRIP-](https://github.com/cvetaevvitaliy/BetaFlight-OMNIBUS-F3---LED_STRIP-)

julc69100, 2019-01-04:
Hi ,

Thanks for your help but i lost my drone there are 1 month ago

I keep your information for my future drone or my friend

Thanks

bye

Provenance : Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows 10

https://github.com/cvetaevvitaliy/BetaFlight-OMNIBUS-F3---LED_STRIP-