betaflight/betaflight #6975: MATEK F405 STD I2C error: 2  when barometer is active

Denys25, 2018-10-23:
**Describe the bug**
Updated my MATEK F405 STD flight controller at firmware betaflight_3.5.2_MATEKF405
when i active barometer in configuration at down status line shows that "I2C error: 2"
(in previous firmwares i not seen any erors)

**To Reproduce**
Behaviour is produced when i use MATEK F405 STD which flashed firmware betaflight_3.5.2_MATEKF405
and in configuration try to active barometer, barobeter is turned on but in status line apears "I2C eror:2"
 
**Expected behavior**
may cause some trouble in flight process when barometer is used in future.

**Flight controller configuration**
 ```
# version
# Betaflight / MATEKF405 (MKF4) 3.5.2 Oct 16 2018 / 22:03:47 (a5e270fee) MSP API: 1.40

board_name MATEKF405
manufacturer_id 

# name
name Denys25

# resources
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 C07
resource MOTOR 4 C06

# mixer

# servo

# servo mix

# feature
feature INFLIGHT_ACC_CAL

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon
beacon RX_SET

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1750 2100 0 0
aux 2 2 1 1350 1650 0 0
aux 3 17 3 1850 2100 0 0
aux 4 28 2 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1500

# master
set mag_hardware = NONE
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT600
set failsafe_throttle = 90
set failsafe_procedure = AUTO-LAND
set small_angle = 180
set pid_process_denom = 2

# profile
profile 0

# rateprofile
rateprofile 0
 ```

**Setup / Versions**
 - MATEK F405 STD bought from Banggod
 - firmware of FC  betaflight_3.5.2_MATEKF405;

**Additional context**
when i use previous betaflight firmwares for my controller this error not apear.
i will appreciate any feedback.

Denys25, 2018-10-23:
> **Describe the bug**
> Updated my MATEK F405 STD flight controller at firmware betaflight_3.5.2_MATEKF405
> when i active barometer in configuration at down status line shows that "I2C error: 2" and in diff is not
> shown that barometer is activated.
> (in previous firmwares i not seen any erors)
> 
> **To Reproduce**
> Behaviour is produced when i use MATEK F405 STD which flashed firmware betaflight_3.5.2_MATEKF405
> and in configuration try to active barometer, barobeter is turned on in menu but in status line apears "I2C eror:2" and in diff is not shown that barometer is activated
> 
> **Expected behavior**
> may cause some trouble in flight process when barometer is used in future.
> 
> **Flight controller configuration**
> 
> ```
> # version
> # Betaflight / MATEKF405 (MKF4) 3.5.2 Oct 16 2018 / 22:03:47 (a5e270fee) MSP API: 1.40
> 
> board_name MATEKF405
> manufacturer_id 
> 
> # name
> name Denys25
> 
> # resources
> resource MOTOR 1 C09
> resource MOTOR 2 C08
> resource MOTOR 3 C07
> resource MOTOR 4 C06
> 
> # mixer
> 
> # servo
> 
> # servo mix
> 
> 
> # feature
> feature INFLIGHT_ACC_CAL
> 
> # beeper
> beeper -GYRO_CALIBRATED
> beeper -RX_LOST
> beeper -RX_LOST_LANDING
> beeper -DISARMING
> beeper -ARMING
> beeper -ARMING_GPS_FIX
> beeper -BAT_CRIT_LOW
> beeper -BAT_LOW
> beeper -GPS_STATUS
> beeper -RX_SET
> beeper -ACC_CALIBRATION
> beeper -ACC_CALIBRATION_FAIL
> beeper -READY_BEEP
> beeper -DISARM_REPEAT
> beeper -ARMED
> beeper -SYSTEM_INIT
> beeper -ON_USB
> beeper -BLACKBOX_ERASE
> beeper -CRASH FLIP
> beeper -CAM_CONNECTION_OPEN
> beeper -CAM_CONNECTION_CLOSED
> 
> # beacon
> beacon RX_SET
> 
> # map
> map TAER1234
> 
> # serial
> 
> # led
> 
> # color
> 
> # mode_color
> 
> # aux
> aux 0 0 0 1800 2100 0 0
> aux 1 1 1 1750 2100 0 0
> aux 2 2 1 1350 1650 0 0
> aux 3 17 3 1850 2100 0 0
> aux 4 28 2 1300 1700 0 0
> 
> # adjrange
> 
> # rxrange
> 
> # vtx
> 
> # rxfail
> rxfail 5 s 1500
> 
> # master
> set mag_hardware = NONE
> set dshot_idle_value = 500
> set motor_pwm_protocol = DSHOT600
> set failsafe_throttle = 90
> set failsafe_procedure = AUTO-LAND
> set small_angle = 180
> set pid_process_denom = 2
> 
> # profile
> profile 0
> 
> 
> # rateprofile
> rateprofile 0
> ```
> **Setup / Versions**
> 
> * MATEK F405 STD bought from Banggod
> * firmware of FC  betaflight_3.5.2_MATEKF405;
> 
> **Additional context**
> when i use previous betaflight firmwares for my controller this error not apear.
> i will appreciate any feedback.

jflyper, 2018-10-23:
@Denys25
When assuming your barometer device is BMP280, two errors on I2C is the expected behavior.

`MATEKF405` target has three barometer devices configured on I2C; BMP085, MS5611 and BMP280. These devices are probed for existence in this order, so time out errors for BMP085 and MS5611 is imminent.

The "older" version of the firmware might have configured only one device, it might have probed BMP280 first, or you might have `baro_hardware` set to `BMP280` specifically.

The two errors does not have any negative effect on flight, but if you don't like it, you can avoid it by setting `baro_hardware = BMP280`.

closed 2018-10-23 by jflyper