betaflight/betaflight #6980: Possible Sim Mode - Like Control Method

totterfree, 2018-10-24:
As I understand it, Betaflight's pid loop focuses on the intended rate of acceleration vs the actual rate of acceleration. Because we can only sample the gyro so fast, this introduces drift in terms of the absolute orientation of the quad. For example, on a day with high amounts of wind, a quad being flown in a straight line (with no pitch, roll, or yaw) inputs can be expected to drift in attitude as time goes on. I believe that without amazingly high sample rates, this effect is much more prominent during split-S turns.

Instead of attempting to match the acceleration as defined by the stick inputs, I propose we attempt to match the absolute orientation of the quad to that of the intended orientation as defined by previous stick inputs. So for example if my quad is oriented at [0,0,0], and I provide a stick input which would result in a desired orientation of [0,90,0], the pid controller's goal is simply to match the orientation as quickly as possible. This could potentially lead to a more predictable flight feel since the same stick input will always yield the same offset in orientation, as opposed to the current method which is dependent on pids, loop times, and outside forces. 

My goal with this feature request is to conceptually develop a more predictable control methodology similar to that of Sim Mode. I believe that a given stick input should provide repeatable results in terms of the quads attitude, as well as similar acceleration speeds. Hopefully this is a stepping stone to that end. It is possible that I have made some untrue assumptions about the way Betaflight works. If this is the case please let me know so that I may correct myself. Ultimately I would like this to be a discussion amongst those with strong control theory knowledge about weather or not this is actually a more effective way of controlling our quads. 

As a side note, if this turns out to be a viable improvement, there will be 🍺 for the person who can implement it (since I don't currently have the necessary skills). 😄

etracer65, 2018-10-25:
How would we know the "desired orientation"? All we know from the pilot's inputs is the desired rate of rotation for each axis. The PID controller already does everything possible to match the requested rotation rate for each axis.

fujin, 2018-10-25:
I do not believe there is any published control theory research regarding the "sim mode" behavior. Anecdotal evidence so far suggests a relationship to [2DOF PIDC (Feed Forward)](https://www.mathworks.com/help/slcontrol/ug/designing-two-degree-of-freedom-pid-controllers.html) capabilities in modern Betaflight.

If you are aware of such research, by all means, please provide links to the papers and such here for review. 

The probability of "sim mode" being a new invention or improvement over the last two centuries of control theory / information theory / control systems engineering corpus generally approaches zero.

ledvinap, 2018-10-25:
@totterfree : What you propose is essentially equivalent to integrating stick input to get reference attitude and then using it as input for attitude (angle mode) controller.
IMO it will mostly combine disadvantages of angle and rate mode, only advantage over angle mode is that reference position will be well defined (in angle mode there are singularities when mapping stick input to attitude)

It is not that hard to implement, starting from quaternion based ANGLE mode (IIRC there are some unmerged PRs). But it is probably not worth the effort

totterfree, 2018-10-25:
@ledvinap What disadvantages are those? I ask because I am genuinely curious. I used Eulerian coordinates for ease of explanation, but quaternion would be better since you avoid gimbal lock, and gain computational speed.

butterSnake, 2018-10-26:
word on the street is that 'sim mode' merely  aggressive FF

totterfree, 2018-10-26:
@fujin It could be how sim mode works, but it's likely that it's not (since the ICM-20602 does not have a magnetometer). Honestly it's irrelevant in my opinion. The problem I'd like to solve with this feature is repeatability of resulting attitude given a series of stick inputs. I think it's possible that sim mode is better than BF in this regard because of aggressive FF, but the problem will surely always exist in both until measurement speeds approach infinity. I hope that PIV could decouple the correlation between fast measurement speeds and good repeatability. So for example an FC with a gyro sample rate of 1MHz would always achieve the same attitude given a series of stick inputs as an FC with a gyro rate of 100MHz. The faster gyro would be able to achieve a given accuracy more quickly, but they would both do it.

totterfree, 2018-10-26:
Something I should also mention (and might be worth updating my post over) is the fact that this feature  might require a magnetometer to be on the quad. As, @etracer65 has pointed out, we don't actually know the orientation of the quad completely given just a gyro, and acc. It's possible we could work out everything except heading. Perhaps someone with more knowledge about this could comment? 

However FCs like the flip32-deluxe-523 do have sensors such as the [HMC5883L](https://cdn-shop.adafruit.com/datasheets/HMC5883L_3-Axis_Digital_Compass_IC.pdf), and I can easily imagine the [Invensense 20948](http://www.invensense.com/wp-content/uploads/2016/06/DS-000189-ICM-20948-v1.3.pdf) being integrated into newer FCs. It's also worth pointing out that these sensors only sample between 75Hz and 100Hz respectively, so the word "supplemental" is key here.

Otarifpv, 2018-11-04:
Reading this my feeling is that Sim mode can be as they say an aggressive FF, people like to give to have a way that does not have to deal with numbers, I do not understand much of the code, but could do a way but about FF do not know if that would work , if you increase the throttle decrease P and increase FF depending on the value of the accelerator a TPA with an increase of FF, not if that would recreate that feeling of sim Mode. I don't understand much just a point of view

RipperDrone, 2018-11-04:
Maybe a little off-topic, but some time back I learnt that BMW transitions from PID controllers to what they called 'gamma control methods'. BMW guy said it was completely different from PID, and faster in homing in on target transients. Have not tried to find out more yet - just thought I'd throw it in here since it seems some people are looking at alternatives to PID control with the aim of making the control faster / more accurate... at least it seems there ARE alternatives out there...

closed 2018-12-11 by stale[bot]