betaflight/betaflight #6982: DALRC F722 Dual ICM20602 Gyro Noise

Dodgetweaker, 2018-10-25:
I have a new DALRC F722 Dual FC and the quad completely flips out when I try to run on the ICM20602.  The quad flies fine on the MPU6000 and on a Hobbywing F4 FC which uses the ICM20602.  The blackbox shows that the pitch gyro is super noisy while the roll and yaw are super quiet.  

Setup:
BF3.5.1
DALRC F722 Dual FC
DALRC Rocket 50A ESC
T-motor F80 Pro 1900kv
TBS Crossfire Nano RX
TBS Unify Pro HV
Foxeer Predator Micro
Rebel High C 1550mah 6S
MAS RS 6" Props

The pictures below were from a hover until the quad decided the do a complete flip before I could disarm.
[image]
[image]
[image]

```# diff

# version
# Betaflight / DALRCF722DUAL (DLF7) 3.5.1 Sep  8 2018 / 05:08:10 (d9fb5ca13) MSP API: 1.40

board_name DALRCF722DUAL
manufacturer_id 

# name
name Steez6Purple

# resources
resource PWM 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_RX 2 NONE
resource I2C_SDA 1 NONE
resource CAMERA_CONTROL 1 B01
resource ADC_EXT 1 A03
resource PINIO 1 B09

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY

# beeper
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH FLIP

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 2 1 19200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led
led 0 0,10::C:1
led 1 1,10::C:1
led 2 2,10::C:1
led 3 3,10::C:1
led 4 4,10::C:1
led 5 5,10::C:1
led 6 6,10::C:1
led 7 7,10::C:1
led 8 8,10::C:1
led 9 9,10::C:1
led 10 10,10::C:1
led 11 11,10::C:1
led 12 12,10::C:1
led 13 13,10::C:1
led 14 14,10::C:1
led 15 15,10::C:1

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 26 0 1700 2100 0 0
aux 5 28 2 1300 2100 0 0
aux 6 35 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1000
rxfail 6 s 2000

# master
set gyro_calib_noise_limit = 148
set gyro_use_32khz = ON
set gyro_to_use = SECOND
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 9
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT1200
set ibata_scale = 138
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2432
set osd_rssi_pos = 2048
set osd_tim_2_pos = 2455
set osd_vtx_channel_pos = 2072
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2423
set osd_craft_name_pos = 34
set osd_warnings_pos = 2442
set debug_mode = GYRO_FILTERED
set cpu_overclock = 240MHZ
set vtx_band = 3
set vtx_channel = 6
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5905
set vcd_video_system = NTSC
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255

# profile
profile 0

set dterm_notch_cutoff = 0
set iterm_relax = RP
set i_pitch = 40
set d_pitch = 32
set f_pitch = 0
set p_roll = 50
set i_roll = 40
set d_roll = 32
set f_roll = 0
set p_yaw = 40
set i_yaw = 120

# rateprofile
rateprofile 0

set thr_mid = 25
set thr_expo = 20
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 73
set tpa_rate = 50
set tpa_breakpoint = 1200```

etracer65, 2018-10-25:
Looks like you have a faulty board and should take it up with the manufacturer.

Strangedave, 2018-10-30:
I have the exact same problem, did you ever find a fix. I have been posting everywhere. DALRC has been unresponsive so far.

GitHub first spot https://github.com/betaflight/betaflight/issues/6893
GitHub second spot https://github.com/betaflight/betaflight/issues/6852#issuecomment-431602860
Also on facebook https://www.facebook.com/groups/betaflightgroup/permalink/241151723247378/

No solution so far. I have two identical Kwads, one runs great the other one is trying to make me spend all of my money.

Strangedave, 2018-10-31:
> @Strangedave I have not found a fix and I have not gotten any response from DALRC after several different means of contact. Do you get the issue on the MPU6000?

No, that Kwad flies reasonably well on the mpu6000 gyro. I had given up and spent some time tuning on that gyro. I have two exact same builds the one that is running on the IMU20602 gyro is much better though. Way more precise. I’m going to try and return the one to getfpv. I’ll keep you updated.

Strangedave, 2018-11-01:
Thought I bought this from GETFPV.com, it was actually from RACEDAYQUADS.com. here is their response. You should probably see if whoever you got yours from will switch it out. 

Hello,

That link is perfect. 
It is pretty clear that the FC is defective at this point.
I really appreciate you being so diligent in trying to troubleshoot it.
Would you prefer we send you a replacement or store credit so you can choose another FC of your choice?

I'm happy to help you get this thing flying right as soon as we can.

Thank you for your continued support,
Derek A. Brown
Customer Service Specialist
www.RaceDayQuads.com
Visit us on Facebook

On Wed, Oct 31, 2018 at 4:19 PM Dave Goddard <dave@dave-goddard.com> wrote:
Here is the entire story in one link I have been working this with others on facebook. https://www.facebook.com/groups/betaflightgroup/permalink/241151723247378/
 
I have pics and logs and so on uploaded there already.
 
Dave G
 
From: RDQ Support [mailto:support@racedayquads.com] 
Sent: Wednesday, October 31, 2018 5:37 PM
To: dave@dave-goddard.com
Subject: Re: I have some issues with the DalRc F722 flight controller I bought from you. DalRc has..
 
Hello, Sorry to hear this! Please share a picture of the setup plus the steps you took to try and debug and we will get this resolved for you ASAP! We currently do not have a shop line, sorry. 
I look forward to helping you with this.

Have a great day,
 
Derek A. Brown
Customer Service Specialist
www.RaceDayQuads.com
Visit us on Facebook

Strangedave, 2018-11-06:
I have an update, there was another post in a different thread that might have some promise. Its too ugly outside here to get in a real test flight, however maybe you want to try this as well. I turned Black Box off and the problem seems to be gone. here is a link to the other thread. https://github.com/betaflight/betaflight/issues/6852#issuecomment-436362742

nyway, 2018-11-07:
@Dodgetweaker  
Hi,we have not yet received any contact in this regard.Which contact information do you use?
For ICM20602,there are very high requirements for shock absorption.
@Strangedave 
We are still working on this issue about BBL,it doesn't look like a hardware failure.

Dodgetweaker, 2018-11-07:
@nyway. I sent another email to zhong@dalrcmodel.com.  Please let me know if I should use another address.  Thanks.

Strangedave, 2018-11-07:
Dodgetweaker - Were you able to test with Black Box logging turned off? I am still wondering if we are all having the same problem.

Dodgetweaker, 2018-11-07:
@Strangedave - I have the issue with the BB not logging as well.  I do not see a way to disable BB in the configuration so I assume you are referring to logging vs not logging.

Strangedave, 2018-11-07:
On the Black Box Tab in configurator the top drop down menu should say "No
Logging" then click save and reboot.

<mention@noreply.github.com>
(#6982)

 

<https://github.com/betaflight/betaflight/issues/6982#issuecomment-436792557
ka43Go3ks5us1dXgaJpZM4X5VRk> .
<https://github.com/notifications/beacon/Ak1mkKKwBPjpIFofAZj3IxSjbQjA5GB2ks5
us1dXgaJpZM4X5VRk.gif> 

  <http://s.bl-1.com/8m65nNg.gif>

Dodgetweaker, 2018-11-07:
Just tried it.  No change.  It still jumps randomly while in a hover.

Dodgetweaker, 2018-11-10:
Issue fixed!!!  I purchased another DALRC F722 Dual FC.  The issue is gone.  It was a faulty gyro.  Now, I need to get a refund.

Strangedave, 2018-11-12:
I was afraid that a high percentage of them had some sort of manufacturing defect. I have 2 that exhibit the known black box/CPU problem, and one that runs great. I am awaiting a 4th unit now. We will have to see. Glad to hear yours is running that makes me feel like I have a chance.. The people at DALRC are saying that there will be a software fix at some point. Maybe that will bring your bad one back to life. Either way it seems like we are all getting squared away one at a time. Good luck out there hopefully we will not meet again the same way.

adamjus, 2018-11-13:
hey....wich uart should connect crossfire and the tbs unfiy on the  Dalrc F722Dual FC and wich gyro is best ?

adamjus, 2018-11-13:
Gonna use only crossfire and tbs unify smartaudio....wich gyro is best ?

nyway, 2018-11-13:
I don't know the crossfire,sorry about that.
About smartaudio connect ,you could  see the pic below,it's TX5,near VTX port:
[image]
About which gyro,reference what I said above,it is up to your preference.
ICM20602 need more higher shock absorption.

dadoribeiro, 2018-12-14:
I just got a F722 and everytime I try to use 32k gyro (with PID at 8k, with static CPU at 20%) with the Blackbox logging at any rate I get severe jitters and the motors get really hot. If I disable 32k and use the second gyro (ICM) ou just use the first gyro (MPU) I can log at any rate.

Strangedave, 2018-12-14:
Been there done that. I had 2 in a row with the same issue. I am not at home right now, but when I get home I’ll send you links to all of pertinent info. You probably need a new flight controller. Everyone else I was in contact with has already received a replacement and the problem was corrected. To be fair the engineer at DALRC told a few of us that it is a software problem, but no one has ever issued any patch, and at this point I would probably not spend the time to solder the old board into a rig to test it. I have two rigs running this flight controller, and once you have a good one they are great. Good luck 

DG

Strangedave, 2018-12-15:
Here are the other threads that pertain to your issue. If I were you I would do my very best to determine if you have a noisy motor or out of balance prop before taking your rig apart again. My rigs are both pretty picky about vibration. The smallest dent in the props will cause jitter on my rigs when I'm running in 32/16k on the IMU gyro. That said I can usually get the thing running smooth again by switching back to the MPU6000 gyro and running 8/8. I race one of the rigs, so it sees a lot of crashes. I have uses 3 arms and 4 motors on that setup already. 

https://github.com/betaflight/betaflight/issues/6852#issuecomment-437772687
https://github.com/betaflight/betaflight/issues/6893#issuecomment-434491918
https://www.facebook.com/groups/betaflightgroup/permalink/241151723247378/

dadoribeiro, 2018-12-15:
Thanks for the info. I contacted RDQ and waiting for them now. I think it is related to the logging, because if it is no the 32k is has heavy jitter on plain hover on a new quad with new props and motors. Without logging it flies fine  but the motors are hot in 32k and with no logging it is impossible to tell why. In 8k MPU it appears to fly fine

Strangedave, 2018-12-15:
That sounds exactly like what mine was doing. It was definitely the flight controller in that instance. Mine still twitched a little though even with the black box turned off in 32/16. It was flyable though. Ran great in 8/8 on the other gyro. Good luck. 

Dave Goddard

Strangedave, 2018-12-15:
Well Boys and Girls, The DALRC F722 is down again. I had one sort of harry crash, and now its all twitchey again. I'm posting this to the various spots so everyone will know the end of the story, and this is the end. I ordered the Strix Binary F10 a few minutes ago. Check back soon for a update with that Flight Controller installed. Here is the video https://youtu.be/M_FU3wcZqBY

Dodgetweaker, 2018-12-16:
@Strangedave Is it both gyros or just the ICM?
I bought a replacement DAL722 and I have not had any issues since.  The ICM was definitely faulty on my original FC.  I wonder if the ICM gets damaged easily.  Not sure why on the FC more than others.

etracer65, 2018-12-16:
@Strangedave That flight controller is not supported on Betaflight.

Strangedave, 2018-12-16:
I’m not really wanting to learn any new stuff, but butterflight seems close enough that I can figure it out. I am hoping that you will add it in when 4.0 is released. If your saying that is not going to happen ever I might need to rethink this decision.

etracer65, 2018-12-16:
No, it won’t be included ever.

Strangedave, 2018-12-16:
I might cancel this order. Can you recommend something F7 that isn’t so fragile. Was also considering matek F7 and the new foxeer F7 but they are both running the same gyro, and I am officially scared of that part.

parallaxfpv, 2018-12-16:
@Strangedave Strangedave Kakute F7 uses an ICM20689, not the ICM20602. I wonder is it any better or worse?

Strangedave, 2018-12-16:
Kakute F7 has the ribbon cable gyro. I had a betaflight F7 board that was killed when some scraggle ripped that ribbon off of the board. That betaflight F7 board still had a mpu gyro so I switched to 8/8 for probably 10 packs and that was when I realized I really like faster loop times/feel of 32/16. I was kind of avoiding that ribbon cable setup. I spent a few hours digging on this. I think I’m going to leave the order in place. I’m hoping that betaflight is added to strix board in the same way that it is available on the helio board. I hope that is not politically incorrect. I’m probably building a 3rd rig anyways. Might preorder the new Lux F7. Same gyro, but I have always had solid products from Lumenier. All of that said this is getting out of controll. The intention when this all started out was to have a few rigs that were exact copies so tuning info from one could be updated to the others. Tuning 3 or 4 rigs that are different uses basically all of my time. My brief test on beraflight 4.0 was all default, and I am sure that ran better than my betaflight 3.5.3 tuned setup, so maybe in the future it’s not going to be so time consuming.

dadoribeiro, 2018-12-16:
Using the ICM gyro at 8k (32k mode disabled, gyro 2) logging works, as gyro 1 (MPU). But I can´t seem to get the logs on a Mac, through the configurator or MSC (it works fine with my Kakute F4 and OmniF4). The Blackbox if definitely a problem in mine at least.

mikeller, 2018-12-17:
@Strangedave:

> hoping that betaflight is added to strix board in the same way that it is available on the helio board

You are wrong there. Betaflight is not available on **any** helio product, and is unlikely ever going to be. What is available is a hacked version that is infringing against the rights of all contributors to Betaflight by breaking the terms that Betaflight is licensed under and has no support from its authors whatsoever.
As long as helio is disregarding the terms of the license that Betaflight is licensed under, there is no way Betaflight is ever going to support it, and I can only recommend against supporting them by buying any of their products.

mikeller, 2018-12-18:
@Strangedave: All good, I am sure it'll fly. ;-)

Re the hardware side, I doubt that using boards from helio or any other manufacturer will be significantly less susceptible to having a bad signal to noise ratio when running in 32 kHz mode, especially after they've been subjected to hard impacts. There is only a limited number of different gyro models that are used across boards from all manufacturers, and they are all subject to this problem.

Strangedave, 2018-12-18:
Not totally certain I'm gonna build it now. Since I have you, are you suggesting that the GYRO is being damaged in the impacts, or that the motors are being knocked out of balance and causing too much noise. My suspicion is that the motors are the culprit, but most of my testing has not been able to prove that out. I have changed motors, balanced prop, built test stands to check vibration with the props on, and so on. I am starting to think there is variation in the GYRO's; how noisy they are. I have always considered the motors and props consumables. Are we thinking the flight controllers are also consumable, or maybe we could recalibrate the gyro from time to time.

mikeller, 2018-12-19:
@Strangedave: I think it's a case of both (or rather all three) causes:
- we know the current generation of gyros are a lot more susceptible to being influenced by vibrations / noise when running at 32k than they are when running at 8k. Unbalanced / damaged motors props cause vibrations, so there is a correlation;
- speaking from personal experience and user reports here, it is not uncommon for gyros to work fine at 32k when they are new, but to degrade to producing too much noise to be ever again usable at 32k after some use / abuse / impacts (hard to differentiate between the 3 in our hobby). These gyros normally keep working fine when used at 8k, so it stands to reason that the manufacturer has designed them to have a reasonable MTBF at 8k, and the 32k modes are just thrown in as 'use at your own risk';
- anectodal evidence from a post in here a while ago by somebody hand building a batch of flight controllers has it that they tested a tube of gyros and binned them according to their noise levels at rest, and there apparently being a lot of variation already in new gyros. This is supported by other posts of users finding that some boards of a certain type will work fine at 32k when new, while others won't.

Strangedave, 2018-12-19:
The binning makes a lot of sense. I’m going to see if I can get a log out of these rigs with no motors attached and see what that looks like. I feel like there needs to be a way to know which component is the primary issue today. Maybe there could even be an automated self test. Take the props off set the thing on a towel and let it run a procedure that checks motor noise and gyro noise against some known standard. Obviously that’s just my brain spouting ideas. I have no programming abilities.

Strangedave, 2018-12-20:
Ok, we tried learned some things today. I disconnected the motor signal wires from the 4in1 esc and ran a 1 minute log on both of my DALRC F722 Rigs. The builds are identical except for the camera which should have no impact on this test. The working super smooth rig has never been crashed, the other rig has been racing 2 times 13 heats. nothing horrible, but it's been banged around enough to develop Parkinsons. here is the video of the twitchy rig https://youtu.be/M_FU3wcZqBY . There are visible differences in the Plazmatree graphs, but i don't think its anything earth shattering. I have attached the BB Logs and Plazmatree jpg's for review. When I test the twitchy rig for motor vibration it really isn't showing any thing more than a brand new motor, but the bearings are definately making audible noise. Here is a video of the bearing noise, nothing super bad. https://youtu.be/kUw6UM4QT4s  I will change those and report back. new bearings are eta Friday. 
[Working DALRC F722 No Motors.zip](https://github.com/betaflight/betaflight/files/2697275/Working.DALRC.F722.No.Motors.zip)
[image]
[Twitchy DALRC F722 No Motors.zip](https://github.com/betaflight/betaflight/files/2697276/Twitchy.DALRC.F722.No.Motors.zip)
[image]

dadoribeiro, 2018-12-20:
I still havent heard back from RDQ and DALRC regarding this gyro/BB issue. But I found out that the MSC command to mount as drive does not work on MacOS (my KakuteF4, OmniF4 and LUXF4 work, so the Mac is "fine"), and using the configurator is incredible slow to pull the logs (about an hour). But I found that you can force mount using the terminal "diskutil mount readOnly /dev/disk2s1" and it works as the others. Maybe a hardware issue with a way to correct by software?

dadoribeiro, 2018-12-20:
> The mass storage mode problem on MacOS was fixed in 3.5.4 (#7145 ). It was not specific to this flight controller. It affected any flight controller with on-board flash for blackbox (instead of SD card).

I´m aware of this. All my other FCs work fine in MacOS without issues after this update #7145. They all work in Windows (including the DALRC F722). But the DALRC F722 will not mount automatically on MacOS unless I do it manually on the Terminal using the command above. I´m on the latest 4.0 (as of December 17th that is, on all my boards). This is the only F7 I have. If someone else has this FC and a Mac would you please test it? I´m pretty sure this board has design / productions issues, but some maybe fixable in software.

etracer65, 2018-12-20:
Unfortunately I don't believe the manufacturer sent any of the devs samples of this flight controller - I know I don't have one.

Strangedave, 2018-12-20:
I have an extra. It probably has a bad imu gyro, but everything else works. If you want it. Let me know.

etracer65, 2018-12-20:
@Strangedave If you want to donate it I'm sure it will help. Contact me as @etracer on the Betaflight slack. If you've not already joined you can start here: https://slack.betaflight.com

Strangedave, 2018-12-20:
Just requested access. I cant program, but I am happy to provide any assistance that I can. What you are doing is MAGIC. I would just like to be involved. It still absolutely blows my mind how much things have improved in just 1 year.

mikeller, 2018-12-20:
They sent me a DALRCF722. I don't have a Mac but I'll see if I can borrow one to check if MSC works on this board (and on another F7 for reference).

etracer65, 2018-12-20:
@mikeller Dave is sending me the spare one he had. He's only 150 miles away from me so I'll have it in a day or so and can test on Mac.

I recall that as part of the MSC fixes for MacOS you already tested on another F7 and was working as expected.  So I expect it will work fine on this board as well unless there's some hardware implementation problem.

etracer65, 2018-12-21:
I received the board from @Strangedave (thanks!) today and was able to confirm that there is still a problem with MSC on MacOS related to F7. I think I know what it is and should be an easy fix.

etracer65, 2018-12-21:
@dadoribeiro (and anybody else that wants to test) Please try the attached firmware. It should fix the MSC mode filesystem mounting on MacOS. It wasn't a problem with this specific board - rather a generalized problem with F7. Not a problem with logging rate etc. Just a bug in the libraries provided by ST.  I fixed the original problem on F4 but it turns out there's another copy of the libraries (with the same bug) specific to F7 so the same fix is needed there.  Just didn't have a F7 with on-board flash to test previously.

[betaflight_4.0.0_DALRCF722DUAL.hex.zip](https://github.com/betaflight/betaflight/files/2704558/betaflight_4.0.0_DALRCF722DUAL.hex.zip)

dadoribeiro, 2018-12-21:
Thanks. Getting late here but i will test early in the morning

dadoribeiro, 2018-12-22:
Just tested it and working fine. Thanks again.

closed 2019-01-29 by stale[bot]