betaflight/betaflight #6983: CC3D and NAZE32 3.2.x

watageek, 2018-10-25:
Does the maintenance release of 3.5.2 have something that could be applied to 3.2.x maintenance release?

I still fly CC3D...

McGiverGim, 2018-10-25:
Not directly. There are lots of changes between 3.2 and 3.5, so the maintenance release is not compatible. Some bug maybe affects both versions, but it will not be a direct application.

McGiverGim, 2018-10-25:
Yes, you can build custom firmware for NAZE (the target files are in the repo). Removing leds and the RX protocols that you don't need can make space for some new feature.

But they are out of support in Betaflight so you must do the custom build yourself or ask for help in the Betaflight Slack chat or other places.

jflyper, 2018-10-25:
I doubt if we can put any new flight behavior improvement code onto F1.

nightflasher, 2018-10-25:
I had a short look over to cleanflight and found this link, maybe this helps us building custom betaflight for F1 flight controllers. 

https://github.com/cleanflight/cleanflight/blob/master/docs/Customized%20Version.md

mikeller, 2018-10-25:
One of the major shifts between 3.2 and 3.5 has been that pretty much all of the data paths for flight related data have been converted from integer to floating point, and normalised. Considering that F1 MCUs do not have native floating point support this means that these changes won't be viable on F1, and all improvements that were made after this (and to some extent were enabled by being able to deal with normalised data) will have to be reimplemented for F1.

Also, F1 was already stripped down to the bare essentials, and still running out of space, so there is not much hope to get more in.

Plus, note that these days, F3 isn't much better off with respect to permanently running low on flash space, and not having a lot of the newer features for this reason. Go for F4 or better, guys.

watageek, 2018-10-25:
The kids and relatives won't touch my F4 build. They are already overwhelmed with the ol'stuff. Thanks for the quick response.

closed 2018-10-25 by watageek