betaflight/betaflight #6984: Flight Controllers with F411 MCU Freezes during mid-flight

msnrooon, 2018-10-25:
**Flighr controllers with F411 MCU freeze during mid flight RANDOMLY.**
Do not get any warnings and it does not appear to be losing signal. 
Have a good signal and full battery when this happens. The drone turns upside down very quickly and then crashes with the motors spinning at what sounds like full speed. 
The OSD on the screen also appears to freeze. this happens randomly and requires me to unplug the battery and reconnect for it to work again.
**Important Point**: The issue mentioned above exists both in brushed and brushless flight controllers.

Can't provide the steps to reproduce as the issue happens randomly.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

MATEKF411RX (M41R) 3.4.1 from BETAFPV (Brushed & Brushless)
A camera that supports Smart Audio is soldered onto the FC.

msnrooon, 2018-10-25:
The Betaflight Setting from one of the brushless FC

# version
# Betaflight / MATEKF411RX (M41R) 3.4.1 Aug  2 2018 / 08:11:22 (1d99444d9) MSP API: 1.39

board_name MATEKF411RX
manufacturer_id 

# resources
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource LED 1 C13
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 B02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
[... 659 lines cut ...]
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
set abs_control_gain = 0
set abs_control_limit = 90
set abs_control_error_limit = 20

# rateprofile
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_type = OFF
set throttle_limit_percent = 100

mikeller, 2018-10-25:
@msnrooon:

> feature TELEMETRY

It is well known and documented that the FrSky SPI RX has issues with lockups when telemetry is enabled. See #6172 for a discussion.

If you turn off telemetry, this should be fine.

closed 2018-10-25 by mikeller