betaflight/betaflight #6987: Camera Control beeping even if Beep options disabled

skorzo, 2018-10-25:
Airbot Omnibus F4 V6 running Betaflight 3.5.2 (OMNIBUSF4FW (OBFW) target ; no V6 target available yet)

The FC has a camera control pad so no resistor needed.

I have configured camera control for a Runcam Swift v2 and everytime I enter the camera menu or I move the right stick, the esc/motors beep two times.

Its the same sound like FC make when it reboots.
Buzzer is operative but Buzzer options for camera control are both disabled.

resource CAMERA_CONTROL 1 B07 (changed from B09 to B07 in order to work in this FC)
camera_control_mode = SOFTWARE_PWM    (camera control does not work in Hardware mode for me)
camera_control_ref_voltage = 328
camera_control_key_delay = 150
camera_control_internal_resistance = 474

No capacitor and no resistance. Osd pin to Camera control pad and GND pin to GND pad

I dont know why are the motors beeping even if buzzer is operative AND even beeper options for camera control are disabled

jflyper, 2018-10-26:
CLI diff please.

skorzo, 2018-10-26:
diff all

# version
# Betaflight / OMNIBUSF4FW (OBFW) 3.5.2 Oct 16 2018 / 22:09:14 (a5e270fee) MSP API: 1.40

board_name OMNIBUSF4FW
manufacturer_id 
mcu_id 001b003d4e48501220373352
signature 

# reset configuration to default settings
defaults nosave

# name

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09
resource CAMERA_CONTROL 1 B07

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon

# map

# serial
serial 0 0 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1750 2100 0 0
aux 1 1 2 900 1200 0 0
aux 2 2 2 1375 1700 0 0
aux 3 13 0 1750 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# rxrange

# vtx

# rxfail

# master
set gyro_to_use = SECOND
set acc_calibration = -115,47,-507
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT1200
set align_board_roll = 180
set ibata_scale = 179
set ibata_offset = 0
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2316
set vtx_band = 5
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = PAL
set camera_control_mode = SOFTWARE_PWM
set camera_control_ref_voltage = 328
set camera_control_internal_resistance = 474

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

TheTape, 2018-11-07:
Excuse my negativity (this is who I am, I don't mean offense, I mean to share my honest opinion in the most direct way possible)... Why blindly ask for a diff when the description clearly outlines the issue? How extremely halfassed, thoughtvoid and irritating... This general theme is really pissing me off. You're idiots treating people as if they were idiots (asking for expected behavior when it's obvious 99% of times is a good example, it's an insult to basic intelligence).

It's obvious that the FC becomes unresponsive since the ESC's loose a valid signal. Whatever could _possibly_ be locking up the FC for some delayed amount of time (camera_control_key_delay)? Oh what will become of us! Wtf is a brain??

It took this: grep --include=\*.{c,h} -rnw -e "cameraControlConfig"
to find this:
```
src/main/drivers/camera_control.c:228:    } else if (CAMERA_CONTROL_MODE_SOFTWARE_PWM == cameraControlConfig()->mode) {
src/main/drivers/camera_control.c:234:            delay(cameraControlConfig()->keyDelayMs + holdDurationMs);
```

Oh look, someone used delay(), halting the CPU and thus motor output... I still don't understand why OP's ESC's are loosing signal??? Remains a mystery to me! _Herpdiderp_.

You can't figure this out in 13 days but some dude who's been awake for 24+ hours burned out on hash from bong just concludes this was the likely issue upon actually reading and applying the most basics of logic. I hope you're really proud of yourself. Congratulations. Not to put you down (thanks for your contributions) but ffs, get a fucking grip dude.

mikeller, 2018-11-07:
@TheTape: You need to tone it down and stop insulting people in here, or you will be banned.

Also, be aware that the users answering support requests in here are volunteering their time to help others, so for them to ask, and for users seeking support to provide the information (without having to ask them twice) to give them the best chance of diagnosing a problem should just be basic courtesy and respect. As anybody with any experience in QA will know, being able to reproduce a problem is an important part of being able to diagnose and fix it - and in order to reproduce a problem with Betaflight accurately, a `diff` is needed - otherwise it's just guesswork with a lot of unknowns.

TheTape, 2018-11-07:
No worries, I have no intention of staying around easily offended people so entitled they think they're supporting bug reporters, people that are this dense in mindset while pretending the opposite. The kind that's reluctant to pursue any real breakthrough. The kind that can only really think statically. BF will never get anywhere truly significant with that holding you back. You'll keep thinking PID control is better than simulating simple physics according to physics related values, given or calculated. We'd have auto output assignment and auto tuning of physical parameters if it wasn't for backwards people like yourself. Keep fixating and tweak the old. If I can't even get my opinion across without turning into some fake stuck up the ass PC egomaniac what is the point? I wish you could handle harsh reality, harsh honest opinions, that's all it is. I didn't know this was the senate. Fuck.

DieHertz, if you can avoid having an angry fit and close the thread for a min, you're now arguing why asking for the diff was even further irrelevant; why are you telling me, and not OP those things? Doesn't seem to be removed from the current 4.0 source, that was what I copy/pasted the relevant code from. What if the ESC recalibrates from the potential elongated high signal? And why would you exclude possibilities just because it currently isn't "meant" so? So what? Again, this cling to the old reluctant mindset, real beneficial for projects like these. How about working towards everybody having what they need instead of trying to dictate it? How about you implement scripts (pawn) dynamically loaded from ext (spi flash, sd), move 2/3 of BF into scripts, saving incredible space and enabling all functionality even on F3, simultaneously allowing anyone to create virtually any functionality or compatibility easily?

fujin, 2018-11-07:
Insulting and derogatory comments are not up to the standards expected of the Betaflight community.

Please refer to [Betaflight Community Code of Conduct](https://github.com/betaflight/betaflight/blob/master/CODE_OF_CONDUCT.md) for our expectations.

We make use of GitHub Issue templates specifically describing required information for a variety of types of problem reports; one of the requirements was missed, then added per our request.

The issue templates cut down on significant amounts of maintainer and community time when used, and enable a maintainer or volunteer from the community to potentially reproduce issues *immediately* by simply flashing the same version and configuring the supplied `diff`.

skorzo, 2018-11-27:
The related issue was initially just a collateral beep tone when using camera control but now it has become a safety issue and I think it should be considered.

Let me explain.

@TheTape 

Probably you were right.

I changed key delay value from 180 to 100 and now ESCs dont beep  (not always al least...)

**First Scenario (widely reproduced)**
I can access camera osd by holding left stick to the right for a couple seconds, as intended, and then I can navigate by moving right stick.  ESCs dont beep in this scenario.

**Second Scenario (rarely reproduced)**
ESCs only beeps if I try to access Camera special menus that require holding a button for a long time. I havent reproduce this particular scenario so much times, but sometimes I managed to access special menus and ESCs started to beep.

So this is another clue pointing to FC unresponsiveness related to key delay / holding a key

**Third Scenario (hardly ever reproduced)**
The issue is now bigger than before.

**Eventually I managed to arm the quad and Motors started to spin** just by moving and holding sticks, while trying to reproduce 2nd scenario.

Of course arm mode is set to a switch so this should not be possible.

As I said, the issue now has become a safety issue and I think it should be considered

And  about @DieHertz  post

> Camera control isn't meant to be used when motors are running

Motors were not running. The quad is on the table, disarmed, and ESCs beeps when using camera control

> Software camera control has long since been removed from the code

AFAIK software mode is still on the code. It still works on 3.5.3 and 4.0

mikeller, 2018-11-27:
@skorzo: Can you describe where the perceived 'safety issue' is? It is not obvious from your statement above.

Also, I think @DieHertz meant 'software PWM'.

skorzo, 2018-11-27:
Not everybody remove props before working on the table, and this issue can arm the quad and make motors spin.

And I mean software pwm too ... that's the mode Im using to make camcontrol work on omnibus f4 v6

mikeller, 2018-11-27:
@skorzo: 

> this issue can arm the quad and make motors spin.

Can you please explain how you think it should be possible that this would arm the quad?

skorzo, 2018-11-27:
I don't know how but it happened. It seems that FC is momentarily locked up due to key delay and ESCs loose and recover the signal. Maybe this disfunction could "make" an arm order.

I know this is crazy because arm is set to a switch and I can't even arm by stick commands, but believe me, playing with camera control key delay and holding sticks I have been able to arm. I can't reproduce it by the way.

In any case the FC is apparently locking up depending on key delay and Escs loose signal and beep, I can reproduce this whenever I want.

Two different side effects, but I'm sure that both are related to this feature

mikeller, 2018-11-27:
@skorzo:

> I don't know how but it happened. It seems that FC is momentarily locked up due to key delay and ESCs loose and recover the signal. Maybe this disfunction could "make" an arm order.

What is an 'arm order'?

mikeller, 2018-11-28:
@skorzo: I am just trying to understand where you think there could be a safety issue in this, and it is not clear to me what this 'arm order' that you think could be triggered by the delay induced by the camera button signal being generated is - do you think that this will cause the flight controller to arm, and if so why would this be?

skorzo, 2018-11-28:
@mikeller 

Let's see,

The safety issue is that the motors could  start spinning unexpectedly and hurt somebody while you are only trying to adjust your camera settings.  This happened to me.  The quad was disarmed and while I was sending stick commands to access the camera via camera control, the motors began to spin.

It was not an arm sequence triggered by sticks commands because arm is set to a switch and this shouldnt be possible.

Why did my motors begin to spin?  I dont know. 
I was wearing the fpv goggles trying to see the camera menu and then I heard the motors spinning and even touched my hand (no props)

In fact I dont even know if the quad was technically armed or not, I cant remember if I heard the arm beep sequence, but the fact is that I did not arm the quad and the motors suddenly spun.

So, honestly, I should not have said "arm" but "motors began to spin"  if that is the point.

Why do I relate this to camera control? 
While trying to adjust camera control parameters, specifically  key delay, I noticed that ESCs were beeping (startup tone) each time I tried to access camera control (holding left stick to the right)

Camera control worked, but the annoying beeps were there.

As pointed by @TheTape  it seems to be some kind of issue with camera control software pwm mode that could lock up the FC.  Maybe this is causing my ESCs beep.

And, maybe, this FC unresponsiveness caused my motors began to spin in further tests.

Regardless you could consider or not the "motors spinning issue" proved or not, I think that the issue of FC becoming unresponsive is there.

mikeller, 2018-11-29:
@skorzo:

> This happened to me. The quad was disarmed and while I was sending stick commands to access the camera via camera control, the motors began to spin.

What ESCs and ESC firmware are you using?

If this was caused by you using software PWM for camera control, then your ESCs are dangerously broken, and should probably be discarded and replaced. What happens when software PWM for camera control is used to send a keypress to the camera is that the flight controller is kept busy for the duration of the keypress, and the ESC 'disarm' or 'idle' stops being sent to the ESCs. Normal and safe ESCs react to this by rebooting (the beeps you hear), and then refusing to spin until the 'idle' signal has been restored for a certain amount of time. If your ESCs are spinning up when not receiving a signal then you should immediately remove them from your build and not use them again.

skorzo, 2018-11-29:
Thanks for your reply @mikeller 

You really scaried me ...
Your conclusion makes a lot of sense.

My ESCs are Airbot Furling32 and I have flown only 4 o 5 lipos.
I guess that when you said "broken" you meant "not well built", and it is not a failure caused by me or by a crash, isnt it?

According to your reply I understand that there are more situations where ESCs could spin up, not only caused by FC kept busy, Is that the situation?

What kind of situations besides FC kept busy by camcontrol are susceptible to make idle signal stop from being sent to ESCs ?

I would like to reproduce and test situations before discarding 4 brand new ESCs, but in fact, I distrust them now.

Many thanks

mikeller, 2018-11-29:
@skorzo:

I don't think there is much reason to be scared - the Airbot Furling32 uses BLHeli_32 firmware which is fairly common, and there are no other bug reports for the behaviour that you describe - it would have been different if you'd used some obscure and uncommon ESCs from a manufacturer writing their own firmware.

Maybe upgrade your ESCs to the latest firmware version with the BLHeli32 suite, to err on the side of caution.

One more thought that I had - if the camera control PWM signal accidentally made it to the signal input of one or more ESCs this could be interpreted as a valid PWM control signal by the ESC. Check your wiring, make sure your cam control output is not accidentally soldered to an ESC or to ground, and that your ESCs have all ground wires from the signal ground pad to a ground pad on the flight controller installed.

skorzo, 2018-11-29:
I just checked my cam control output and it is not soldered to an ESC or to ground.

Unfortunately furling32 ESC has no ground pad, it just has signal and tele.

Thank you very much, Im going to update the ESC and of course I will remove Camera control wires and I will discard this feature, just in case.

closed 2019-01-05 by stale[bot]