betaflight/betaflight #6988: GPS auto config leave HDOP (tmp2 sensor) @ zero

RotorVulura, 2018-10-25:
Hi,
As per subject, I can see in my taranis tmp2 telemetry sensor, something like 3024 where 3 is 3D fix (values are from 1 to 4), 0 Quality maybe HDOP (values should be from 0 to 9) and 24 is the number of the linked satellites (values may vary from 1 to whatever...).
So I’ve got 24 sat and 3D fix. I think that 0 quality is almost impossible. What I’m doing wrong ?
GPS rescue works perfectly, anyway. Very good features :)
Thx for your time and RotorVulura !

fsciam, 2018-10-25:
Hi RotorVulura! I think that tmp2 is not computed as you think.
You can see in betaflight/src/main/telemetry/smartport.c file (precisely line 722 )
`              
                    smartPortSendPackage(id, (STATE(GPS_FIX) ? 1000 : 0) + (STATE(GPS_FIX_HOME) ? 2000 : 0) + gpsSol.numSat);`

that the value is built as follow (final value is obtained summing numbers obtained at each "step"):

- 1000 if you "have" a 3d fix, 0 otherwise;
- 2000 if an home point has been recorded, 0 otherwise;
- number of satellites "seen" by your gps sensor;

The number you see means that you see 24 satellites, you have a 3d fix and home point has been recorded. 

Hope it helps

RotorVulura, 2018-10-25:
It makes sense, but usually up the mountain where I like to fly, there are no more than 8 sat linked instead 24/26 reached in the country. Long story short, I would like to know how good that signal is. iNav works like esxplained above, but it’s not so good for aerobatic session ...
Thx for looking into this, all the best and RotorVulura !

closed 2018-12-01 by stale[bot]