betaflight/betaflight #6990: Cannot enable LED Strip on Eachine Aurora90 Omnibus on Betaflight 3.5.2

archont, 2018-10-26:
**Describe the bug**
LED Strip configuration option toggles itself off after reboot

**To Reproduce**
1. Flash 3.5.2 on Eachine Aurora 90
2. Enable LED Strip
3. Observe LED Strip tab is missing after reboot

**Expected behavior**
1. Flash 3.5.2 on Eachine Aurora 90
2. Enable LED Strip
3. Observe LED Strip tab is present after reboot

**Flight controller configuration**
```name Aurora90

feature -RX_PPM
feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature RX_SERIAL
feature MOTOR_STOP
beacon RX_LOST
beacon RX_SET
map RETA1234
serial 0 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
aux 0 0 0 1625 2100 0 0
aux 1 1 1 1900 2100 0 0
aux 2 2 1 900 1075 0 0
aux 3 4 1 1400 1600 0 0
aux 4 13 2 1900 2100 0 0
aux 5 15 2 900 1050 0 0
aux 6 19 4 900 1200 0 0
aux 7 28 0 1700 2100 0 0
set acc_trim_pitch = -8
set acc_trim_roll = -14
set baro_hardware = NONE
set min_check = 1000
set max_check = 2000
set rc_interp_ch = RP
set fpv_mix_degrees = 15
set serialrx_provider = SBUS
set airmode_start_throttle_percent = 35
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set bat_capacity = 800
set battery_meter = ESC
set small_angle = 0
set pid_process_denom = 2
set osd_vbat_pos = 2528
set osd_rssi_pos = 2496
set osd_remaining_time_estimate_pos = 2521
set osd_flymode_pos = 2554
set osd_throttle_pos = 2540
set osd_crosshairs_pos = 2253
set osd_disarmed_pos = 2315
profile 0

set throttle_boost = 0
set p_pitch = 58
set d_pitch = 35
set p_roll = 40
set i_roll = 40
set d_roll = 30
set p_yaw = 70
set d_yaw = 20
set f_yaw = 0
set i_level = 0
set d_level = 0
rateprofile 0```

**Setup / Versions**
 - Flight controller Stock Omnibus in Eachine Aurora90, bought from banggood 
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable]; All stock from Eachine Aurora 90 except camera, replaced to TX05
 - how are the different components wired up: tidily

**Additional context**
I really like my LED strip :(

archont, 2018-10-26:
A couple of flashes later, testing reveals this feature was last working on 3.4.1 and broke with 3.5.x

archont, 2018-10-26:
Duplicate of

https://github.com/betaflight/betaflight/issues/6966
https://github.com/betaflight/betaflight/issues/6802
https://github.com/betaflight/betaflight/issues/6773
https://github.com/betaflight/betaflight/issues/6594

Some information about this being by design should probably be presented to the user.

closed 2018-10-26 by archont

mikeller, 2018-10-28:
@archont: The targets that have had features removed are all listed in the release notes for 3.5, which are shown to the user when updating the firmware - we can only put it there, but we can't make you read it.

archont, 2018-10-28:
While this certainly belongs in the release notes section, the configurator, as the primary method of interacting with the FW, should be aware of this and either grey out/remove the option or present a pop-up. 

Down the line in a couple of years, expecting the user to read through few dozen notes is not optimal.

archont, 2018-10-28:
Rather than adding runtime FW feature supported queries, assuming official builds, would a static approach of parsing FW makefiles for feature defines and exporting them in a configurator-readable format be a solution?

mikeller, 2018-10-28:
@archont: Not really. Feature changes happen in patch releases, and the configurator is not designed to to tell these apart.
But this could probably be worked around, feel free to open a pull request.

cvetaevvitaliy, 2019-01-04:
Hi!
I compiled the latest firmware from the source with LED_STRIP, but without the SD Card, since there is no free space in FLASH [(STM32F3)
[https://github.com/cvetaevvitaliy/BetaFlight-OMNIBUS-F3---LED_STRIP-](https://github.com/cvetaevvitaliy/BetaFlight-OMNIBUS-F3---LED_STRIP-)