betaflight/betaflight #6991: 3d mode needs flip over after crash motor direction reversing or direction choice tick box

bobdogfpv, 2018-10-27:
**Describe the bug**
A clear and concise description of what the bug is.
in 3d mode enabled turtle mode spins motors wrong direction

**To Reproduce**
Steps to reproduce the behavior:
enable 3d mode in configurator  arm quad ,activate turtle ,props now wrong direction

**Expected behavior**
A clear and concise description of what you expected to happen.
props should spin as before and flip over after crash

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

mikeller, 2018-10-28:
@bobdogfpv:

> enable 3d mode in configurator arm quad ,activate turtle 

What mode were your ESCs in when you tried this?

bobdogfpv, 2018-10-31:
Bi directional esc and I fly props into sides to avoid camera water splash.
There is no option to reverse turtle motor direction

bobdogfpv, 2018-10-31:
> @bobdogfpv <https://github.com/bobdogfpv>: Would you mind sharing a CLI
> diff, this could potentially help in diagnosing the cause of the problem.
> (Maybe next time share a diff right from the start, like you were asked
> when opening the issue, most of the time they can help to rule out
> configuration issues.)
>
> Also, what type of ESC are you using (again, providing information up
> front can speed up things), and are they set to 'bidirectional' or
> 'bidirectional inverted'?

Yes certainly
Do you need cli from each quad?
They are all built with different hardware but obviously they all have the
same problem.
And bi directional and bidirectional reversed obviously suited to match
motor wiring

mikeller, 2018-11-01:
@bobdogfpv: Just one is enough to start with, thanks.

bobdogfpv, 2018-11-01:
here you go

diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.5.1 Sep  8 2018 / 05:40:39 (d9fb5ca13) MSP API: 1.40

board_name OMNIBUSF4SD
manufacturer_id 

# name
name frog yellow

# resources
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature INFLIGHT_ACC_CAL
feature RX_SERIAL
feature TELEMETRY
feature 3D
feature RSSI_ADC
feature CHANNEL_FORWARDING
feature AIRMODE

# beeper
beeper -RX_LOST
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 2 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led
led 0 6,15::CO:2
led 1 7,15::COW:0
led 2 8,15::COW:0
led 3 9,15::CO:0

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 3 1700 2100 0 0
aux 2 2 1 1700 2100 0 0
aux 3 13 1 1300 1700 0 0
aux 4 29 2 900 1700 0 0
aux 5 35 2 1300 1700 0 0
aux 6 39 4 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1900

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_calibration = -21,13,-8
set mag_hardware = NONE
set baro_hardware = NONE
set baro_noise_lpf = 0
set baro_cf_vel = 0
set baro_cf_alt = 0
set min_check = 1000
set max_check = 2000
set rssi_scale = 30
set rc_interp_ch = RP
set fpv_mix_degrees = 40
set serialrx_provider = SBUS
set blackbox_device = SERIAL
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set align_board_pitch = 180
set vbat_min_cell_voltage = 30
set vbat_scale = 105
set yaw_motors_reversed = ON
set 3d_deadband_low = 1450
set 3d_deadband_high = 1550
set 3d_neutral = 1500
set 3d_deadband_throttle = 100
set servo_lowpass_hz = 400
set small_angle = 180
set deadband = 32
set yaw_deadband = 50
set pid_process_denom = 2
set sdcard_dma = ON
set osd_units = IMPERIAL
set osd_vbat_pos = 2540
set osd_rssi_pos = 2512
set osd_tim_1_pos = 2528
set osd_tim_2_pos = 2546
set osd_flymode_pos = 364
set osd_throttle_pos = 129
set osd_vtx_channel_pos = 2504
set osd_crosshairs_pos = 2253
set osd_ah_pos = 200
set osd_current_pos = 97
set osd_mah_drawn_pos = 355
set osd_craft_name_pos = 392
set osd_gps_speed_pos = 66
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 386
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 161
set osd_pid_roll_pos = 322
set osd_pid_pitch_pos = 354
set osd_pid_yaw_pos = 386
set osd_debug_pos = 1
set osd_power_pos = 47
set osd_pidrate_profile_pos = 387
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2534
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 354
set osd_disarmed_pos = 2281
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 403
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_freq = 5880
set dashboard_i2c_bus = 1

# profile
profile 0

set dterm_lowpass_type = BIQUAD
set dterm_notch_hz = 260
set anti_gravity_threshold = 350
set anti_gravity_gain = 1000
set iterm_windup = 50
set throttle_boost = 0
set i_pitch = 20
set d_pitch = 20
set f_pitch = 50
set p_roll = 50
set i_roll = 20
set d_roll = 20
set f_roll = 50
set p_yaw = 30
set i_yaw = 40
set d_yaw = 20
set f_yaw = 0
set d_level = 100

# rateprofile
rateprofile 2

#

mikeller, 2018-11-01:
Thanks. Looks like this is a bug indeed, and `yaw_motors_reversed` isn't respected when the 3D feature is on. Will look into a fix.

mikeller, 2018-11-01:
Turns out the solution was a bit more complicated, and required changing the behaviour for 3D with / without `yaw_motors_reversed`, probably because crash flip with 3D on was never a thing before.

bobdogfpv, 2018-11-02:
That's fantastic news, let me know when it's available
Also some advise please, occasionally in 3d on fast throttle trick moves
like tic toc LOS it desyncs and falls out of sky but re-arms OK and flies
again.  what settings would help me stop this? Is idle speed a factor? Does
esc soft start have any effect?
Many thanks bob
Bobdogfpv

PR #7026 (2018-11-07): Fixed crash flip with 3D enabled.
Fixes #6991.
closed 2018-11-07 by mikeller