betaflight/betaflight #6993: HoTT telemetry using hardware 1-wire mode (only uart TX pin)

mmabln, 2018-10-27:
I have a question about the Betaflight / HoTT telemetry implementation / possible enhancement request:

In the documentation (and also source code) about HoTT telemetry, it is stated that you have to connect TX/RX pins from FC (using a diode towards the TX pin, making it kind of "open drain" when nothing is transmitted).

Modern controllers, such as STM32F4xx, have already build in support for 1-wire half-duplex serial transmission, so externally connecting TX and RX should not be necessary, only the TX pin can be used in bi-directional mode.

This hardware 1-wire mode is already used in the Betaflight SmartAudio implementation. Please check if HoTT telemetry connection could be simplified in the same way.

jflyper, 2018-10-28:
I think `tlm_halfduplex = ON` will do the job. If it doesn't, it is not some arbitral person who "checks" it, but someone with a "HoTT"; I believe none of the devs have Graupner system.

mikeller, 2018-10-28:
@mmabln: This is actually specifically listed in the Betaflight documentation: https://github.com/Scavanger/betaflight/blob/b907e584d1834b3c6ce769acdf85155821b109c1/docs/Telemetry.md#new-way

mmabln, 2018-10-28:
Thanks much for pointing me to that, I didn't see it at a first glance. So everything is fine, thanks!

closed 2018-10-28 by mmabln