betaflight/betaflight #6995: Flip Over After Crash - Safety Issue

timmyknuckles, 2018-10-28:
In "flip over after crash" mode there is currently no visual or audible feedback (no beeper installed) to warn that the mode is active. So during testing it is possible to be unaware and bump your stick, creating a potentially dangerous situation, especially if you've picked up the quadcopter and your transmitter is dangling from your neck.

I'm suggesting that "Flip over after crash" should automatically cancel after a set number of seconds of zero stick input. Perhaps around 3-5 seconds. 

I've considered other auto-cancelling possibilities where the software recognizes unexpected movement (e.g. picking up the quadcopter). However this may prove difficult given similar instances of movement -  for example where a quad is stuck in a tree on a windy day.

Another consideration was having the ESC's "beep" when the mode is active. However I was given this feedback by another user:
"Depending on the ESC version it can take up to 1 second after the beep command before the ESC will respond to any further commands. That would be particularly painful when trying to flip over under pressure in a race."

mikeller, 2018-10-28:
@timmyknuckles: Again, I do not see where this 'layer of safety' would be in what you are suggesting. Maybe you want to explain by making an example of how this should be helpful?

timmyknuckles, 2018-10-28:
@mikeller 
1. User is experimenting with crash-flip feature, learning how it functions. Line of sight only.
2. User becomes distracted momentarily and forgets the quad is effectively armed. All other hours 
 of experience flying for this pilot tells them if the props aren't spinning, the quad is disarmed. 
3. The pilot is wrong. They lean down to disconnect battery. Inadvertently bumping stick on transmitter. Chaos ensues.

mikeller, 2018-10-28:
@timmyknuckles:

1. No difference there - the pilot has to learn how it works either with or without your proposed change.
2. If a user has a habit of manipulating a craft without making sure it is disarmed this is a recipe for disaster, with or without crash flip. Reinforcing this bad habit by pretending to make it acceptable will only make this worse.
3. 'Being wrong' on disarming a craft is simply not acceptable. Trying to make it 'acceptable' will only encourage users to pick up bad habits.

timman2er, 2018-10-28:
There are so many ways to screw up in this way and it certainly needs to be consistent. Also this will lead to quite unexpected behavior for a pilot which is much worse.

If I'm not wrong taz will trigger if a drone hits someone/something so the damage should not be big.

Mechotronic, 2018-10-30:
Turn on the OSD element that shows the mode. "CRASH FLIP" will be on the screen when the mode is active.

leocb, 2018-10-30:
Is there an LED state for crash flip? If not we could add it at least, this safety concern the OP is talking about is while flying LOS, so the OSD does nothing in this case

mikeller, 2018-10-30:
@leocb: That is actually a good idea - I guess we could change the 'warning' overlay to blink when crash flip is active.

timmyknuckles, 2018-10-31:
Rapid flashing of LEDs would be good for sure. 

Is this achievable for the flight controller LEDs or only for auxiliary LEDs?

I think there should at least be a warning note in the modes tab in configurator.

For example:

“There will no feedback confirming that FLIP OVER AFTER CRASH mode is ARMED, unless a beeper has been installed.”

timman2er, 2018-11-01:
I think a universal warning LED could be configured that is turned on is these cases.

PR #7028 (2018-11-04): Added 'crash flip active' to the list of warnings to be shown on LED_STRIP.
Also unifies the crash flip warning to be on when disarmed everywhere.
Fixes #6995.
closed 2018-11-04 by mikeller