betaflight/betaflight #7001: Omnibus F3 BF 3.51 remapping issues

Sixks, 2018-10-28:
Hi

The LED STRIP function is no longer in v 3.51 for Omnibus F3
I read here that it was possible to remap in the cli.
With version 3.4.1 led strip was assigned in A08.
In version 3.5.2 it is the transponder.

  resource TRANSPONDER 1 none
  feature LED_STRIP
  resource LED_STRIP A08
  save

I tried this in 3.5.1 without success, did I forget something?

Thanks for your help

jflyper, 2018-10-28:
@Sixks As the LED strip feature itself has been dropped, assigning a pin for the feature does not have any effect. You have to build your own firmware.

Sixks, 2018-10-28:
Ok, thank you vor your reply.

mikeller, 2018-10-28:
Duplicates #6594.

closed 2018-10-28 by mikeller

cvetaevvitaliy, 2019-01-04:
Hi!
I compiled the latest firmware from the source with LED_STRIP, but without the SD Card, since there is no free space in FLASH [(STM32F3)
[https://github.com/cvetaevvitaliy/BetaFlight-OMNIBUS-F3---LED_STRIP-/blob/master/betaflight_4.0.0_OMNIBUS.hex](url)