betaflight/betaflight #7004: Matek F722 Battery Voltage Reporting

jcutcher01, 2018-10-28:
Flashed BF 4.0 #1199 on a Matek F722 FC and a Flip32 Omnibus FC. Last known good flights with no issues were on 3.5.0 RC2. On #1199 both fly great and I was interested in seeing how the dynamic LFP worked. The Matek FC reports the correct vbat but seems to “freeze” or hold its value when I am flying. This is observed on my radio via telemetry and on my OSD. Only when I land does it update to the correct voltage level. The omnibus is working without a problem and its vbat is updated fast and constantly as expected.

jcutcher01, 2018-10-29:
I confirmed with a hex build off of #1197 from @fujin that VBAT works on my Matek. #1199 build is not working.

fujin, 2018-10-29:
There is for sure something weird going on, although I've not tracked down what yet. I can't find anything in 6aa3b78629b1bb62f268c5eef52d3bd11ddcda80 (which triggered build 1199) that would cause any interaction.

[debug-mode-battery-logs.zip](https://github.com/betaflight/betaflight/files/2523713/debug-mode-battery-logs.zip)

In my case it looked like the battery voltage is higher than expected and doesn't respond to e.g. pulsing the throttle. It seems to get stuck at a particular value.

mikeller, 2018-10-29:
Just taking potshots here, but could it be that we're looking at a DMA conflict here that started with #6958?

jflyper, 2018-10-29:
It's easy to find out; just do CLI `dma` to find out if any TX DMA is activated and overriding ADC. I thought no target is using UART DMA at this moment, but might have overlooked.

fujin, 2018-10-30:
```
Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

jflyper, 2018-10-30:
@fujin No dshot?

fujin, 2018-10-30:
@jflyper that was with a minimal reproduction config on the board. (8/2/2 oneshot125). with dshot the expected DMA is visible.

sorry :-)

```
DMA2 Stream 1: TIMUP 8
```

fujin, 2018-10-30:
diff. pretty basic, just enough to arm/hover with the vbat related widget OSD.
```

# version
# Betaflight / MATEKF722 (MKF7) 4.0.0 Oct 29 2018 / 17:01:44 (9e1cf2add) MSP API: 1.41

board_name MATEKF722
manufacturer_id 

# name
name FUJIN F1

# resources
resource MOTOR 1 C07
resource MOTOR 2 C06

# mixer
mixer QUADX1234

# servo

# servo mix

# feature
feature AIRMODE
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RETA1234

# serial
serial 0 1024 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1950 2050 0 0
aux 1 13 3 1950 2050 0 0
aux 2 26 2 1450 1550 0 0
aux 3 35 3 1450 1550 0 0
aux 4 0 1 1950 2050 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set rssi_channel = 12
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 44
set vbat_warning_cell_voltage = 34
set pid_process_denom = 1
set osd_vbat_pos = 2251
set osd_current_pos = 2283
set osd_mah_drawn_pos = 2347
set osd_debug_pos = 2402
set osd_power_pos = 2315
set debug_mode = BATTERY
set vtx_band = 5
set vtx_channel = 2
set vtx_freq = 5695
set vcd_video_system = NTSC

# profile
profile 0

# rateprofile
rateprofile 0
```

fujin, 2018-10-31:
I [flashed and tested a branch yesterday](https://www.facebook.com/aj.christensen/posts/1073621419478451) based on master + #6992 + #7003 + akk vtx fix) and battery voltage was behaving normally (see vid). Also have logs. We need to find a 100% reproducible case. I have asked a few others with MATEK F722-STD to try (@McGiverGim).

McGiverGim, 2018-11-01:
I have flashed latest master today and the voltage was ok in my Matek F722... I've flown 6 packs.

jcutcher01, 2018-11-01:
I will try it out sometime today.

jcutcher01, 2018-11-02:
I confirmed that #1212 works and reports vbat during flight.

closed 2018-12-09 by stale[bot]