betaflight/betaflight #7005: GPS Rescue sometimes dropping just after activation of Stage2

TonyBlit, 2018-10-28:
Hi, I've been testing GPS Rescue with this fix https://github.com/betaflight/betaflight/pull/6979

The fix works well, however, in some cases the quad has suddenly disarmed just after the activation of Stage2. In the three failed attempts that I've recorded today, this disarm has happened between 300ms and 400ms after the activation of GPS Rescue.

I have the Sanity Check set to ON, so maybe the sudden activation of the motors is detected by the crash detection algorithm and hence the disarm? Is there any way to check the reason for the GPS Rescue to have stopped? Note that I'm testing with Stage1 set to AUTO instead of HOLD, as I'm trying to test the worst conditions. Maybe I'm doing a dive when I get a real RX Loss, so GPS Rescue should be able to gracefully recover this situation.

As a bonus, a feature request related to this: it may be desirable to soften the transition from normal flight to GPS Rescue, so the briefer the RX Loss, the less disturbed the flight.

Successful recover with GPS Rescue, Air Mode disabled:
[image]

Disarm after GPS Rescue, Air Mode enabled:
[image]

etracer65, 2018-10-29:
Can you please supply your `diff` as requested in the issue template?

Also please provide the log files - screen shots are of limited value.  From the limited information in the  pictures I don't see any indication of a disarm.

etracer65, 2018-10-29:
Based on the fact that you have a failsafe mode switch set up I'm assuming you're triggering the failsafe that way and not though an actual signal loss?

If you're not more than 100m away from your home location when GPS Rescue is triggered via failsafe then it is designed to disarm for safety. I think that's what you're seeing.

etracer65, 2018-10-29:
Do you have any DVR footage showing the OSD when the problem occurs? Also can you try and capture a blackbox log with `debug_mode = RTH`?

etracer65, 2018-10-29:
Just a few seconds before and the actual event. Maybe include some successful ones for comparison.

TonyBlit, 2018-10-29:
Cut1: BB 1/7 at 1:15: GPS Rescue suddenly disarms
Cut2: BB 5/7 at 0:39: GPS Rescue works well (2:46 previously flown in another run)
Cut3: BB 5/7 at 1:21: GPS Rescue suddenly disarms (2:46 previously flown in another run)
Cut4: BB 7/7 at 0:29: GPS Rescue works well (0:40 previously flown in another run)
Cut5: BB 7/7 at 0:43: GPS Rescue suddenly disarms (0:40 previously flown in another run)

[Cuts.zip](https://github.com/betaflight/betaflight/files/2525529/Cuts.zip)

TonyBlit, 2018-10-30:
I've spotted something:

     // Stalled movement detection
    static int8_t gsI = 0;

I think this variable should be reset to zero at RescueStart(), otherwise subsequent activations of GPS Rescue will start accumulating on old data, potentially allowing for an unexpected Rescue Abort (Drop). Think of a quad hovering or moving at very low speed while having multiple short failsafes. This code will end dropping the quad.

I'll also do previousTimeUs = currentTimeUs; at RescueStart(), as it feels a bit sloppy to be accumulating onto gsl while GPS Rescue has not even started controlling the quad. At this point in my tests I'm hovering or barely moving, so it's pointless to accumulate the initial state as stalled movement.

Also there are a host of variables that would be nice if they're reset each time GPS Rescue is activated, just for the sake of determinism:

    // Minimum sat detection
    static int8_t msI = 0;

    static float previousSpeedError = 0;
    static int16_t speedIntegral = 0;
    static float previousAltitudeError = 0;
    static int16_t altitudeIntegral = 0;

I've already setup a development environment here, so tomorrow I can create a PR if needed.

mikeller, 2018-10-30:
Good find, @TonyBlit!

closed 2018-11-01 by mikeller

TonyBlit, 2018-11-01:
Ok, today I've been able to test with debug mode and I can confirm the crash detection is being triggered by the own operation of GPS Rescue.

Details:

- The debug code had a few flaws, so I've been testing with these changes applied: https://github.com/betaflight/betaflight/pull/7022
- In order to get the GPS Rescue failure code in the blackbox, I've had to set SANITY_CHECK=OFF. Otherwise, once a problem is found, the Sanity code immediately disarms, so the failure code is not stored in the Blackbox. Thankfully enough, disabling Sanity Checks is programmed to keep doing the checks and updating the debug variables on the blackbox, just it doesn't disarm the quad on most failures, which is very convenient for testing.
- I've forced a lot of failsafes while flying in very different scenarios.
- The quad has recovered in all cases, starting the GPS Rescue procedure, including recovering from falling and being completely upside-down
- When analyzing the blackbox, those difficult recoveries were triggering a CrashDetection failure. If sanity checks would have been enabled, this would have dropped the quad, as I initially reported in this issue.

Conclusions:

No matter how you setup Stage1 (Air mode, HOLD, whatever), there's no way to prevent GPS Rescue to fail due to a false Crash Detection *IF* Sanity checks are enabled (which is counter-intuitive). If you have a real RX Loss in the middle of a Dive or a Power Loop, the Sanity Check code is probably going to drop the quad due to a false positive. Just check the videos I posted, the start of the recovery can involve very sharp maneuvers not possible with normal input, so it all makes sense.

Next steps:

In order for the Sanity Checks to work properly, a new internal state has to be included in GPS Rescue to account for the initial stabilization phase, where crash detection should be ignored. I'll start working on a PR for this. Hope this is the last one. :-P

etracer65, 2018-11-01:
You're thinking about it a little bit backward. It's not that GPS Rescue should handle or ignore crash recovery, it's that crash recovery shouldn't activate while GPS Rescue is enabled. GPS Rescue already enters angle mode as part of the process - the same as crash recovery uses to regain control.

And if you're in the middle of crashing and get a failsafe, well there's not much we can do.

TonyBlit, 2018-11-01:
I'm talking about this:

    // Check if crash recovery mode is active, disarm if so.
    if (crashRecoveryModeActive()) {
        rescueState.failure = RESCUE_CRASH_DETECTED;
    }

This should not be checked during the first 0.5s or so after engaging GPS Rescue, because when the quad is leveled (see pid.c), this can make the quad to turn very sharply (ie: if it was upside down)

The amount of time the initial leveling takes is something that I still have to figure out.

etracer65, 2018-11-01:
I had forgotten that GPS Rescue used crash recovery to disarm if it crashes into something. Probably have to prevent crash recovery from activating if GPS Rescue has just been triggered within a short period of time. When GPS Rescue is activated it switches into level mode so that should take care of the initial leveling.

But the case where crash recovery is already active and then a failsafe occurs can't really be handled. But that's an unlikely edge case and GPS Rescue disarming in that case is the best option.

etracer65, 2018-11-01:
Overall however this adds a level of risk since even disabling crash recovery for a short period then prevents the safety aspect of disarming the quad if it runs into something immediately after GPS Rescue initiates.

One could argue that being upside down is not a "recoverable" state for GPS Rescue.

TonyBlit, 2018-11-02:
I think there are two unrelated things in your explanation:
The autoleveling code can recover the quad from being upside down (I strongly relied on this when learning acro...)
The crash detection feature will do its best to level the quad after a crash, but doesn't care about failsafes or disarming.
GPS Rescue cleverly uses both things: in the initial stage it takes advantage of autoleveling code for recovering the quad. And when flying home, it uses the crash detection code as part of the sanity checks.
The mistake is to to use the crash detection code for all the duration of the GPS Rescue operation. Because a side effect of using auto-leveling for recovering an upsidedown quad is it will trigger the crash detection code. I've been seeing this on DVR for the past weeks while scratching my head, but yesterday I got the proof this is not an intended behavior.

Worst case: you crash into something, but are still in the air, and then have an RX Loss. GPS Rescue will trigger the auto-leveling. Let's say the auto-level triggers the crash detector yet another time (or still), no problem, that simply has to be ignored and leave auto-level to keep doing its thing. Once you're leveled (this part of code is currently missing), GPS Rescue will fly you home monitoring only SUBSEQUENT crashes, that is the intended part for the sanity checks. This is a completely recoverable scenario AFAIK.

Of course, there should be a timeout for the leveling phase. ie: if the quad has not leveled in 1s or 2s, just drop it.

TonyBlit, 2018-11-02:
Quick prototype: https://github.com/TonyBlit/betaflight/commit/ff99e7f05838ea6f3a6f61bcc197886f2a0ab773

Hopefully tomorrow I'll be able to test it on the field. After that, I'm unsure on how to proceed with the tuning of the magic numbers I had to include. Asking other people on the slack channel to help test it?

etracer65, 2018-11-02:
The problem is that crash recovery *also* tries to self-level the quad and takes over control to do so - that's its "recovery" part. While crash recovery is active GPS Rescue has no control over the quad. Crash recovery needs to be aware of the GPS Rescue state and behave appropriately. Remember that crash recovery may not even be enabled and is actually off by default so to code around it in GPS Rescue is the wrong approach and the above is overly complicated.

TonyBlit, 2018-11-02:
Yes, GPS Rescue activating the Crash Detection by itself combined with the normal Crash Detection behavior is quite a bit messy. I'll write down all the combinations between Crash Recovery activated by the config or GPS Rescue, and handle Recovery before and after a failsafe, Sanity Check enabled or disabled and being in the leveling phase of the GPS Rescue. This way we'll have a complete view of the problem. Otherwise it's easy to overlook some scenarios.

TonyBlit, 2018-11-02:
Ok, I think it will be fine going with this condition for both detect and handle:

  if ((crash_recovery || GPS_Rescue_AlreadyLeveled) && !GPS_Rescue_Leveling) {
    do something
  }

This will allow for:

- Detection&Handling without GPS Rescue work as usual
- Detection&Handling when GPS Rescue is activated are disabled during leveling
- Detection&Handling after GPS Rescue leveling will depend on sanity check. If it's enabled, the quad will be dropped. If it's disabled, normal handling will occur. After normal handling happens, GPS Rescue will continue doing its thing.

Another topic is why pid.c has to know what stage is GPS Rescue in, but that's another discussion. :-)

If the above looks good, I'll hack it into the prototype this afternoon.

TonyBlit, 2018-11-02:
Even better, I can replace FLIGHT_MODE(GPS_RESCUE_MODE) with a function GPSRescueForceCrashDetectionAndHandling, and hide those conditions inside gps_rescue.c

etracer65, 2018-11-02:
Arrr! Not going to be that simple. https://github.com/betaflight/betaflight/pull/7033#issuecomment-435386877

Complicated by the async nature of how GPS Rescue transitions through its states in the relatively slow GPS task. Need to look at it deeper.

TonyBlit, 2018-11-02:
I've updated the crash detection code, and ifdef'ed with GPS_RESCUE:
https://github.com/TonyBlit/betaflight/commit/0eb7379b8e777427bcf67cc2a32c57dfb0ee2eca

@etracer65 I know this solution is more verbose but it's worth noting that: it ends the leveling phase as soon as the quad is leveled, so it disables the crash detection as little as possible, not for a fixed amount of time (that can be potentially longer than 1sec if tests confirm so). Also, having a separate internal state allows for different parameters in the leveling phase (ie: delay the speed controller, and altitude controller and just focus on stabilizing the quad also taking advantage of the throttle hover value). This is something that I've not tweaked (yet).

etracer65, 2018-11-02:
I still think this is far too complicated. We don't actually need to wait for the quad to be "level" as it's the initial reaction to correct the orientation that will trigger crash recovery. Also fine-tuning the settings to ensure that the "correct" amount of level is attained before timeout and GPS Rescue proceeding is problematic.

I closed the previous PR I created and went a different direction. In this case it's not GPS Rescue specific but rather related to entering a self-level mode. This fixes the problem for GPS Rescue and improves behavior for people using self-level modes as they're now unlikely to cause a false crash recovery trigger when switching to self-level (like in a panic to prevent being out of control).

https://github.com/betaflight/betaflight/pull/7034

TonyBlit, 2018-11-02:
Good job! Now that your code is totally self-contained I agree this is the best approach hands down. However, I still think my idea of detecting when the leveling has finished fits perfectly well in this approach. The timeout should be there only as a last resort. pidlevel knows exactly how far it is the quad from where the sticks/rescue are asking for. This way, when activating angle/rescue on an already leveled quad, the crash detector won't be disabled for 1 full second.

TonyBlit, 2018-11-02:
Actually, this is how handleCrashRecovery is implemented to know that it has already finished:

                // check aircraft nearly level
                if (ABS(attitude.raw[FD_ROLL] - angleTrim->raw[FD_ROLL]) < crashRecoveryAngleDeciDegrees
                   && ABS(attitude.raw[FD_PITCH] - angleTrim->raw[FD_PITCH]) < crashRecoveryAngleDeciDegrees) {

So it would be almost identical code to know when the switch to level mode has also finished.

TonyBlit, 2018-11-03:
Today I've been able to test this: https://github.com/betaflight/betaflight/pull/7034

It *almost* works perfectly :-)

So I've flashed those changes (btw, the timer2 is now broken, it doesn't show up on the osd, this didn't happen 2 days ago). My intention today was to test extreme recoveries, so I preferred to disable the sanity checks and forced debug_mode=RTH. Just by examining the BB post flight would be enough to know if sanity check would have dropped the quad. I've tried whatever combination I could think: being upside down with air mode, falling without air mode, flying in angle both with air mode enabled and disabled.

And right now I've just finished examining the BB. And found that just one of the combinations triggered the crash detector: flying in angle without air mode. When you cut the throttle or get an RXLoss with default parameters (0.4s stage1 with throttle LOW), the quad falls without control for 0.6s. Then, upon activation of GPS Rescue, as the quad is already in angle mode, the crash detector is not guarded:

    const bool levelModeActive = FLIGHT_MODE(ANGLE_MODE) || FLIGHT_MODE(HORIZON_MODE) || FLIGHT_MODE(GPS_RESCUE_MODE);
    static timeUs_t levelModeStartTimeUs = 0;

    // keep track of when we entered a self-level mode so that we can
    // add a guard time before crash recovery can activate
    if (levelModeActive) {
        if (levelModeStartTimeUs == 0) {
            levelModeStartTimeUs = currentTimeUs;
        }
    } else {
        levelModeStartTimeUs = 0;
    }

I think there should be two separate timers, one for levelMode and one for rescueMode. And then use both for guarding detectAndSetCrashRecovery()

Further thinking into it, I'm sure I'm able to trigger a false Crash Detection in Angle with Air Mode. It's actually what happened in some of the cases I initially reported in this issue. When falling for 0.6s with Air mode, even being already leveled, the GPS Rescue startup introduces so much change at once that it can certainly trigger the Crash Detector. So I think the problem I'm describing today applies to Angle mode, regardless of Air mode.

I'm sorry to be so picky, but I think the feature has to be 100% rock solid for the users to trust it. Of course there are hardware factors into it, but at least the software part should be completely deterministic.

etracer65, 2018-11-03:
While I see your point, at the same time we're getting pretty deep down in the weeds for scenarios that the pilot shouldn't really allow to happen. The fundamental flaw here is that the pilot's config allows the motors to stop during a stage 1 failsafe. At that point the quad is uncontrolled and really should just crash. In other words coding around an invalid config was not the goal.

The existing changes address a valid flight case where the pilot is in controlled flight but might be in an orientation where the transition to self-level may be aggressive enough to trigger crash recovery. Just because it happens to also to work in cases where the quad is in uncontrolled flight with the motors stopped is secondary.

We will discuss this internally to see if it makes sense to account for the described scenario. The downside is that it would *always* add the crash recovery delay when entering GPS Rescue - not just when entering it from another self-level mode (which should be under control).

Also just to be clear, this feature will never be "100% solid". GPS Rescue is a best-effort based on the available tools. There is no aspect of trust or guarantees implied.  Besides, even if it fails 1 time out of 10, that's still 9 times it did better than the alternative which was a failsafe drop.

TonyBlit, 2018-11-03:
Sorry, I didn't express myself correctly. I completely agree with you, and the crazy things I've tested today were just to explore the boundaries of the recovery, this is not what I expect to be recovered. My expectations are still the same since day one: with default settings, I expect the quad not to drop when normally flying, which is what happened to me (by chance).
So my point is, when you activate GPS Rescue on failsafe, you don't expect the sanity checks to be fooled by the start of GPS Rescue itself, specially when using default Stage1 settings. You can argue that messing with failsafe parameters requires you to understand them (and that's true), but maybe it's worth thinking what to do to prevent other people to get into this trap. I have to admit I wasn't totally sure of the implications of my stage1 settings the very first time I activated GPS Rescue on failsafe. And I'm sure this will happen to other people.
So maybe other Stage1 settings can be suggested in the wiki (I'll be happy to document this), or enforced by the GPS Rescue itself, but I think just adding the additional timeout I mentioned previously should cover everything I would expect.
Hope my point is clear now, sorry.

etracer65, 2018-11-04:
The fixes including the additional resetting of the crash recovery detection guard time when entering GPS Rescue (instead of just when self-level state changes) have been merged.

TonyBlit, 2018-11-04:
Today I've been able to test your last changes @etracer65, the sanity checks during the activation of gps rescue are now rock solid!!! :-)

Thanks for your help and patience.