betaflight/betaflight #7009: Why most `xxx_INVERTED` are actually not inverted at all?

jflyper, 2018-10-29:
I've always wondered why there is at least a couple of things called `xxx_INVERTED` which are not inverted at all; they are all active high logic.

- Beeper
- Buttons

mikeller, 2018-10-29:
I guess the 'default' is pull down for these...

jflyper, 2018-10-29:
Ah... for buttons, it looks like the default is active low.

What do you think about the beeper? It's really confusing when mixed with OD...

mikeller, 2018-10-29:
@jflyper:

I think 'inverted' is confusing, since it comes with the implied assumption that something is the default.

I also think that now is the time (if any) to change the terminology used for the configuration of pins, as there are not many of them around, and they aren't often used. Once we've moved to universal targets there will be more, and they will be used by many targets.

Do you have a proposal for how to name the different modes?

jflyper, 2018-10-29:
Postfixing with `_ACTIVE_HIGH` is one way.
`BUTTON_INVERTED` (input) -> `BUTTON_ACTIVE_LOW`
`LED_INVERTED` (output) -> `LED_ACTIVE_LOW`

mikeller, 2018-10-29:
@jflyper: The suggestions sound good to me.

As for the lists of pins that are currently done as parameter arrays, I think there will be more lists of configurable items with universal targets (other pins, UARTS, buses). Expressing their config as array parameters will make modifying them very awkward. We should probably look into adding a facility to support parameter arrays:

```
led_polarity[1] = [ACTIVE_LOW | ACTIVE_HIGH]
```

or even

```
led[1].polarity = [ACTIVE_LOW | ACTIVE_HIGH]
```

closed 2026-09-26 by blckmn