betaflight/betaflight #7012: F7 3.5.2 Runcam Split V2 and GPS UART not working

pediegoivan, 2018-10-29:
**Describe the bug**
I have tried the Runcam Split V2 in two different UARTs and the GPS and neither work.
I even tried gpspassthrough with nothing coming out through the CLI afterwards.
I double and triple checked all the connections and I am 100% sure that everything is correctly wired.

**Expected behavior**
I was expecting to start stop recording and navigate through Runcam osd with my aux switches
And also I was expecting to see the GPS icon light up when I finished the configuration of the GPS and port set up and reboot.

**Flight controller configuration**
```# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 3.5.2 Oct 16 2018 / 22:00:17 (a5e270fee) MSP API: 1.40

board_name KAKUTEF7
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 3 2 115200 115200 0 115200
serial 6 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 1 1700 2100 0 0
aux 3 35 1 1300 1700 0 0
aux 4 47 1 1300 1700 0 0

# adjrange

# rxrange
rxrange 0 1007 1991
rxrange 1 996 2008
rxrange 2 995 2005
rxrange 3 995 2011

# vtx

# rxfail

# master
set acc_trim_pitch = -20
set acc_trim_roll = -32
set min_check = 1010
set max_check = 1990
set rssi_channel = 15
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set current_meter = ESC
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 30
set gps_rescue_min_sats = 0
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 30
set osd_cap_alarm = 1550
set osd_tim2 = 1025
set osd_vbat_pos = 2210
set osd_rssi_pos = 2242
set osd_tim_2_pos = 2274
set osd_remaining_time_estimate_pos = 2282
set osd_current_pos = 2338
set osd_mah_drawn_pos = 2371
set osd_gps_speed_pos = 2230
set osd_gps_sats_pos = 2198
set osd_home_dir_pos = 2262
set osd_home_dist_pos = 2263
set osd_warnings_pos = 2379
set osd_avg_cell_voltage_pos = 2178
set vtx_band = 1
set vtx_channel = 3
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5825

# profile
profile 0

set dterm_notch_cutoff = 0

# rateprofile
rateprofile 0

set roll_rc_rate = 199
set pitch_rc_rate = 199
set yaw_rc_rate = 180
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 25
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 55
```

**Setup / Versions**
I ran into this issue with a bare new kakute F7 board (the one without PDB);
I am running Lua Scripts and SmartAudio to control an akk x2 ultimate without any issues
I have an R9 slim+ running F Port hooked up to the FC.

Fport to uart6
GPS to uart4
Runcam to uart3
akk x2 ultimate to uart1

McGiverGim, 2018-10-30:
Only to be sure, have you connected the TX pad of the FC to the RX pad of the GPS/Runcam and the RX pad of the FC to the TX pad of the GPS/Runcam? I have seen other users connecting TX to TX and RX to RX.

pediegoivan, 2018-10-30:
I have certainly done that and double checked

McGiverGim, 2018-10-30:
The Runcam Split has been generating problems and not working in the latest versions. A PR has been merged some weeks ago that in theory fixed the problem, but I didn't test it. But the GPS must work (a lot of people is using it now).

What GPS unit are you using?

pediegoivan, 2018-10-30:
TS100 Radiolink (https://www.amazon.com/Radiolink-Pixhawk-Controller-Quadcopter-Multirotor/dp/B07DPN8WNR/ref=mp_s_a_1_1?ie=UTF8&qid=1540893877&sr=8-1-spons&pi=AC_SX236_SY340_FMwebp_QL65&keywords=gps+ts100&psc=1)

McGiverGim, 2018-10-30:
I have it and it works (at least in my Matek F722).
- Have you tested it in another FC? To see if it can be a problem of the GPS unit.
- Have you tried to activate auto baud for the GPS in the configurator?

pediegoivan, 2018-10-30:
I tried auto baud and it auto selected done baud rate.
I haven't tried the gps and neither the runcam serial through a different FC. It's the only one I have, but it is a bar new FC.
And Fport and smart audio are still working.
I will try the gps with a ftdi adapter later

McGiverGim, 2018-10-30:
Another silly idea... have you connected the lipo while testing in the configurator? New FCs didn't power up the 5V outputs while connected by USB.

pediegoivan, 2018-10-30:
Yes! I did try that!
I connected the Compass of the TS100 and got it working like a charm through the SDA and SCL pads. My issue is only with the UARTs

SchonRichtig, 2018-11-02:
I also use the kakute f7 without any UART problems.

ESC telemetry from tekko32 4in1,  GPS,, R9 mini via F.Port, Tramp Protokoll - no problem

pediegoivan, 2018-11-02:
What bf version are you running? Are you using the kakute f7 with pdb? I must have a faulty board then. :(
Can you post your diff?

SchonRichtig, 2018-11-05:
> # version
# Betaflight / KAKUTEF7 (KTF7) 3.5.2 Oct 16 2018 / 22:00:17 (a5e270fee) MSP API: 1.40

board_name âÞ<2mg[Ùí¤<ù
manufacturer_id ìÇVù

# name
name Flosstyle6-V2

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 8192 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 3 1700 2100 0 0
aux 4 13 3 1300 1700 0 0
aux 5 28 4 900 1300 0 0
aux 6 35 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h
rxfail 5 s 1000
rxfail 14 s 1000
rxfail 15 s 1000

# master
set gyro_sync_denom = 2
set gyro_use_32khz = ON
set max_check = 1950
set rssi_channel = 15
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 25
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set failsafe_delay = 8
set failsafe_procedure = GPS-RESCUE
set current_meter = ESC
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_initial_alt = 65
set gps_rescue_descent_dist = 100
set gps_rescue_min_sats = 0
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set osd_vbat_pos = 2103
set osd_rssi_pos = 2157
set osd_tim_2_pos = 2487
set osd_current_pos = 2092
set osd_mah_drawn_pos = 2097
set osd_gps_speed_pos = 2088
set osd_gps_lon_pos = 2496
set osd_gps_lat_pos = 2513
set osd_gps_sats_pos = 2509
set osd_home_dir_pos = 2476
set osd_home_dist_pos = 2477
set osd_altitude_pos = 2464
set osd_avg_cell_voltage_pos = 2082
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set vtx_band = 2
set vtx_channel = 5
set vtx_power = 4
set vtx_freq = 5809

# profile
profile 0

set dterm_notch_cutoff = 0
set iterm_relax = RP
set i_pitch = 70
set i_roll = 65

# rateprofile
rateprofile 1

set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 120
>

Do you use the original antennas from the R9 System?
Change them to the new T Type Antennas an the Super 8 on transmitter side.
Or use TBS Antennas, or True-Mox.
The stock antennas are bad.

pediegoivan, 2018-11-05:
I will get one of those by december. I am stuck with the stock ones now. But, I don't get any telemetry lost from my taranis when it falls off the sky so I assume it is another UART related issue

pediegoivan, 2018-11-10:
Ok. I've swapped the F Port pad, I will give it a try.
I made the GPS work. It seems that the only documentation that I managed to get was wrong. I connected TX-TX and RX-RX to make it work. Radiolink TS100 square black box thing does not have even a pdf manual. only the Banggood product page.
I checked that my Runcam Wires and runcam serial side are working ok. I grabbed my FTDI board and send the serial commands with my laptop. However, this issue can be closed now, GPS is working, and there is some other issues dealing with Runcam Device protocol problems.
Thanks for all the help.

closed 2018-11-10 by pediegoivan