betaflight/betaflight #7024: LED_STRIPE feature leads to a 100% CPU load for OMNIBUSF4SD 3.5.2

wuhao19920619, 2018-11-01:
while I test 3.3.0 it works prefectly. But for 3.5.2 it will suddenly goes to 100% load when I turn on the feature LED_STRIPE. My FC is EXUAV F4 PRO, the pink one.

MY diff is:

`# DIFF

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 Mar  2 2018 / 03:56:35 (177472b4f) MSP API: 1.37

# name

# resources

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 5 64 115200 57600 0 115200

# led
led 0 10,5::LI:0
led 1 10,6::LI:0
led 2 10,7::LI:0
led 3 9,8::LI:0
led 4 8,8::LI:0
led 5 7,8::LI:0
led 6 6,8::LI:0
led 7 5,7::LI:0
led 8 5,6::LI:0
led 9 5,5::LI:0

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0
aux 1 1 1 900 1200 0
aux 2 2 1 1400 1600 0
aux 3 13 3 1800 2100 0
aux 4 15 3 900 1200 0
aux 5 19 2 1400 1600 0
aux 6 28 2 1800 2100 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1000
rxfail 6 s 2000

# master
set gyro_lowpass_hz = 35
set gyro_notch1_hz = 140
set gyro_notch1_cutoff = 80
set gyro_notch2_hz = 58
set gyro_notch2_cutoff = 50
set moron_threshold = 2
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1000
set max_check = 2000
set rssi_channel = 16
set serialrx_provider = SBUS
set airmode_start_throttle_percent = 20
set rx_min_usec = 986
set rx_max_usec = 2012
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 0
set failsafe_off_delay = 200
set failsafe_throttle = 1235
set failsafe_procedure = AUTO-LAND
set vbat_min_cell_voltage = 34
set vbat_warning_cell_voltage = 36
set small_angle = 180
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set osd_warnings = 63
set osd_vbat_pos = 2504
set osd_rssi_pos = 2305
set osd_tim_1_pos = 2103
set osd_tim_2_pos = 2081
set osd_throttle_pos = 2241
set osd_crosshairs = 2282
set osd_ah_sbar = 2282
set osd_ah_pos = 2282
set osd_current_pos = 2510
set osd_mah_drawn_pos = 2544
set osd_compass_bar_pos = 2090
set osd_altitude_pos = 269
set osd_avg_cell_voltage_pos = 2536
set osd_pit_ang_pos = 2263
set osd_rol_ang_pos = 2327
set osd_disarmed_pos = 2186
set osd_nheading_pos = 2126
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_tim_1 = ON

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass = 35
set dterm_notch_hz = 45
set dterm_notch_cutoff = 35
set yaw_lowpass = 35
set p_pitch = 36
set i_pitch = 72
set d_pitch = 21
set p_roll = 30
set i_roll = 60
set d_roll = 18
set p_yaw = 20
set i_yaw = 40
set d_level = 90
set level_limit = 75

# rateprofile
rateprofile 0

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# `

mikeller, 2018-11-01:
@wuhao19920619: 

> Betaflight / OMNIBUSF4SD (OBSD) 3.3.0 

You are reporting a problem with 3.5.2, but you supply a `diff` from 3.3.0. In order to get help with a problem with 3.5.2, please supply a `diff` for 3.5.2 that has the problem.

mikeller, 2018-11-01:
@wuhao19920619: Looks like your board has got a hardware issue. I just tried to reproduce with 3.5.2 and your original config, and the load is only 5%.

closed 2018-11-01 by mikeller

wuhao19920619, 2018-11-01:
@mikeller I tested 3.3.0 everything works fine. With 3.5.2 when I use all default setting the load is 5-6% , but when I turn on LED_STRIPE it suddenly goes to 25%. After all these setting the load is around 55% for 1~2 second after startup, then goes to 100%. 

Because with other version(all before 3.3. After 3.3.0 only 3.5.2 is tested) it works good. So I am not sure if this is really a hardware problem.