betaflight/betaflight #7031: Hexa and more Motorconfiguration

sknyetcom, 2018-11-02:
My Problem:
I am flying Hexa configuration. If i break one propeller, i am still able to fly. Thats great.
But if one propeller has lost some of their blades, or got loose, this motor will spin with 100% until it dies.

Is there an option to add an guard, if one of the motor is spining with a lot more power then the rest? I know on most quads it is difficult. But on a hex, there should be a certant level between motors on the same axes.

That would be great to have a feature which could be enable  to save my motors 

First step maybe compare the maximum PWM output to the next PWM output. If the difference for a certain time (default maybe 3 sec,  paramter) excceds a threshold, it should turn the motor off and displays the motor ID via osd.

I dont know, but maybe it makes sence, to turn allso the opposite or the motor next to it of, to get back to a quad configuartion is this case?

Any better suggestions are welcome

some video evidence

at 1:44 i stripped of the blades of the propeller 
at 1:54 the motor burn out

https://www.youtube.com/watch?v=5oIMuFpP-t0

jflyper, 2018-11-02:
I believe the output adjustment scheme doesn't work on most cases.

The only case it may work is the case when you lose a prop on a motor that is contributing to one of R or P (left or right for 6X, or front or rear for 6P). In this case, the motor with lost prop and an opposing motor could be shut down, falling back to a 4X topology. In all other cases, you will be burning two motors instead of one if you continue to fly like you do in the vid.

sknyetcom, 2018-11-02:
that are some good arguments, and it will not be easy to get an working solution for this. And Hex is not the mainstream i know.

But maybe how about only a small inidicator?
If the osd provides me an Warning if one motor spins at 100% for a certain time?
Than i have the information i need and i could decide by my own if i stop flying/disarm or something else?

And maybe is this information good for something else, who knows?

jflyper, 2018-11-02:
Over-rev warning. That would be a decent feature request.

ledvinap, 2018-11-02:
@jflyper : The configuration is symmetrical, disabling any pair of opposite motors will result in controllable quad configuration (with reduced control in direction of lost motors).
You can probably even use 5-motor configuration, but it may cause unexpected problems ..

jflyper, 2018-11-02:
Probably. But you can't win the race any more with cripped craft. So there isn't a point to keep flying.

However, I'm starting to get some interest in the degraded configuration now... (I'm partially convinced) I might do some experiment personally.

Btw, 5 motor configuration is inherently unstable because of the extra yaw torque, isn't it?

ledvinap, 2018-11-02:
It may be better to implement protection in ESC, for example by limiting maximum RPM. PIDs may handle disabled motor without modification.

> Is it worth the effort? That is my question now

Maybe by implementing generic motor protection - use some estimator for motor temperature and reduce power to keep temperature bounded. But that may fail in propeller lost situation (loses in motor may be considerably higher at given input command)

ledvinap, 2018-11-02:
5 motor configuration: from mathematical point of view the system should be controllable.

assuming 6X and lost leftmost CCW motor:
2 remaining CCW motors will speed up to control yaw, causing roll to left. This will be compensated by reducing power to rightmost (CW) motor. Eventually I-terms will adapt with rightmost motor almost disabled.

sknyetcom, 2018-11-02:
There are a lot of options to try keep that hexa copter with less motors in air. That would be awesome.

But at the moment, everytime a touch a gate, i worrie about my motor could burn out.

i think first we need something wich is able to detect that the props are gone. And then we can go one with more.

Test is easy,
Just replace on a 5" quad one prop with an 3" low pitch. Have the same effect like cut 2 blade from a 5" treeblade

TheTape, 2018-11-07:
It's called sanity checking and you need some of that too jf. Please. Your reasoning is utterly ridiculous. Can't win the race? Let's just trash the motor then! Back at the beginning... I dub you Jeff from now on.

1: It's pretty easy to compare relatively constant I states to a sudden change.

2: It's pretty easy to calculate the center of throttle (via throttle values required to stay balanced), calculate it's offset from average flight, and if it moved significantly, calculate the direction it moved, to detect a faulty motor opposite.

3: It's pretty easy to detect which specific motor throttle is much higher in ratio compared to the others average, and to shut it down slowly to ensure stability isn't significantly worsened from shutting down the wrong one, and if we did try the next most likely candidate.

5 motors works fine, how do you think he's keeps flying the thing?

I prefer motor shut-down warning rather than over rev warning, as that would be irrelevant to a motor that automatically shut down.

closed 2018-12-15 by stale[bot]