betaflight/betaflight #7042: OSD and OLED do not work simultaneously

ujjwalk450, 2018-11-04:
I am using BF 3.5.2 on a matek F405-CTR board. OSD was working fine until now but last week I added an OLED display (SSD1306), and OSD freezes up within 5 seconds of boot of the flight controller. OLED works fine, but OSD freezes up. Disabling the OLED in the configuration menu makes the OSD up and running again after reboot. Please help

etracer65, 2018-11-04:
I think this is normal as there can only be one display port active.

jflyper, 2018-11-04:
Nah, OLED should be working as dashboard until it is selected as an active display port for CMS.

jflyper, 2018-11-04:
I can look into it once I'm back at the bench on Wed. Are you in a hurry? @ujjwalk450

ujjwalk450, 2018-11-04:
no, i am not in so much hurry, i was just fiddling around and came to know about this. Thank you for helping

jflyper, 2018-11-06:
@ujjwalk450 I'll look at this sometime tomorrow when I'm back at my bench. Can you provide a CLI "diff" output with OLED disabled?

ujjwalk450, 2018-11-06:
here you go sir @jflyper 

# diff all

# version
# Betaflight / MATEKF405 (MKF4) 3.5.2 Oct 16 2018 / 22:03:47 (a5e270fee) MSP API: 1.40

board_name MATEKF405
manufacturer_id 
mcu_id 0022003e3436470b37333638
signature 

# reset configuration to default settings
defaults nosave

# name
name THE_EC_GUY

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource LED_STRIP 1 A15
resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL
feature INFLIGHT_ACC_CAL
feature GPS
feature RSSI_ADC
feature LED_STRIP
feature AIRMODE

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200

# led
led 0 0,0::C:2
led 1 1,0::C:2
led 2 2,0::C:2
led 3 3,0::C:1
led 4 4,0::C:1
led 5 5,0::C:1
led 6 6,0::C:1
led 7 7,0::C:1
led 8 8,0::CB:1
led 9 9,0::A:1
led 10 10,0::A:1
led 11 11,0::CB:1

# color

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 4 900 1350 0 0
aux 2 2 4 1300 1700 0 0
aux 3 32 2 1400 2100 0 0
aux 4 35 3 1700 2100 0 0
aux 5 39 1 1300 2050 0 0
aux 6 47 4 1700 2100 0 0

# adjrange

# rxrange
rxrange 0 1049 2011
rxrange 1 1030 2006
rxrange 2 1035 2002
rxrange 3 1030 2002

# vtx

# rxfail

# master
set acc_trim_pitch = -20
set acc_trim_roll = 2
set acc_calibration = 76,91,-259
set fpv_mix_degrees = 20
set min_throttle = 1043
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set failsafe_throttle = 1500
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 177
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_initial_alt = 20
set gps_rescue_ground_speed = 500
set gps_rescue_min_sats = 4
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_vbat_pos = 396
set osd_rssi_pos = 2337
set osd_flymode_pos = 2381
set osd_throttle_pos = 2369
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 352
set osd_craft_name_pos = 2282
set osd_gps_speed_pos = 2243
set osd_gps_lon_pos = 2113
set osd_gps_lat_pos = 2129
set osd_gps_sats_pos = 2157
set osd_home_dir_pos = 2190
set osd_home_dist_pos = 2262
set osd_compass_bar_pos = 2091
set osd_altitude_pos = 2389
set osd_warnings_pos = 2281
set debug_mode = RANGEFINDER
set vtx_channel = 4
set vtx_freq = 5800
set displayport_max7456_wht = 0

# profile
profile 0

set dterm_notch_cutoff = 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save

ujjwalk450, 2018-11-06:
one thing more i discovered is that if OLED is enabled, i also cannot change the OSD layout

ujjwalk450, 2018-11-08:
any alternative to the issue @jflyper

jflyper, 2018-11-13:
@ujjwalk450 
It took sometime to setup a test environment; couldn't find my OLED module - had to make another one from spare parts.

The issue is reproducible. Looking into it.

jflyper, 2018-11-13:
Hmm... Doesn't reproduce with Omnibus F4.  Matek F405 CTR specific problem???

ujjwalk450, 2018-11-13:
i dont know, i have a cc3d revo and sp racing f3 only, both of them do not contain osd module, i have ordered another matek board, will check and see if this piece was faulty

jflyper, 2018-11-13:
F405CTR here exhibits the same behavior. It's a software problem specific to F405CTR.

ujjwalk450, 2018-11-13:
oh, i see, any solution to fix that ?

jflyper, 2018-11-13:
Trying to debug it with a nucleo, but as soon as I connect the OLED module, SWD interface dies... replaced an MCU, and it's dead again. Hmm... thinking of something else.

ujjwalk450, 2018-11-13:
so is it a mcu related problem ?

jflyper, 2018-11-13:
No. This has something to do with SWD. The chip itself is okay. Hang on.

jflyper, 2018-11-13:
WTH?

ujjwalk450, 2018-11-13:
> 
> 
> WTH?

didn't understand that

jflyper, 2018-11-13:
I think there's a buffer overrun condition when dashboard switches to a second page. The timing OSD stops updating matches with the dashboard page switching. More specifically, a variable called `max7456Lock` spontaneously becomes 1, and this will cause OSD not to update.

This needs some further investigation ... I'm calling it a day. It's 5:14 in the morning 🤣

jflyper, 2018-11-13:
Yeah, the `max7456Lock` is immediately behind `lineBuffer`, which is used by dashboard.

ujjwalk450, 2018-11-13:
ok sure sir, thank you so much
btw its 2 in the night in India 🤣🤣

etracer65, 2018-11-13:
`static char lineBuffer[SCREEN_CHARACTER_COLUMN_COUNT + 1];`

`SCREEN_CHARACTER_COLUMN_COUNT` is 21 (128 / 6).

There's a buffer overflow generated here when displaying the rcRates. Default rates would use 23 characters and look like:
```
RRr:100 PRR:100 YRR:100
```
https://github.com/betaflight/betaflight/blob/a2cfa7b1f08aebe8d74b2518cc960460078638ca/src/main/io/dashboard.c#L327-L331

Also a potential buffer overflow here if current was over 99A and > 999mah drawn.

https://github.com/betaflight/betaflight/blob/a2cfa7b1f08aebe8d74b2518cc960460078638ca/src/main/io/dashboard.c#L454

jflyper, 2018-11-14:
@etracer65 Yes, I noticed the `rcRate`s are overflowing. Didn't notice the amp case, though 😉

PR #7086 (2018-11-14): [DASHBOARD] Fix buffer overruns
Fixes two buffer overrun conditions.

As described by @etracer65 in https://github.com/betaflight/betaflight/issues/7042#issuecomment-438441670

---
`static char lineBuffer[SCREEN_CHARACTER_COLUMN_COUNT + 1];`

`SCREEN_CHARACTER_COLUMN_COUNT` is 21 (128 / 6).

There's a buffer overflow generated here when displaying the rcRates. Default rates would use 23 characters and look like:
```
RRr:100 PRR:100 YRR:100
```
https://github.com/betaflight/betaflight/blob/a2cfa7b1f08aebe8d74b2518cc960460078638ca/src/main/io/dashboard.c#L327-L331

Also a potential buffer overfl [... cut]
closed 2018-11-14 by mikeller