betaflight/betaflight #7052: VTX Low Power on Disarm - Enable Settable Delay & AUX Channel Toggle Reset

TehllamaFPV, 2018-11-06:
Feature Request: Add (or expose) timing parameters to VTX Control low power (or pit mode on disarm) settings.  

The idea is that the same functionality of low power on disarm, expose a timer if disarmed for a set time (e.g. 0.1s to 1000s).  
The other added feature portion is enabling any change on a specified AUX channel to revert to flight power, and reset the timer - e.g. any AUX channel toggle resets this timer, or the Buzzer/Beeper, PreArm, and FlipOver channels reset the timer.  This could also be implemented with any channel change (i.e. all channels but a RSSI channel), so that in only a transmitter idle state will this default to low power.

For many builds, I like being able to default to low power when sitting idle for thermal reasons, but I don't want to be constantly toggling power.  Realistically, I would tend to set this at 100s default, with all non-RSSI AUX channels resetting and resetting the timer, so that setting up to fly in a group the FC isn't commanding VTX power changes constantly, but will command low power after some delay, but that toggling the beeper or pre-arm switch sets the video power back to specified flight power output.

Currently, I'm reliant on OSD or LUA script to manually toggle VTX power in cases where I suspect overheating VTX is likely - being able to configure this once (even if its a bit fiddly and requires CLI) would be very valuable to me, so that it's a set-and-forget and if I'm distracted I'm less likely to leave a transmitter running at full power, or need to arm in order to enable video transmission to locate a downed craft.

wind0r, 2018-11-06:
You could use vtx_low_power_disarm in the UNTIL_FIRST_ARM mode and for the cases you need to power down after the first arm you can use the BOXVTXPITMODE switch.

i think with the pit mode switch and vtx_low_power mode everything is possible to do.
if there still is something missing pleace descripe that usecase. adding some function that monitors every input should be the last last resort

TehllamaFPV, 2018-11-06:
You're correct - everything I'm asking for is currently possible with setting a pit_mode_switch and vtx_low_power mode, the feature request is to automate it on a settable timer, and enable aux channels (or any channel - preferrably ignoring a RSSI channel) to reset that timer.

I have a couple of practical use cases - long explanation, so tried to leave it out of the initial post.

First one: setting up to do a race heat (practice/qualifying/race).  Inevitably somebody is slower than I am, which the until_first_arm does a great job of handling, but if there's a false start or holeshot rule failure (i.e. somebody else has a small fixable malfunction), my only option is to rely on vtx pit mode switch... and for me to remember that I've done this prior to launching the next time.
Similarly, for racing, I would prefer to be able to test that my video is good in flight configuration (and arguably more importantly not interfering with other pilots), but not sit there overheating if somebody else has to go walk out there and fix a minor issue.  The more direct tied to disarm concerns me because I also have unusual crashes in races.

Second one: When flying with friends, and/or if serving as a race director (this is especially amusing when somebody is borrowing my quad while I'm running timing systems and flying), if they have a situation where I need to land urgently and fix a safety issue, I'd like for my quad to not overheat, but be able to re-enable video and fly back easily.   I'm 3 for 3 on my loaner quads catching fire while somebody else is flying them and I'm also up in the air.  To say I'm usually distracted in these situations is an undestatement, so I'd like to automate that VTX low-power switchover, without risk of losing video if I have an unusual crash in a race where I can rearm and resume.

On some stupid-light builds and flying in the desert southwest, I'm up against overheating issues with integrated FC/VTX solutions... these warnings only show up in OSD as currently configured, and if I'm helping extinguish a fire or find a quad, I won't notice them.
I also want to keep my builds portable to my i6x controller, which has a grand total of four aux switches (Arm, Prearm, Buzzer, and Turtle are my default allocation, which includes some duplicated mapping that captures flight mode, blackbox enable).  
The other workaround if video jumps to low power is to arm the quad in turtle mode but leave the sticks alone - but I've also broken stuff by failing to realize I was in turtle mode (even with the warning right on the OSD).

ijustwannafly, 2018-11-08:
Not sure how constantly toggling power would happen on just "low power on disarm" setting alone.

When disarmed it goes to the lowest power except pit mode and stays at the power level.

When armed it goes to the highest power for that protocol and stays there until disarmed.

The only way the power would be toggled constantly is if you are arming and disarming in a short amount of time for whatever reason.  If that's the case then you could/should disable the "low power on disarm", set your max vtx power and reboot the fc to save the vtx settings, then configure pit mode on a switch.

There is also paralyze mode.  "Low power on disarm" shouldn't drop your video for more than a second when the power level changes.  Unless your goggle setup just can't keep a lock on the band/channel.  Especially in racing where raceband would be used for the purpose of minimizing bleedthrough between channels.

25mW (default low power on disarm setting) shouldn't hurt anything...it hasn't on any of my quads when sitting idle on a table in the field.  Flying with friends we always use raceband to fly together and even on low power disarm I stay locked into my vtx feed and use my model finder feature to retrieve my quad while my buddies are still zipping around me on 200+mW.

You mentioned using an i6x controller and I believe that is a flysky tx.  Sadly, you aren't able to use logic/lua options unless you are using opentx.  I haven't touched flysky in ages and I thought the i6x might have had an option to use otx but it's highly likely I am mistaken on this point.

TehllamaFPV, 2018-11-08:
Correct on all accounts.  

My thinking is that this is fairly low-hanging fruit in terms of cost to implement.
That said, definitely not a high priority because in all likelihood, the user base for VTX control and desires for this feature that does not intersect with LUA script capability via CRSF/SmartPort/FPort might be smaller than even I suspect.
Practically, as a LUA script adjustement that might answer the question for enough people, just not necessarily for me (I'm already in the process of robbing XSRs off perfectly good quads to shift around capability to achieve this in my fleet, but that isn't a complete solution)

I am still running very cost-conscious hardware (Flysight Spex02 goggles, FlySky X6B Receivers on many builds), which is another part of the impetus for this - you're absolutely correct that it's unusual to combine budget RC hardware with smartAudio/Tramp control, but the latter is getting a lot cheaper, and I can incorporate these features into $150 builds.

I have had issues with certain quads overheating on 25mW - arguably more of a hardware issue (and the fundamental brilliance of integrating an F411 chip AND VTX on the same 26x26mm PCB) - highest reading on that was 82°C at the CPU core thermocouple reading (showed up on the OSD), and that was in a matter of minutes.   

Pit mode on a switch would be the default answer, but without overloading that capability onto a switch that's already allocated on my i6x becomes fairly tricky (since every available switch is already in use, at least for me).

ijustwannafly, 2018-11-08:
The i6x is 10 channel capable.  You could map pit mode to one of the pots at the top center of the tx.  From your description it sounds like you are using 8 out 10 channels possible.

closed 2026-09-26 by blckmn