betaflight/betaflight #7054: Binding IO from freed resource motor 4-6 to servo causes motor 3 pwm to stop

TheTape, 2018-11-07:
**Describe the bug**
1: Binding IO from freed resource motor 4-6 to servo causes motor 3 pwm to stop in tricopter config.
2: smix servo doesn't correspond to resource servo number (5 means 1)

**To Reproduce**
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 NONE (was A12)
resource MOTOR 5 NONE (was B08)
resource MOTOR 6 NONE (was B09)
resource MOTOR 7 NONE (was A02)
resource MOTOR 8 NONE (was A03)
resource SERVO 1 <A12, B08 or B09 disables MOTOR 3, but A02 and A03 works>
mmix reset
mmix 0 1.000 0.000 1.000 0.000
mmix 1 1.000 -1.000 -0.200 0.000 
mmix 2 1.000 1.000 -0.200 0.000
smix reset
smix 0 5 2 100 0 20 80 0
save

**Expected behavior**
Basic sanity (seriously, wtf all across 95% of BF and it's near identically insane siblings, any aspect is flawed to some degree)

**Flight controller configuration**
```resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 B09
resource GYRO_EXTI 2 NONE
mixer CUSTOMTRI
mmix 0  1.000  0.000  1.000  0.000
mmix 1  1.000 -1.000 -0.200  0.000
mmix 2  1.000  1.000 -0.200  0.000
servo 7 1000 2000 1500 99 -1
smix 0 5 2 100 0 20 80 0

feature -GPS
feature -DISPLAY
set acc_calibration = 106,-25,133
set serialrx_provider = SBUS
set max_throttle = 1930
set motor_pwm_protocol = OFF
set pid_process_denom = 5
profile 0

rateprofile 0```

**Setup / Versions**
 - SPRACINGF3 with BF 4.0 compiled from modified rep source (enabled/disabled functionality, added manual dashboard page cycling and stick commands; nothing related to IO modified in source, except defines)

**Additional context**
target.h
`#pragma once

//INIT
#define TARGET_BOARD_IDENTIFIER "SRF3"
#define USE_SLOW_MSP_DISPLAYPORT_RATE_WHEN_UNARMED
#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT
#define DEFAULT_RX_FEATURE      FEATURE_RX_SERIAL
#define DEFAULT_FEATURES        (FEATURE_RSSI_ADC | FEATURE_GPS | FEATURE_RANGEFINDER | FEATURE_DASHBOARD | FEATURE_ANTI_GRAVITY | FEATURE_AIRMODE)

//SENS
#define USE_ADC
#define ADC_INSTANCE            ADC2
#define DEFAULT_VOLTAGE_METER_SOURCE VOLTAGE_METER_ADC
#define VBAT_ADC_PIN            PA4
#define DEFAULT_CURRENT_METER_SOURCE CURRENT_METER_ADC
#define CURRENT_METER_ADC_PIN   PA5
//#define USE_MSP_CURRENT_METER
#define RSSI_ADC_PIN            PB2

//PERPS
#define USE_FLASHFS
#define USE_FLASH_M25P16
#define FLASH_CS_PIN            PB12
#define FLASH_SPI_INSTANCE      SPI2

#define USE_BEEPER
#define BEEPER_PIN              PC15
#define BEEPER_INVERTED

#define USE_EXTI
#define USE_GYRO_EXTI
#define GYRO_1_EXTI_PIN          PC13
#define USE_MPU_DATA_READY_SIGNAL
#define ENSURE_MPU_DATA_READY_IS_LOW
#define USE_GYRO
#define USE_GYRO_MPU6050
#define GYRO_1_ALIGN            CW270_DEG

#define USE_ACC
#define USE_ACC_MPU6050
#define ACC_1_ALIGN             CW270_DEG

#define USE_BARO
#define USE_BARO_BMP280

#define USE_MAG
#define USE_MAG_QMC5883
//#define MAG_HMC5883_ALIGN       CW270_DEG
//#define USE_MAG_DATA_READY_SIGNAL
//#define ENSURE_MAG_DATA_READY_IS_HIGH
//#define MAG_INT_EXTI            PC14

#define USE_RANGEFINDER
#define USE_RANGEFINDER_HCSR04
#define RANGEFINDER_HCSR04_TRIGGER_PIN       PB0
#define RANGEFINDER_HCSR04_ECHO_PIN          PB1
//#define RANGEFINDER_HCSR04_SOFTSERIAL2_EXCLUSIVE

//PORTS
#define USE_UART1
#define USE_UART2
#define USE_UART3
//#define USE_SOFTSERIAL1
//#define USE_SOFTSERIAL2
#define SERIAL_PORT_COUNT       3

//#define SOFTSERIAL1_RX_PIN      PB4 // PWM 5
//#define SOFTSERIAL1_TX_PIN      PB5 // PWM 6
//#define SOFTSERIAL2_RX_PIN      PB0 // PWM 7
//#define SOFTSERIAL2_TX_PIN      PB1 // PWM 8

#define UART1_TX_PIN            PA9
#define UART1_RX_PIN            PA10
#define UART2_TX_PIN            PA14 // PA14 / SWCLK
#define UART2_RX_PIN            PA15
#define UART3_TX_PIN            PB10 // PB10 (AF7)
#define UART3_RX_PIN            PB11 // PB11 (AF7)
#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C_DEVICE              (I2CDEV_1)
#define USE_SPI
#define USE_SPI_DEVICE_2 // PB12,13,14,15 on AF5

#define REMAP_TIM17_DMA

// IO - stm32f303cc in 48pin package
#define TARGET_IO_PORTA         0xffff
#define TARGET_IO_PORTB         0xffff
#define TARGET_IO_PORTC         (BIT(13)|BIT(14)|BIT(15))
#define TARGET_IO_PORTF         (BIT(0)|BIT(1)|BIT(3)|BIT(4))

#define USABLE_TIMER_CHANNEL_COUNT 17
#define USED_TIMERS             ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(15) | TIM_N(16) | TIM_N(17) )`

common_pre.h
`#pragma once

#define USE_PARAMETER_GROUPS
#define DEBUG_MODE DEBUG_NONE // change this to change initial debug mode

#define I2C1_OVERCLOCK true
#define I2C2_OVERCLOCK true

#define MINIMAL_CLI
//#define USE_DSHOT
#define USE_GYRO_DATA_ANALYSE

#define DEFAULT_AUX_CHANNEL_COUNT       MAX_AUX_CHANNEL_COUNT
#define MAX_PROFILE_COUNT 3
#define FAST_CODE
#define FAST_CODE_NOINLINE
#define FAST_RAM_ZERO_INIT
#define FAST_RAM
#define TASK_GYROPID_DESIRED_PERIOD     1000 // 1000us = 1kHz
#define SCHEDULER_DELAY_LIMIT           100
#define SRAM2

//#define USE_BRUSHED_ESC_AUTODETECT  // Detect if brushed motors are connected and set defaults appropriately to avoid motors spinning on boot
#define USE_CLI
//#define USE_SERIAL_PASSTHROUGH
#define USE_TASK_STATISTICS
#define USE_GYRO_REGISTER_DUMP  // Adds gyroregisters command to cli to dump configured register values
#define USE_IMU_CALC
//#define USE_PPM
//#define USE_PWM
#define USE_SERIAL_RX
//#define USE_SERIALRX_CRSF       // Team Black Sheep Crossfire protocol
//#define USE_SERIALRX_IBUS       // FlySky and Turnigy receivers
#define USE_SERIALRX_SBUS       // Frsky and Futaba receivers
//#define USE_SERIALRX_SPEKTRUM   // SRXL, DSM2 and DSMX protocol
//#define USE_SERIALRX_SUMD       // Graupner Hott protocol
//#define USE_SERIALRX_SUMH       // Graupner legacy protocol
//#define USE_SERIALRX_XBUS       // JR
//#define USE_ACRO_TRAINER
#define USE_BLACKBOX
#define USE_RESOURCE_MGMT
#define USE_RUNAWAY_TAKEOFF     // Runaway Takeoff Prevention (anti-taz)
#define USE_SERVOS
//#define USE_TELEMETRY
//#define USE_TELEMETRY_FRSKY_HUB
//#define USE_TELEMETRY_HOTT
//#define USE_TELEMETRY_LTM
//#define USE_TELEMETRY_SMARTPORT
//#define USE_CAMERA_CONTROL
#define USE_CMS
//#define USE_EXTENDED_CMS_MENUS
//#define USE_DSHOT_DMAR
#define USE_GYRO_OVERFLOW_CHECK
#define USE_YAW_SPIN_RECOVERY
#define USE_HUFFMAN
//#define USE_MSP_DISPLAYPORT
//#define USE_MSP_OVER_TELEMETRY
#define USE_PINIO
#define USE_PINIOBOX
#define USE_RCDEVICE
//#define USE_RTC_TIME
//#define USE_RX_MSP
#define USE_SERIALRX_FPORT      // FrSky FPort
//#define USE_TELEMETRY_CRSF
//#define USE_TELEMETRY_SRXL
//#define USE_VIRTUAL_CURRENT_METER
//#define USE_VTX_COMMON
//#define USE_VTX_CONTROL
//#define USE_VTX_SMARTAUDIO
//#define USE_VTX_TRAMP
#define USE_GYRO_LPF2
//#define USE_ESC_SENSOR
//#define USE_ESC_SENSOR_INFO
//#define USE_CRSF_CMS_TELEMETRY
//#define USE_BOARD_INFO
#define USE_SMART_FEEDFORWARD
#define USE_THROTTLE_BOOST
#define USE_RC_SMOOTHING_FILTER
#define USE_ITERM_RELAX
#define USE_DYN_LPF

#define USE_DASHBOARD
#define USE_GPS
//#define USE_GPS_NMEA
#define USE_GPS_UBLOX
#define USE_GPS_RESCUE
//#define USE_GYRO_DLPF_EXPERIMENTAL
//#define USE_OSD
//#define USE_OSD_OVER_MSP_DISPLAYPORT
//#define USE_MULTI_GYRO
//#define USE_OSD_ADJUSTMENTS
//#define USE_SENSOR_NAMES
//#define USE_SERIALRX_JETIEXBUS
//#define USE_TELEMETRY_IBUS
//#define USE_TELEMETRY_IBUS_EXTENDED
//#define USE_TELEMETRY_JETIEXBUS
//#define USE_TELEMETRY_MAVLINK
//#define USE_UNCOMMON_MIXERS
//#define USE_SIGNATURE
//#define USE_ABSOLUTE_CONTROL
//#define USE_HOTT_TEXTMODE
//#define USE_LED_STRIP
//#define USE_VARIO`

mikeller, 2018-11-07:
@TheTape: You need to tone it down. Insulting people is not going to make people more likely to help you. Also, keep in mind that Betaflight is a community project, and a lot of the features in it have been contributed by community members with not much involvement in the project. Getting new features is seen as more valuable to not losing consistency by not adding a feature because it's not 100% consistent with everything else.

Your configuration cannot work because motor outputs 3 - 6 share the same timer, and cannot be used for a mix of servo and motors. Rearrange your outputs so that servos and motors use separate timers (usually motors use outputs 3 - 6 for tricopter on SPRACINGF3).

Read the documentation (https://github.com/betaflight/betaflight/blob/master/docs/Mixer.md#custom-servo-mixing) to get an idea how the servo mixer outputs work.

jflyper, 2018-11-08:
> Your configuration cannot work because motor outputs 3 - 6 share the same timer, and cannot be used for a mix of servo and motors.

In other words, you can't use one of your car wheels to rock your chair if you still want to drive it.

closed 2018-12-15 by stale[bot]