betaflight/betaflight #7055: FrSky fport telemetry broken

adrianmiriuta, 2018-11-07:
FrSky fport telemetry broken
for all targets.  (tested F3 F4 F7)

Only intermittent sensor data update.
At boot for 1 sec all sensors get updated
after that some get updated others not.

my configuration for F7

# diff

# version
# Betaflight / OMNINXT7 (ONX7) 4.0.0 Nov  7 2018 / 16:29:31 (3492bcff3) MSP API: 1.41

board_name OMNINXT7
manufacturer_id 

# name
name OmniNXT7

# resources

# mixer

# servo

# servo mix

# feature
feature -LED_STRIP
feature -OSD
feature -ANTI_GRAVITY
feature TELEMETRY
feature AIRMODE

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 1 1950 2075 0 0
aux 1 1 0 1525 1650 0 0
aux 2 2 0 1700 1825 0 0
aux 3 6 2 1950 2075 0 0
aux 4 7 3 1950 2075 0 0
aux 5 13 1 1425 1550 0 0
aux 6 26 1 1950 2075 0 0
aux 7 30 2 1450 1550 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass2_hz = 0
set gyro_calib_duration = 333
set gyro_calib_noise_limit = 13
set gyro_overflow_detect = OFF
set acc_calibration = 24,-17,-311
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1000
set max_check = 2000
set rc_interp_ch = RPY
set rc_interp_int = 9
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1500
set vbat_warning_cell_voltage = 34
set current_meter = ESC
set battery_meter = ESC
set small_angle = 180
set deadband = 7
set yaw_deadband = 11
set pid_process_denom = 1
set frsky_unit = METRIC

# profile
profile 0

set dterm_lowpass_type = PT1
set throttle_boost = 0
set p_pitch = 61
set i_pitch = 41
set d_pitch = 41
set p_roll = 61
set i_roll = 41
set d_roll = 41
set p_yaw = 71
set i_yaw = 81
set d_yaw = 3
set horizon_tilt_effect = 50

# rateprofile
rateprofile 0

set thr_mid = 21
set thr_expo = 61
set roll_rc_rate = 137
set pitch_rc_rate = 137
set yaw_rc_rate = 137
set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 73
set tpa_rate = 1

mikeller, 2018-11-07:
@adrianmiriuta: What RX are you using, and what firmware version are you running on it?

adrianmiriuta, 2018-11-08:
@mikeller 

TX. taranis 9xd plus se  opentx 2.1.9
RX. R-XSR R-XSR_FPORT_LBT180228

stable release 
# Betaflight / OMNINXT7 (ONX7) 3.4.0 Jul  8 2018 / 04:26:00 (cce8a59be) MSP API: 1.39
works !

mikeller, 2018-11-08:
@adrianmiriuta: Weird, this should work (can't confirm myself though, as my hardware is all FCC). There still is a bug in the bytestuffing in this version, but it normally does not completely disable telemetry. Can you try the beta version from https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-423505314 to see if that fixes it in your case?

SchonRichtig, 2018-11-08:
I use FPort on R9 Mini together with a kakute f7 on BF 3.5.2 without any problems.

adrianmiriuta, 2018-11-08:
@mikeller 

with 
R-XSR_LBT180921Beta.frk
i get no rc-data and no telemetry at all

jflyper, 2018-11-08:
> R-XSR_LBT180921Beta.frk
i get no rc-data and no telemetry at all

What does this suggest???

adrianmiriuta, 2018-11-08:
i cannot revert to old firmware
RX does not get recognized by flasher any more....
I think I'v got a bad RX ...

closed 2018-11-08 by adrianmiriuta

SchonRichtig, 2018-11-08:
R-XSR_FPORT_LBT180827.frk should be the right firmware for F.Port

mikeller, 2018-11-08:
@adrianmiriuta: So sorry, the link to the comment that I posted was for the SBus / SmartPort version. The correct beta version for FPort is in https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-416183224. But having flashed the SmartPort version should not prevent you from flashing the FPort version later on, so there must be something else wrong.

@SchonRichtig: Yes, correct. My bad.

smpotter, 2019-01-10:
All is working with the latest #1341betaflight 4.0! Be sure to to check your ports for proper setup. I.e. tramp smart audio etc. Happy Flying!