betaflight/betaflight #7057: Buzzer not working / DSHOT beacon OK

Quadtime, 2018-11-08:
**Describe the bug**
After updating to BF 4.0.0 (from 3.5) I noticed that the (piezo) buzzer is not working. DSHOT beacon (set to 3) is working as it should. I tried some beeper settings (beeper_inversion, beeper_od, beeper_frequency), but that doesn't help. Even trid differend buzzers and tested if they are working when connect them to 5V.
 
**Flight controller configuration**
# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Oct 29 2018 / 10:57:01 (0e7a4d6e1) MSP API: 1.41

# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Oct 29 2018 / 10:57:01 (0e7a4d6e1) MSP API: 1.41

board_name OMNIBUSF4
manufacturer_id AIRB

# name
name Matse

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200

# led
led 0 6,10::CW:2
led 1 7,10::CTW:11
led 2 8,10::CTW:11
led 3 9,10::CW:2

# color

# mode_color

# aux
aux 0 0 2 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 15 1 1300 1700 0 0
aux 3 35 0 1700 2100 0 0
aux 4 47 0 1325 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = -116,5,-100
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 8
set serialrx_provider = SBUS
set min_throttle = 1070
set dshot_idle_value = 400
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set bat_capacity = 1300
set vbat_warning_cell_voltage = 34
set beeper_od = ON
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_fail_safe = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1500
set osd_vbat_pos = 2466
set osd_rssi_pos = 2487
set osd_tim_1_pos = 2519
set osd_craft_name_pos = 2498
set osd_warnings_pos = 2409
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5732

# profile

**Setup / Versions**
FC Dys Omnibus F4 Pro V2
4 x Airbot wraith32 ESC
Matek VTX @ UART6 f. smartaudio
R9MM receiver @ UART 1 with smartport (uninverted) @ UART 3

Maybe its a bug because I had a similar problem with 2 5" quads based on a Josh Bardwell F4.

jflyper, 2018-11-08:
@Quadtime Can you start with a fresh config (i.e. with CLI `defaults`) and test your buzzer again?

Quadtime, 2018-11-09:
@jflyper I reset the quad to defaults, setup SBUS on UART 1, select beeper mode with switch AUX2, but no buzzer. I'll try it with a new firmware install (4.0.0 #1203) again. No buzzer, but DSHOT beacon works.

Quadtime, 2018-11-09:
Maybe I solved it, it seemed to be a hardware problem with the DYS omnibus F4. I just flashed my old BF 3.5, did a reset and configured SBUS and switch for buzzer only. When switching I heard a tickering noise at a very low volume from the buzzer. I take a measuring with my oszilloscop, detecting a squarewave at 5.8 kHz with a 4 V amplitude. 
I take another buzzer and it was tweeting very weak. The thing is that I did a new wiring when updating to BF 4.0.0 with a R9 receiver and a new Matek tailpiece with LED and buzzer. I checked my photo from the build before with a standalone buzzer and for some reason I used 9V at the buzzer +. May be this is a problem with the DYS providing a voltage to low for the buzzer at the Matek tailpiece. Due to the LEDs the 9V isn't suitable, but I'll wire the Buzzer separately and check it with BF 4.0.0!
[image]

Quadtime, 2018-11-10:
So I attached the "Buzzer +" to 9V, and the beeper runs @BF 3.5.2 (clean install). Tried Betaflight 4.0.0 Build #1234, clean install, beeper works. When rewriting the DIFF, beeper stops working. FInd out that it ist the "set beeper_od = ON" command, must be "set beeper_od = OFF". Now both beeper and DSHOT beacon are working. Beeper sounds a little strange, some beeps are like "strangled". Maybe an DMA issue, but not important. DSHOT beacon is the way to go, because it's not so directionally sounding like the small piezo speakers. I'll close this now.

closed 2018-11-10 by Quadtime