betaflight/betaflight #7064: Can I reverse the artificial horizon on the OSD?

GeckoLibre, 2018-11-10:
Hello,

I have a 2 axis gimbal on my quadcopter with the FPV cam on the gimbal. Because of this, the view from my perspective will always stay level despite the roll and pitch angle of the quad. I want to use the artificial horizon to track the quad's angle to help me fly, but with standard settings it is reversed. Is there a way to go into the OSD settings of the CLI and put a negative sign somewhere to get the pitch and roll angles of the artificial horizon to be reversed?

I have the 10.4 configurator with BF 3.5.2 on my Matek F405 CTR.

mikeller, 2018-11-10:
@GeckoLibre: No, you'd have to build custom firmware for this - it should not be too hard to get this effect, but it's not possible with configuration only.

GeckoLibre, 2018-11-10:
Ah darn okay thanks for that crucial piece of info. I guess I'll have to get used to the backwards method then. Thanks!

etracer65, 2018-11-11:
@GeckoLibre I have implemented this in #7072 

Below is a version for you to try. Since the Matek F405 comes in different variants, be sure to check/adjust your board orientation to make sure it's correct. Some variants have the gyro mounted differently and may not match the defaults in the target.

[betaflight_4.0.0_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/2569485/betaflight_4.0.0_MATEKF405.hex.zip)

etracer65, 2018-11-11:
Forgot to mention, you need to go into the CLI and `set osd_ah_invert = ON`.

PR #7072 (2018-11-12): Add osd_ah_invert to allow reversing the direction of the OSD artificial horizon
Fixes #7064
closed 2018-11-12 by mikeller