betaflight/betaflight #7075: BF2.0 build 1203 yaw spin issue

dabit20, 2018-11-11:
Not sure where to post this, please point me to the correct place if this is the incorrect location.

I had a yaw spin issue today with BF 4.0 build 1203. Never had those with 3.4 and 3.5, so this is either new in BF4.0 or I have a yet undetected issue with the quad hardware.
The highdef video captured this spin: https://www.youtube.com/watch?v=BgV_V5TabIo
After disarming, hitting the ground and rearming I was able to continue flying. 

Note that the quad is not shooting very high in the air, which it would do if two motors were at full throttle and the other two were not.

Weird thing: I cannot find the issue in the BB logs. Unfortunately the goggles DVR was not running and file date/time from the Runcam Split is bogus, so I have only a rough idea what log should contain this issue. I checked them, but I have not found it. 
The issue happened 2 minutes, 26 seconds after arming. I think the log which should contain this issue is LOG00099, but I am not 100% sure. 
After this issue I flew a few batteries without issues. 
I zipped all flight logs, these can be found here:
https://www.icecoldcomputing.com/directlink/temp/BF40_1203_BBLogs_11nov2018_sibeliuspark.zip

Overflow and yaw spin recovery are set:
set gyro_overflow_detect = ALL
set yaw_spin_recovery = ON
set yaw_spin_threshold = 950

FC board is a Matek F405STD with ICM20602 gyro. ESC is a DALRC Engine BLHeli32 4-in-1, and I am running 2306 2450kV Hypetrain motors and 5S batteries on a Chameleon Ti 5" frame. There is 3300uF+470uF filtering capacitance directly on the Vbat connection to the ESC.

I sacrified a few props in trying to reproduce this, but so far I failed. 

Diff:
```
# name
name BvH_Ti

# resources
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 C07
resource MOTOR 4 C06
resource MOTOR 7 NONE
resource PWM 3 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02

# mixer

# servo

# servo mix

# feature
feature GPS
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 1 64 115200 57600 115200 115200
serial 2 16384 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200
serial 4 2 115200 115200 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 0,0::C:3
led 1 1,0::C:2
led 2 2,0::C:2
led 3 3,0::C:7
led 4 4,0::C:1
led 5 5,0::C:1

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 2 1 900 1100 0 0
aux 2 46 1 1900 2100 0 0
aux 3 13 2 1300 1550 0 0
aux 4 28 1 1350 1600 0 0
aux 5 30 1 1350 1600 0 0
aux 6 33 3 1850 2100 0 0
aux 7 34 1 1850 2075 0 0
aux 8 35 2 1800 2050 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 7 s 1000

# master
set yaw_spin_threshold = 950
set acc_calibration = -82,11,-23
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 16
set fpv_mix_degrees = 10
set blackbox_p_ratio = 64
set dshot_idle_value = 400
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 7
set failsafe_throttle = 1200
set failsafe_throttle_low_delay = 20
set failsafe_procedure = AUTO-LAND
set align_board_yaw = 90
set bat_capacity = 1300
set vbat_scale = 109
set ibata_scale = 212
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 30
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1500
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set frsky_vfas_precision = 1
set osd_warn_core_temp = OFF
set osd_warn_fail_safe = OFF
set osd_cap_alarm = 1300
set osd_rssi_pos = 2099
set osd_tim_1_pos = 233
set osd_remaining_time_estimate_pos = 465
set osd_flymode_pos = 2105
set osd_throttle_pos = 448
set osd_current_pos = 65
set osd_mah_drawn_pos = 2113
set osd_craft_name_pos = 2507
set osd_gps_speed_pos = 2145
set osd_gps_lon_pos = 146
set osd_gps_lat_pos = 114
set osd_gps_sats_pos = 2499
set osd_home_dir_pos = 2090
set osd_home_dist_pos = 2091
set osd_altitude_pos = 2177
set osd_warnings_pos = 2314
set osd_avg_cell_voltage_pos = 2082
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set debug_mode = GYRO_SCALED
set cpu_overclock = 216MHZ
set vtx_band = 5
set vtx_power = 5 ###CORRUPTED CONFIG###
set vtx_freq = 5658
set rcdevice_init_dev_attempts = 4

# profile
profile 0

set feedforward_transition = 5
set iterm_relax_type = GYRO
set p_pitch = 55
set f_pitch = 70
set p_roll = 51
set f_roll = 70

# rateprofile
rateprofile 0

set thr_mid = 0
set roll_rc_rate = 93
set pitch_rc_rate = 93
set yaw_rc_rate = 93
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_rate = 15
set tpa_breakpoint = 1050
set throttle_limit_type = SCALE
set throttle_limit_percent = 85
```

etracer65, 2018-11-12:
Can't open the log zip file. Getting a damaged archive error.

dabit20, 2018-11-12:
I am getting the same error when trying to extract the contents; LOG00089.BFL seems to be corrupt. The rest seems to be fine. Weird. 
Sorry for the inconvenience. I will recreate the archive tonight.

In the meantime there were a few questions and answers in the Youtube comments. For completeness I will copy/paste the important bits here:

Q: Can it be that one Motor was blocked due to the branches you hit?﻿
A: There were no branches or leaves on the arms or between motor base and bell when I flew back and landed. Props were not loose or damaged either. But yes, it is a possibility.
Also, if one motor was blocked I would expect that a motor trace would go up to the max in blackbox. I did not see it (or I did not search for it thoroughly enough)﻿

Q: did you have crash recovery on or off?﻿
A: at the time this video was recorded, crash recovery was OFF.
However, yaw spin and gyro overflow prevention are turned ON

dabit20, 2018-11-13:
Something goes wrong as soon as I copy the ZIP to my webhosting. 
Oh well.. This works too, although bandwidth is not very high: https://nextcloud.icecoldcomputing.com/s/qt8dEAqaTq3XPJ9

etracer65, 2018-11-13:
Looks like log 91 has the event. The log shows that the gyro rate exceeded 1950dps and gyro overflow detection enabled. This disabled the PIDs and the spinning began to slow down and once < 1850dps it disabled. Then yaw spin recovery took over and disabled roll/pitch and used the full yaw resources to recover. This worked and the yaw rate got below the threshold and yaw spin recovery disabled and returned normal PID controller response. Everything worked as expected to this point.

But when the PID's reenabled something was still wrong and the spin began again. Again overflow detect and yaw spin recovery did their things. So something was wrong mechanically that was causing the spin. Maybe a loose prop, branch wedged in a motor, etc. Can't really tell from the log as the activity is too chaotic to identify a specific motor.

dabit20, 2018-11-13:
OK, thanks for looking into the issue. 
If we can conclude that the issue was caused by a mechanical problem and all software recovery mechanisms worked as expected, this issue can be closed. 

I will flash a fresh build and keep on crashtesting :)
BF4.0 works so much better on this quad than 3.5.2, the difference is night and day. Thanks for doing all that magic!

closed 2018-11-13 by dabit20