betaflight/betaflight #7077: Runcam Device control through stick commands

dabit20, 2018-11-11:
One can control an FPV-camera with 2-pin OSD connector using stick commands. However, with the FC talking to the Split using a uart/Runcam Device Protocol this is not possible (or I missed something).

This means that fully controlling the Split through the FC requires three RC channels, or more accurate: three modes. This is quite a lot given the fact that using more than 8 active RC channels is not preferred (at least on FrSky hardware; latency jumps from 9ms to 18ms) and recycling existing channels used for things like flight mode control is limited. 

I would like to have the camera control through stick commands more generic so it also works for Runcam Device protocol devices and possibly future devices.

McGiverGim, 2018-11-12:
Yes, this will be a very useful feature. I asked for it to Runcam without luck. Let's see if some developer can do something for it, but the protocol of the Split has more features than the standard OSD, and is not very good designed, so it will not be easy...

etracer65, 2018-11-12:
Or maybe something more explicit like extend `camera_control_mode` to add a RunCam mode. I realize the existing parameter is meant to control the PWM/DAC signaling behavior of camera control but maybe this could be abstracted to be more generic. Or maybe move it up a level and add `camera_control_type`.

dabit20, 2018-11-13:
@mikeller: IMHO 'everybody in charge' only works for things that do not require any change to existing architecture. As soon as a discussion goes in a 'maybe move it up a level' direction someone with a good overview of the current architecture and where it is heading can make better decisions than 'the new guy' ;)

I will set up the development environment, browse through the code, and see if I am capable of doing a serious contribution instead of a hack. I am fluent in C, slightly less fluent in C++ and I also have experience with STM32 and Cortex-M in general, but I am no software engineer.

mikeller, 2018-11-13:
@dabit20: I never meant to imply that myself or other experienced members of the Betaflight community will not help you as 'the new guy' with advice on how things can best be done in Betaflight. But there are more aspects than pure technical merit, like user preferences and priorities of features (in the light of an acute shortage of flash space on F3), and these have to be considered in the final decision as well.

dabit20, 2018-11-13:
@mikeller : I did not interpret your text that way. Just that it is a waste of time to hack something together with limited future useability, and that I do understand that there are many other aspects than just technical issues that needs to be considered, history and future amongst them.

If I have questions like 'how do I ..', 'where can I find ..', 'can you guide me', etc., what is the best place to ask them?

mikeller, 2018-11-13:
@dabit20: I think the approach to extend the existing camera control feature / stick combinations to work for RunCam devices hits about the right level of feature richness vs. configuration complexity, so it will have a good chance of being accepted.

You can always ask questions in here, or you can join the Betaflight Slack (registration at https://slack.betaflight.com/).

dabit20, 2018-11-13:
I am currently browsing through the BF codebase, figuring out the general structure. Setting up the build environment and generating a hex file went completely smooth and without any trouble. So far so good. Let's see if I can find the SWD pins on my FC somewhere to attach the STLink.

ghostface, 2018-12-03:
Last time I checked the splits protocol I was under the impression that it was changed to use a newer runcam specific protocol which already supports cam controls via the splits uart.

dabit20, 2018-12-04:
Cam controls work fine, that is not the issue. The issue is that I need 3 modes to accomplish full camera control. With FrSky RC hardware set to 8-channel mode or when using RC hardware such as the FlySky i6 it is close to impossible to implement these three extra modes on top of all the other desired functionality such as arm, flight mode control, turtle mode, beeper, etcetera.  

Maybe muxing multiple mode-control commands/switches over a single RC channel would be even more useful. The standard 1000-2000 range is almost 10 bits, so it should be possible to encode 4 4-pos switches into a single RC channel and have almost two bits left to cope with arithmetic/rounding errors somewhere in the system. OpenTX is capable enough to do this mixing on the TX side. If only BF could unmix these and create logical channels... 

Regarding the coding: I managed to kill the Split (and much more) by hitting a steel soccer goal post at high speed. Replacement parts are on their way. Did not find SWD/SWCLK pads on the FC either, and soldering to the pins of the uC is going to be hard also since I put a nice protective layer of coating on thet board. Bummer. But we probably have a couple of F4 Discovery kits here at work, lets see if I can get one of those.

closed 2019-01-10 by stale[bot]