betaflight/betaflight #7084: RXLOSS

emorffpv, 2018-11-13:
I write here because I just tried all the solutions.
Since a couple of months  I get often RXLOSS during flight, and falls down. But happens only on one of my quad. RSSI is always high, and I fly n 50-60 meters max.
What I tried:
-another radio (taranis qx7 and another qx7)
-different rx (2 frsky r-xsr and a R9 Mini with R9 module)
-two brand new Flight controller (fortini F4 OSD)
-different antenna position
-different location
-checked each soldering joint and remake all

This happens always with BF 3.5 and 3.4 (on each lipo), sometimes with 3.3 and rarely with 3.2.
I can exclude components derivated problems, because I changed all.
My suspect is a wrong configuration on BF about filtering, because the only thing I changed is PT1 from BIQUAD, first to have a new RXLOSS in BF 3.2

etracer65, 2018-11-13:
The fact that it's only happening on one of your quads is a a clear indication of a hardware or component problem.

For hardware troubleshooting try these other resources:

https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

https://slack.betaflight.com

emorffpv, 2018-11-13:
yes, but I changed ALL hardwares, and on other quad I have older version of BF and no PT1.
So I can exclude hardwares

I forgot, I use D16 and 9 channel..if helps

jflyper, 2018-11-14:
@emorffpv 
Go back to 3.2 and test if the issue disappears when you changed PT1 to BIQUAD.

emorffpv, 2018-11-14:
never happened with BIQUAD. 
I was reading around about some issues withD16 and  more than 8 channels..

jflyper, 2018-11-14:
Nnnn... I wanted you to actually go back to 3.2 and test if the issue develops or not...

BatesKG, 2018-12-18:
I am having the exact problem on 2 other quads.  I thought the problem was more related to the R9M system.  I will check what settings I have on the weekend for the filters and post back.  My quads fly fine for a few flights and then just drop out of the sky.  It's really annoying!   This was on 3.5.3.

BatesKG, 2018-12-19:
Well.....I hovered for about 30 packs in my basement tonight.   I thought the switching to BIQUAD filter had fixed it......but then it did it two more times in the last couple packs.   Mid-flight you get RXLOSS in the OSD for a split second and then it's like you threw the disarm switch, the motors stop, and the stats come up.   I also noticed on one occasion the MinRSSI in the OSD was reported as 5%, yet on the Horus x10s, the telemetry srcreen read the minRSSI as 94%.   I have the RSSI wire running from the R9 slim+ to the Matek F722 RSSI pad.  How would the two values be so far apart?  Normally they are nearly identical on this quad.
This was on the latest BF 3.5.4.

poulpito, 2018-12-22:
i'd like to add some information as a friend and i are experiencing the same behaviour on betaflight quad (we both have other hardware running inav or naza with same TX/RX without any issue) 

random RXLOSS (with sensor lost some time before or not)  at short range (10 to 20m) after a random time but usually between 1/2min 
with R9 + R9mm (try with 10m 200m and 500m on TX) 
and we can recover almost immediately as i was able to disarm/re-arm before the quad hit the ground (but too late )  

Here are some screenshot  at rxloss (rssi was 98% before) 
and right after the crash with stats when i disarm 

[image]

[image]

hardware on this one is diatone F405 mamba with lastest stable betaflight 
R9mm & R9 with latest FCC firmware 

i'm running the same hardware on a TBS discovery on same spot without any issue

BatesKG, 2018-12-25:
Could one of the developers tell me where BF gets the data to determine "RX Signal Received" and "RX Flight Ch. Valid" flags?  (from the Blackbox logs)    I have a new quad that won't make more than 10 minutes of flight before just dropping out of the air....usually while the RSSI is in the 90 range.    I'm trying to use an R9 Slim+ receiver with a Matek F711 / Spedix GS40 4in1 ESC / Tramp HV Vtx / Matek GPS-Compass module.    Everything is going fine when out of the blue, the RX Signal Received goes to 0 (from 1) and a short while later the RX Flight CH. Valid goes to 0 and the quad shuts down.  Sometimes the signal received flag goes to 0 but comes back soon enough that the "valid" flag doesn't trip.   I've tried 3 different R9 Slim+ receivers, two different wire harnesses, and multiple antenna layouts.   Nothing helps.  Here is a Blackbox log with a cutout right at the end.  Just remove the .txt extension from the end.....I'm surprised log files aren't supported here?
[LOG00040 .BFL.txt](https://github.com/betaflight/betaflight/files/2709007/LOG00040.BFL.txt)
  Any suggestions of what is going on would be appreciated.  Everything is new.
I was starting to lean towards a HW/SW incompatibility of some kind with BF the the R9slim+ specifically.  So, I wired a Furious 2.4 Ghz receiver instead.   No more cutouts.    Any ideas?  

BTW, I've tried the last stable firmware for the R9Slim+ as well as the Flex, and latest FCC beta firmwares.  All have the dropouts.   The latest versions also have a glitch where the current telemetry will suddenly have an impossible value (like a million amps) which throws off the Power and mah used values as well.

Pretty much ready to give up and go Crossfire at this point....except that I have a Hours x10s.....which leads to more issues to be able to use CF.

Thanks!    and Merry Christmas!

BatesKG, 2018-12-27:
To follow up, I went to the field today and flew 6 packs back to back without a single hiccup using the Furious 2.4GHz Rx.    There must be some kind of "RX Signal Received" bug between the R9slim+ and BF?  Seeing that the FC thinks it has lost signal even when the RSSI reports fine at the radio and in the logs.   
I'm using the RSSI signal wire from the Rx to the FC.   The cutouts happened when I tested using F.Port firmware as well.  Is there any CLI config regarding how BF monitors "RX Signal Received"?
Thanks,

KevinB

BatesKG, 2018-12-31:
Frsky just posted new firmware for the whole R9 line.   Bullet #1 for fixes implemented........"1. Fix the random incorrect RSSI issue".   I'm guessing the random values were low, and would nicely explain why my quad was randomly falling out of the sky in FC failsafe.

I updated one of my R9 quads and put 4 packs through it without issue.  That one had been working ok and only had a few unexplained RXLOSS all last season.   It then had the motors pulled for another project and sat until now.    I'll reinstall the R9slim into the quad that was giving me the recent problems and test fly it when the weather warms up a bit.   Hopefully, this is it.

KevinB

poulpito, 2018-12-31:
Same thing here
Had a rx loss this afternoon with some damage... 

notice the flex upgrade (not only flex, all) and update all my r9 rx/tx
will try tomorrow if it's better or not but at least rssi on aux12 now display properly each time i boot the Quad on betaflight
(was random before)

siamect, 2018-12-31:
I have something similar with Crossfire. It's been on and off since I bought crossfire almost a year ago. Two receivers, three tx modules (two micros and one big crossfire) all have same problem. Currently frying the TBS help desk again but it may be a Betaflight glitch. The thing is falling out of the sky at close distance 20m sometimes.

Lazy8onez, 2019-01-04:
@BatesKG can you confirm the new Firmware (181228) fixed that issue?

I have this issue on two of my quads, both using R9 mini running 181116 firmware (which is not available for download anymore).
My tx is R9M on X9D.

BatesKG, 2019-01-05:
I've flown about a dozen packs so far on the new 181228 production firmware.   I haven't had a single dropout since switching to this version.  Apparently the randomly incorrect RSSI is what was tripping the flight controller into failsafe.
Now they just need to fix the random high Current telemetry values being sent back to the Tx over S.Port/F.Port.   I have it happen quite often where the tx suddenly displays over a million mAh used in the consumption telemetry value.  This is caused from an incorrect Current telemetry value which I can see in the Max Amps recorded for the flight.  The Power value is also thrown of by the phantom current spike.   It appears to be totally random like the RSSI issue.   Probably happens more often though.   It only happens on my R9 quads.    When I converted the one to 2.4Gz (because of the RSSI problem), the random current spike telemetry went away as well.   Note, this incorrect telemetry is only visible on the Tx.   The values in the OSD remain accurate.

KevinB

mikeller, 2019-01-07:
@BatesKG:

> Apparently the randomly incorrect RSSI is what was tripping the flight controller into failsafe.

This is definitely not the case - the flight controller only uses the RSSI value for informational purposes, but not to detect a 'failsafe' state. But there are flags in the SBus protocol format that allow the RX to signal to the flight controller if frames were lost, of if the connection has been lost completely and failsafe should be initiated:

https://github.com/betaflight/betaflight/blob/4778ad6c0fa4cf2c145513601b1378a98901e6cd/src/main/rx/sbus_channels.c#L76-L85

My guess is that older versions of the RX firmware did not only send misleading RSSI information, but also misinterpreted this data internally and sent `SBUS_FLAG_FAILSAFE_ACTIVE` or `SBUS_FLAG_SIGNAL_LOSS` to the flight controller.

Lazy8onez, 2019-01-17:
I have just upgraded the firmware on the R9M module and on one of my quads r9 mini Rx to 181228, It seem to have solved the issue.

mikeller, 2019-01-17:
@BatesKG: The telemetry protocol used over the radio link is rather crude, so it's well possible that the spikes are the result of transmission errors / missed frames.

BatesKG, 2019-01-18:
LazyBonez, do you see any of the Tx telemetry spike issues?   If not, what Tx/ OS are you using?   (OpenTx or FrOS?)   What FC board?
I'm also running 6s.   That could be making the issue worse than 4s.  I do have a capacitor on the battery input.   I'll be changing the motors on my 7" to go to strictly using 4s.   I'll see if there is any difference.
Thanks,
Kevin

poulpito, 2019-01-18:
Hi guys, additionnal question for frsky owners 
do you always have AUX12 filled with rssi value ?

from time to time AUX12 remain at 1500 (default value) on 2 different quad
* mamba F4 controller , Kakute F7  
* R9mm and R9mini flash to latest firmware
* BF 3.5.4 on first, 4.0.0 someday build on the other 

first is set on taranis with CH1-8  and the second one is with CH1-16 
(to see if radio configuration can help or not) 

I could not found what it causing this random rssi value missing. 
when i start flying with missing rssi, it can show up after some time and the work as expected. 

is there any way to track if received sbus trames are ok ?
any debug options or CLI ? 
i mean except when it's so bad that SBUS_FLAG_FAILSAFE_ACTIVE or SBUS_FLAG_SIGNAL_LOSS  are triggered 

thks

mikeller, 2019-01-19:
@poulpito:

> is there any way to track if received sbus trames are ok ?
any debug options or CLI ?
i mean except when it's so bad that SBUS_FLAG_FAILSAFE_ACTIVE or SBUS_FLAG_SIGNAL_LOSS are triggered

Not sure what you mean by this, do you want to analyse the time difference between received frames (`set debug_mode = FPORT`, `debug[0]` will give you this)? If a frame cannot be properly decoded after being received over the air, then the RX will simply not forward it to the flight controller (or forward a frame with `SBUS_FLAG_SIGNAL_LOSS`).

Lazy8onez, 2019-01-20:
> LazyBonez, do you see any of the Tx telemetry spike issues? If not, what Tx/ OS are you using? (OpenTx or FrOS?) What FC board?
> I'm also running 6s. That could be making the issue worse than 4s. I [...]

I will have to check the RSSI value. currently I am not monitoring it.
I am using X9D SE with R9M module. running OpenTX 2.2.1.
My board is OmnibusF4V6 running BF 3.5.4.
I have a 4s setup with a 1000uF 35V capacitor.

Regards.

yryz, 2019-02-15:
190201 | Fix the incorrect telemetry issue

https://www.frsky-rc.com/r9-mini/

BatesKG, 2019-02-19:
Yeah.....I just saw the telemetry fix.   I can't wait to try it.    Hopefully it fixes the problem in the R9 series.   I think at least part of the issue is in the Tx firmware as well though.   I'm running FrOS on a Horus x10s and still get the telemetry glitch on SPI-FrSky X connections to tiny Whoop controllers.    Or maybe it's just a poor implementation on the tinyW controller?   I've never had the issue when connecting to FuriousFPV receivers .

BatesKG, 2019-03-22:
Just to follow up.....
I've been out a few times since the telemetry fix firmware was released and have had no more telemetry value spikes with the R9 system.

closed 2019-04-28 by stale[bot]

Nullstr1ng, 2020-03-09:
Worst feeling ever. RXLOSS over a river
Flew the first battery with no issue, 2nd battery -- RXLOSS

I'm using Mamba F405 MK2 + XM+ 

[](https://www.youtube.com/watch?v=IHcpDPo-JgY)
[](https://imgur.com/a/Ov680o0)

Mutoryu, 2020-03-13:
Having this issue too.. 
Taranis 2019 with opentx 2.3.4
Getting rxloss randomly even at 5 meters from me...
The r-xsr receiver was flashed to access and now its on accss, nothing changed, still getting rxloss.
Checked all soldering. 
My copter hit the ground so much time because of that bug...

deanrather, 2020-03-23:
I'm seeing this on my new quad, too.

RSSI is reading high, sometimes 95%, and still getting RXLOSS FAILSAFE.
Half the time the signal comes back in time for the drone to recover, but on several occasions it didn't, and it fell to the ground.
It was reliably always at a certain altitude/distance/angle from myself, and I was under a metal roof as well...

X-Lite Pro TX with an XMPLUS RX. BF 4.0.4.
valid pulse range is 885-2115
failsafe guard time 0.4
failsafe throttle low delay 100

I'm new to this, and I just picked up the drone 2nd hand, but the seller seemed trustworthy, i suspect the issue is with my tx or antenna or maybe i mucked up the settings when i was re-binding it to my tx...

video collection of rx issues from my first flights here -- https://youtu.be/dH9k4O4aVos

Ceaserxl, 2020-05-31:
> Worst feeling ever. RXLOSS over a river
> Flew the first battery with no issue, 2nd battery -- RXLOSS
> 
> I'm using Mamba F405 MK2 + XM
> Betaflight 4.1.1 firmware
> 
> [link to youtube video](https://www.youtube.com/watch?v=IHcpDPo-JgY)
> [image]

I have this exact issue. I have a AKK reciever so I'm forced to use the patch from betaflight in order to use smart audio. Was your issue fixed after 4.1.1. Because i have two different quads with the exact same FC and Reciever, two different transmitters and it's the same issue.
I swapped from a omnibus f4 to the mamba. The omnibus was patched to 4.1.7 and i didnt see this problem.

mikeller, 2020-05-31:
@Ceaserxl:

> Edit: the mamba f405 mk2 doesn't have a patch above 4.1.1.

[image]

Ceaserxl, 2020-05-31:
> @Ceaserxl:
> 
> > Edit: the mamba f405 mk2 doesn't have a patch above 4.1.1.
> 
> [image]

The info page on it says FuryOSD (Legacy)

mikeller, 2020-05-31:
@Ceaserxl: That target has been disontinued - contact your supplier for support.

Ceaserxl, 2020-05-31:
Hmm, ill just flash it and see what happens, SEND IT!

Ceaserxl, 2020-05-31:
So just as an update the MAMBAF405US firmware was a bust. But I was able to flash the AKK patch for the FURY4OSD (DIAT) 4.1.7 with success. I haven't flown yet to see if it fixes the issues stated. But my smart audio works perfectly now it wasn't working properly previously.

mikeller, 2020-05-31:
@Ceaserxl: If 'FuryOSD (Legacy)' used to work then it is likely that 'FURY4OSD (DIAT)' will work as well - would just be good to know why the manufacturer specifically asks users to use an unsupported target.

Nullstr1ng, 2020-08-29:
[image]
  
had another RXLOSS with my new build with 93% RSSI then an RXLOSS warning,

Mamba F405 MK2 Mini flashed with FURYF4OSD 
```
2020-08-30 @ 00:23:57 -- Flight controller info, identifier: BTFL, version: 4.2.2
2020-08-30 @ 00:23:57 -- Running firmware released on: Aug 16 2020 01:46:41
2020-08-30 @ 00:23:57 -- Board: DIAT/FURYF4OSD(STM32F405), version: 0
2020-08-30 @ 00:23:57 -- Unique device ID: 0x1f0043374d500c20313048
```  
XM+ updated with ACCST_2.1.0_RSSI16_FCC  
Taranis X9D Plus updated with the same ACCST version.

illusive-c, 2020-12-05:
This happens to me with my TX16S and FrSky XM+ on a tiny whoop. I use the same radio on my 250 quad to fly to 1.5km. So I know the radio and the module is good. 

The strange thing I noticed is, the whoop only gets the RXLOSS when it's literally next to me. It never happened when it's more than 2 meters away. I'd chalk this up to a dead zone in the antenna's transmission area but I get RSSI on screen and it never drops below 95%

I'm thinking maybe somehow XM+ is sending an invalid packet and BF is thinking that's a sign of signal loss?

mikeller, 2020-12-05:
@illusive-c:

> The strange thing I noticed is, the whoop only gets the RXLOSS when it's literally next to me.

This is a known limitation of the tiny whoop hardware - it does not have protection from 'swamping' of the receiver, which happens when it is too close to the transmitter.