betaflight/betaflight #7087: Serialpassthrough with softserial

ijustwannafly, 2018-11-14:
I have been attempting to flash some of my frsky rx's using serialpassthrough.

All of my quads are using f4 boards and as a result I have to use softserial for my smartport setup.

I have one omnibusf4pro and the rest are the dysf4pro.

On all the boards the rx is powered through a 5v pad that does not have power when USB is connected.

In betaflight I use the command "serialpassthrough 30 57600" where 30 is the port number assigned to the softserial port according to diff.

After setting the passthrough, disconnecting from betaflight (not the USB), and opening the frsky tool (run as admin, load firmware file) the frsky tool goes into the "finding device" mode.

After powering the rx the frsky tool will see the rx and display hardware/firmware versions.  When clicking download it goes straight to "device timeout".

Does serialpassthrough not work at all with softserial?

jflyper, 2018-11-14:
It is at least working until the device is found.

After that, we have to scope the line what is going on to further investigate the issue.

ijustwannafly, 2018-11-14:
Is it possible that the softserial is capping the baud rate?  I thought I read somewhere that softserial has baud rate limitations for certain cases.

I thought it might be a timing issue but a bardwell video showing the same process had him sitting at the step of the frsky tool seeing the firmware and he selected download a while after it was detected.

jflyper, 2018-11-14:
No speed capping with F4.

I'm suspecting the tool is using some unknown method to toggle something that works with UART but not with software serial.

KPeng17, 2018-12-13:
I'm having exactly the same problem here. Was anyone lucky to find a solution?

jflyper, 2018-12-14:
Dumb question.

Does the rx flashing method use two wires between FC and RX or one?

ijustwannafly, 2018-12-14:
@jflyper 

It should be one wire, either sbus or smartport depending on rx.

When flashing from my taranis I use the smartport wire or sbus wire (xm+) but never both data wires (sbus and smartport) at the same time.

jflyper, 2018-12-14:
@ijustwannafly Which version you are trying this with?

ijustwannafly, 2018-12-14:
Bf 3.5.3 on the fc boards.

jflyper, 2018-12-14:
What is the message printed when you execute CLI `serialpassthrough 30 57600`?

ijustwannafly, 2018-12-14:
# serialpassthrough 30 57600
Port 30 already open, setting baud = 57600.
Forwarding, power cycle to exit.

jflyper, 2018-12-14:
I'm sorry for small question...

The SS port under question is configured as F.Port/S.Port when flying, and it's working, correct?

ijustwannafly, 2018-12-14:
Yes, the softserial is configured as smartport and I receive telemetry data in flight.

mikeller, 2018-12-15:
@ijustwannafly: Have you tried using `serialpassthrough 30`? The speed should be picked up from how the port is configured.

KPeng17, 2018-12-15:
@mikeller: I have the exact same symptoms, i.e. telemetry is working fine but I cannot flash over softserial.
I tried it with `serialpassthrough 30` and `serialpassthrough 30 0` and even with `19200` and `57600`. I also set the baud rates of the USB VCP and SOFTSERIAL on the ports tabs to 19k2, 57k6 and 115k2 - all combinations. But unfortunately, none of these work, I always get the same "device timeout" when clicking the Download button in the Frsky tool.

mikeller, 2018-12-15:
This seems to be a hard one. The flashing process seems to be doing something (sending a break?) that the hardware UART supports, but SOFTSERIAL doesn't. I've contacted FrSky to try and get some information on what is done at this point in the process.

closed 2019-01-22 by stale[bot]

randomstash, 2019-04-07:
bumping this open again.

ijustwannafly, 2019-04-08:
Would also like to see this resolved.  I haven't updated any of my rx's since because my quads are all flying.

SlipstreamFPV, 2019-04-12:
Bump also. Too many f4 boards with FrSky receivers using softserial for smartport. Did anyone hear back from FrSky?

sansha, 2019-12-04:
I'd also love to have this possibility as I am in the same situation as you guys. Any update?

LouDnl, 2020-01-29:
I'm also running into this. I have multiple F4 boards where the smartport is on softserial. Can this be reopened?

AlpayY, 2020-11-02:
Me too. I have the same issue with a MATEKF405 on BTFL 4.1.1 and FrSky R-XSR Receiver on Softserial.