betaflight/betaflight #7088: RSSI in osd without configuration

ijustwannafly, 2018-11-14:
Not entirely sure if betaflight is somehow reading the RSSI value when it is not configured.

Two quads: both using the frsky r9 system (r9m lite on x-lite with r9 mini).

One quad is using the omnibusf4pro and another is using the dysf4pro.

The r9m lite and the r9mini rx's both currently using FCC firmware (non-flex) with RSSI out on sbus 16.

Because I set the channels to 1-8 I have the RSSI channel in the receiver tab set to disabled.
In cli I set max aux channels to 4.

I have no functions on the tx that sends RSSI back to the rx from the tx.  I have a function on the tx that calls out my RSSI value especially since going to 8 channels and all sbus output keeps the RSSI on channel 16.

On the omnibusf4pro with RSSI channel disabled and RSSI on the osd enabled it reports RSSI value steadily.  It does drop to 0 when the tx is off and fluctuates when flying.

On the dysf4pro with RSSI channel disabled and RSSI on the osd enabled it reports RSSI value but goes from 0 to 99 even when the tx hasn't lost the signal.

Is it a fluke that the omnibusf4pro with the r9mini and sbus RSSI out on 16 can still read and report the RSSI value over osd despite the RSSI channel in the receiver tab being disabled?

jflyper, 2018-11-14:
Can we see your mysterious configuration as CLI `diff` output?

ijustwannafly, 2018-11-14:
I can get the diff of both quads after work today.

Eganwp, 2018-12-12:
RSSI is baked into the firmware by default on Aux 4 for nearly all FrSky (openTX) rx’s. They will report RSSI no matter what BF is telling it to do because it’s a function of Aux 4 on the radio link, unaffected by BF. So whether you set Aux 4 as the RSSI or not, the radio is still feeding BF RSSI values over Aux 4, hence why you’re seeing this. 

We could make it so setting RSSI channel to Disabled in configurator disables the RSSI value on the OSD, however a lot of noobs who don’t set this wont benefit from RSSI in the their osd by default.

BatesKG, 2018-12-28:
So if the FrSky is sending RSSI over Aux4 regardless of the setup of BF, and in the above person's case, BF is using that info for the OSD with no configuration....if I run the RSSI wire from my R9slim+ to the FC and turn on the analog RSSI switch, is BF using the RSSI from the extra wire?   Or from Aux4?   
 
I'm having a problem with the FC interpreting a loss of Rx and tripping the failsafe while the RSSI on my Tx and in the BB logs says RSSI is fine.   Just wondering if the failsafe system is maybe using this other RSSI source that may or may not be correct???

Thanks for any additional info you can provide on this aux4 RSSI.
KevinB

closed 2019-02-04 by stale[bot]