betaflight/betaflight #7089: Motors going ~50% speed when arming

xen0nnn, 2018-11-14:
**Describe the bug**
After updating to firmware 3.5.2 the motors start spinning uncontrollably after arming. The drone takes off immediately. This only happens when airmode is turned on.

I was running betaflight version 3.2 before and everything seemed to be working alright. I only had problems with oscillations, which made me upgrade to version 3.5.

**Flight controller configuration**
DSHOT1200
Motor Idle Throttle: 4.5%
8kHz Gyro
4kHz PID
SBUS receiver: UART1
OSB, ANTI_GRAVITY, DYNAMIC_FILTER, AIRMODE ON
Stock PID
Filter Settings off
[log.zip](https://github.com/betaflight/betaflight/files/2581661/log.zip)

**Setup / Versions**
- Racerstar TattooF4S 30A BLHELI_32 4in1
- Version 3.5.2 (LUX4)

I attached the log of the "flight"
First log was with props on, the drone took off about 20m and then I disarmed and it crashed into the ground. 
Second and third log are tests without the props on.

jflyper, 2018-11-14:
How did you configure your 3.5.2? Did you try to start with a fresh config and manually enter least configs?

xen0nnn, 2018-11-14:
Yes all the configurations match the old ones, except for the dynamic filtering. But even when the dynamic filter is turned off, the motors are spinning randomly.
What is weird is, that the gyroscope values change very rapidly even when no props are mounted. I think there is an issue with the sensors.

wind0r, 2018-11-14:
Filter Settings off .. This never will work.
Please use a sane filter setup and do not blindly disable everything.

Also please provide a diff as stated in the issue template

closed 2018-12-21 by stale[bot]