betaflight/betaflight #7102: HID USB Joystick doesn't work on Android phone

lmnn, 2018-11-18:
**Describe the bug**
FC with () connected to Android phone via usb otg cable. The phone (e.g. cursor in Home screen or controls in FPV simulator) does not react to transmitter sticks.
The same setup works fine if FC is connected to Windows 10 laptop. It works fine if the transmitter is connected to the phone directly using the same otg cable.
FC - Matek f405-CTR, the phone - Xiaomi Redmi 5 (Android 7.1.2),

**To Reproduce**
Connect receiver with FC. Bind the receiver with the transmitter, configure BF to be controlled by the receiver. Set usb_hid_cdc to on, save configuration, connect to Android phone. Check if FC acts as a joystick.

**Expected behavior**
FC with usb_hid_cdc set to on acts as a joystick on Android phone.

**Flight controller configuration**
```
# version
# Betaflight / MATEKF405 (MKF4) 3.5.3 Nov 15 2018 / 00:01:03 (61d5abf00) MSP API: 1.40
board_name MATEKF405
manufacturer_id 
# name
# resources
# mixer
# servo
# servo mix
# feature
# beeper
# beacon
# map
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# master
set min_check = 1010
set max_check = 1990
set serialrx_provider = IBUS
set deadband = 7
set yaw_deadband = 7
set usb_hid_cdc = ON
# profile
profile 0
# rateprofile
rateprofile 0
```
**Setup / Versions**
Flight controller: Matek F405-CTR (https://www.banggood.com/Matek-Systems-BetaFlight-F405-AIO-Flight-Controller-Built-in-PDB-5V-2A-9V-2A-Dual-BEC-for-RC-FPV-Racing-Drone-p-1165338.html),
receiver: Flysky FS-iA6B (https://www.banggood.com/Flysky-2_4G-6CH-FS-iA6B-Receiver-PPM-Output-With-iBus-Port-p-978603.html),
transmitter: Flysky FS-i6S (https://www.banggood.com/Flysky-FS-i6S-2_4G-10CH-AFHDS-2A-Transmitter-With-FS-iA10B-Receiver-p-1061496.html),
receiver connected to FC's RX2 pad (UART2), uses IBUS protocol.

**Additional context**
Android log attached (FC connecting to phone moment logged
[android_log.txt](https://github.com/betaflight/betaflight/files/2592716/android_log.txt)
).

mikeller, 2018-11-20:
@lmnn: What device driver are you using on the android device, and how did you install it? As far as I can tell, android does not support joystick devices natively, and apps wanting to use joystick devices need to supply their own drivers, so this is not a limitation in Betaflight.

lmnn, 2018-11-22:
@mikeller I didn't install any drivers. The transmitter works out of the box as a joystick via otg cable.
Android log after connecting the transmitter:
```
2018-11-22 22:38:13.136 1580-2473/? W/ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:914 com.android.server.usb.UsbSettingsManager.deviceAttached:739 com.android.server.usb.UsbHostManager.endUsbDeviceAdded:225 com.android.server.usb.UsbHostManager.monitorUsbHostBus:-2 com.android.server.usb.UsbHostManager.-wrap0:-1 
2018-11-22 22:38:13.137 1580-2031/? D/EventHub: No input device configuration file found for device 'Flysky FS-i6S emulator'.
2018-11-22 22:38:13.139 1580-2473/? W/ResourceType: No package identifier when getting name for resource number 0x00000000
2018-11-22 22:38:13.146 1580-2031/? W/EventHub: Unable to disable kernel key repeat for /dev/input/event7: Function not implemented
2018-11-22 22:38:13.147 1580-2031/? I/EventHub: New device: id=9, fd=379, path='/dev/input/event7', name='Flysky FS-i6S emulator', classes=0x80000101, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true
2018-11-22 22:38:13.147 1580-2031/? I/InputReader: Device added: id=9, name='Flysky FS-i6S emulator', sources=0x01000111
````

mikeller, 2018-11-23:
@lmnn: I suspect that there is a built in list of device ids that are recognised as joystick devices in android (or potentially in an android app making use of the joystick) - your report suggests that this list is probably specific to device / manufacturer.

I am not an expert for low level device support in android, but I suspect making our HID device supported is potentially something that needs to be done on the android side and not in the firmware. If there are changes needed in the firmware to make this possible then I am all for making them.