betaflight/betaflight #7106: problem with telemetry on Kakute F7 AIO + ESC 4in1 ver. 3.5.X

CpeDratDoPC, 2018-11-19:
**Describe the bug**
i have a problem with telemetry on Kakute F7 AIO + ESC 4in1

**To Reproduce**
BF 3.4.1 telemetry work (more than 32 entries) but without GPS
BF 3.5.1 - 3.5.3 only works telemetry from RSSI + RxBat but not from F7 ESC and no GPS
in Betaflight  satelites OK,  FIX GPS OK,
 but in telemetry is without information (BF 3.4.1 - 3.5.3)

**Expected behavior**
3.5.3 trelemetry OK.

**Flight controller configuration**
``` 
# diff all

# version
# Betaflight / KAKUTEF7 (KTF7) 3.5.1 Sep  8 2018 / 05:25:40 (d9fb5ca13) MSP API: 1.40

board_name KAKUTEF7
manufacturer_id 
mcu_id 003700323436510731393931
signature 

# reset configuration to default settings
defaults nosave

# name
name Faust

# resources
resource MOTOR 1 E09
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource SERVO 1 C09

# mixer

# servo
servo 0 1000 2000 1500 100 7

# servo mix

# feature
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon

# map

# serial
serial 3 2 115200 115200 0 115200
serial 6 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1675 2100 0 0
aux 1 1 1 900 2100 0 0
aux 2 13 4 1700 2100 0 0
aux 3 28 1 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = 63,38,9
set mag_hardware = NONE
set serialrx_provider = FPORT
set min_throttle = 1000
set use_unsynced_pwm = ON
set motor_pwm_protocol = BRUSHED
set motor_pwm_rate = 16000
set smartport_use_extra_sensors = ON

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
``` 
**Setup / Versions**
Kakute F7 AIO
Tekko32 4in1 35A
Frsky X4R  	FW:FPORT_EU_180111.frk
Horus X12S 	FW:OpenTx 2.2.2
GPS Beidou 220 	FW: 3.0.1 (Galieo fix OK)
BetaFlight 10.4.1

**Additional context**
smartport_use_extra_sensors = ON / OFF  unchanged

mikeller, 2018-11-22:
@CpeDratDoPC:

> set smartport_use_extra_sensors = ON

Set this to `off` and it should work.

CpeDratDoPC, 2018-11-22:
Even he does not solve the problem with telemetry

mikeller, 2018-11-23:
@CpeDratDoPC: Ok, try updating the firmware on your RX to this one: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-437283806

CpeDratDoPC, 2018-12-08:
By canceling the F.port to X4R  and by connecting the SBUS but  Telemetrie with another wire, the problem is solved especially in BF 3.5.X only GPS does not work!

smpotter, 2019-01-10:
I followed Bardwell instructional for 3.5. All worked fine. I updated to betaflight 4.0 and now no telemetry, no LUA script are blank. What happened?

closed 2019-02-16 by stale[bot]

CpeDratDoPC, 2019-05-04:
Solved.
Run in telemetry Detect new sensors only when GPS has FIX!

smpotter, 2019-05-07:
Is this thread only referring to "GPS" not working with lua? Because it appears I don't have telemetry now either, although I thought I was running 4.0 . I also don't want this thread to stop.