betaflight/betaflight #7108: SPRacingF7DUAL no DSHOT on M5/M6 outputs.

hydra, 2018-11-20:
**Describe the bug**
Customer reported motor M5 and M6 do not produce any DSHOT output signals. PWM/Oneshot was OK, Programming ESCs was OK.

**To Reproduce**
Enable M5-M8 outputs by disabling onboard RTC6705 VTX support by commenting out the following lines in `src/main/target/SPRACINGF7DUAL/target.h`  - This is a known issue, further details here https://github.com/cleanflight/cleanflight/issues/3049#issuecomment-405357538

```
//#define USE_VTX_RTC6705
//#define USE_VTX_RTC6705_SOFTSPI
//#define VTX_RTC6705_OPTIONAL    // VTX/OSD board is OPTIONAL
//
//#define RTC6705_SPI_MOSI_PIN                PB0  // Shared with PWM8
//#define RTC6705_CS_PIN                      PB6  // Shared with PWM5
//#define RTC6705_SPICLK_PIN                  PB1  // Shared with PWM7
//#define RTC6705_POWER_PIN                   PB7  // Shared with PWM6
```

Build, flash.

Reset to defaults using CLI commands (or full-chip erase when flashing)

```defaults```

Then

```mixer OCTOFLATX
set motor_pwm_protocol = DSHOT1200
save```

Observe that M1-M4 and M7/M8 produce DSHOT signals while M5/M6 do not.  Low signal when observed on a scope.

**Expected behavior**
M5/M6 to produce valid DSHOT signals.

I did a git bisect and the issue commit that breaks it is this one:

https://github.com/betaflight/betaflight/commit/bf984f39b14c685d20924e05874e4fc53b8df635

It's been broken for a while it seems.

hydra, 2018-11-20:
The line in the commit mentioned above that causes the issue is this one.

```
#define ENABLE_DSHOT_DMAR       true
```

https://github.com/betaflight/betaflight/commit/bf984f39b14c685d20924e05874e4fc53b8df635#diff-97f29bb4b146044c9af3cbb69eeb70eaR34

When that line is commented out it works again, so there's an implementation issue in the DMAR code that needs investigating and fixing when someone has time.

hydra, 2018-11-20:
having a little look into this reveals that `motor_DMA_IRQHandler` is never called for TIM4.

hydra, 2018-11-20:
For users with this issue, the current workaround is to disable burst dshot in the cli, as follows:

```
set dshot_burst=ON
save
```

hydra, 2018-11-20:
An alternative workaround is to disable the LED strip feature.

The issue is caused by a resource conflict on DMA1 STREAM 6 between TIM4_UP and TIM2_CH2.

TIM4_UP is used for M5/M6 when using Burst DMA.
TIM2_CH2 is used for LED strip.

mikeller, 2018-11-20:
@hydra: Have you checked if it is possible to use an alternative timer arrangement that supports DShot DMAR **and** LED_STRIP concurrently?

hydra, 2018-11-20:
at first glancenit seems like I could swap tim5 and tim2 but when i tried it didnt work. needs more investigation but the workarounds are ok for now.

mikeller, 2018-11-22:
Could be a DMA conflict with something else, like SPI. I don't have the hardware so I can't provide much help with solving it.

mikeller, 2018-11-22:
@hydra: No, I never got one of those - if you sent it it must have been lost in the mail.

hydra, 2018-11-22:
Pretty sure i sent one, PM me your details again on Slack and I'll send again.

hydra, 2018-12-31:
This still needs investigation. Pinned to remind me.

hydra, 2023-04-11:
Closing, will create a new issue if the problem if found to exist still.

closed 2023-04-11 by hydra