betaflight/betaflight #7110: Is "applyBetaflightRates" wrong?

klutvott123, 2018-11-20:
Or "rcCommandRawToDegreesPerSecond"(in configurator)?

There is a difference in the way rate curves are calculated in betaflight and betaflight configurator.

In Betaflight the "rcCommandfAbs" used when applying super rate is taken directly as an input argument.
https://github.com/betaflight/betaflight/blob/f57b3f3dc7fb09fc1aa142153bbfea3e394d9d5a/src/main/fc/rc.c#L119

In the configurator the absolute value is calculated from "rcInput = inputValue / maxRc", where "inputValue" has already been modified when applying expo.
https://github.com/betaflight/betaflight-configurator/blob/27cf3a356b5a585097f6746417a5bbfc5a2fc77f/src/js/RateCurve.js#L83

Here's a comparison:
[image]

Which one is correct?

mikeller, 2018-11-20:
@klutvott123: If in doubt, the configurator is wrong. The rates calculation is often improved in the firmware, and the calculation in the configurator which was implemented as an emulation of what the firmware is doing is not updated to reflect the changes.

McGiverGim, 2018-11-20:
I modified the way the rates were calculated in the latest version of the configurator, maybe I made some mistake.

klutvott123, 2018-11-20:
@mikeller Thanks for clarifying!

@McGiverGim I just made a pull request that makes the calculation the same a it currently is in betaflight. Check it out

McGiverGim, 2018-11-20:
@kluvott123 thanks for this! I will check it soon, now I'm on mobile and is difficult to me 😉

klutvott123, 2018-11-20:
@McGiverGim 👍 😄

closed 2018-11-20 by klutvott123

McGiverGim, 2018-11-21:
Seeing the code, I modified it for the Blackbox Explorer, but not for the Configurator, so this time the mistake was not mine :) I will check now your PR.