betaflight/betaflight #7120: iBUS failsafe OMNIBUSF4

smarq8, 2018-11-24:
I setup mu FS-i6X failsafe throttle channel to produce 985 pulse and i setup BF rx_min_usec=990. Now I should expect failsafe after turning off my receiver but BF always set my THROTTLE channel to rx_min_usec value and I'm unable to enter failsafe independet of what values I use. In short, any received value is unable to go below rx_min_usec.

my FW OMNIBUSF4 3.5.3

mikeller, 2018-11-24:
Please provide the information you were asked for when opening the issue.

jcutcher01, 2018-11-25:
From my search FS-i6x is your transmitter. When the RX loses signal that means no reception from the transmitter so your opening statement makes no sense. Failsafe settings are required in your receiver. Most receivers let you set the desired channels outputs (i.e. throttle) and then Betaflight can be set accordingly to respond as a Failsafe. I always use “CUT” on my crossfire receivers and that is detected by BF as a Failsafe.